1technophile/OpenMQTTGateway #564: LoRa gateway - Guru Meditation error - panic

spattinson, 2020-03-02:
Before submitting a problem please check the troubleshooting section
https://github.com/1technophile/OpenMQTTGateway/wiki/Troubleshooting

**Describe the bug**
LoRa gateway - Guru Meditation error - panic

**To Reproduce**
git clone default branch today. Setup Lora parameters for packets, change LoRa RST pin to the correct one for V1.0 TT-Go T-Beam (23). Setup wifi in user config. Build and flash with PlatformIO/VSCode

**Expected behavior**
No crash

**Screenshots**
```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
N:
************* WELCOME TO OpenMQTTGateway **************
T: Connecting to home
T: .
N: WiFi ok with manual config credentials
T: OpenMQTTGateway mac: 24:6F:28:B2:00:44
T: OpenMQTTGateway ip: 192.168.1.226
T: Port: 1883
T: Mqtt server connection by IP: 192.168.1.217
N: Setup BME280 on adress:
N: Bosch BME280 Initialized - Result of .begin(): 0x
N: LORA_SCK: Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400014dc  PS      : 0x00060c30  A0      : 0x800d9468  A1      : 0x3ffb1e30
A2      : 0x00000005  A3      : 0x00000001  A4      : 0x000000ff  A5      : 0x0000ff00
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x80088ddc  A9      : 0x3ffb1de0
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000001
A14     : 0x00060c20  A15     : 0x00000000  SAR     : 0x0000000e  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000005  LBEG    : 0x400da658  LEND    : 0x400da6ba  LCOUNT  : 0x00000002

Backtrace: 0x400014dc:0x3ffb1e30 0x400d9465:0x3ffb1e40 0x400d953d:0x3ffb1e60 0x400d54c6:0x3ffb1e80 0x400d5529:0x3ffb1ea0 0x400d1bb2:0x3ffb1ed0 0x400d1c7a:0x3ffb1f20 0x400d1ccc:0x3ffb1f40 0x400d22a8:0x3ffb1f60 0x400db05b:0x3ffb1fb0 0x4008939d:0x3ffb1fd0
```
**Environment (please complete the following information):**
Version: Current default branch "git clone https://github.com/1technophile/OpenMQTTGateway.git" downloaded today, dont know where version is stored?

LoRa library version 0.5.0

**Additional context**
I decoded the stack trace using the tool from me-no-dev:
```
PC: 0x400014dc
EXCVADDR: 0x00000005

Decoding stack results
0x400d9465: Print::write(char const*) at /home/scott/.platformio/packages/framework-arduinoespressif32/cores/esp32/Print.h line 66
0x400d953d: Print::print(char const*) at /home/scott/.platformio/packages/framework-arduinoespressif32/cores/esp32/Print.cpp line 89
0x400d54c6: Logging::printFormat(char, __va_list_tag*) at .pio/libdeps/ttgo-lora32-v1/ArduinoLog/ArduinoLog.cpp line 188
0x400d5529: Logging::print(__FlashStringHelper const*, __va_list_tag) at .pio/libdeps/ttgo-lora32-v1/ArduinoLog/ArduinoLog.cpp line 99
0x400d1bb2: Logging::printLevel__FlashStringHelper const*>(int, __FlashStringHelper const*, ...) at .pio/libdeps/ttgo-lora32-v1/ArduinoLog/ArduinoLog.h line 285
0x400d1c7a: Logging::notice__FlashStringHelper const*, int>(__FlashStringHelper const*, int) at .pio/libdeps/ttgo-lora32-v1/ArduinoLog/ArduinoLog.h line 219
0x400d1ccc: setupLORA() at /home/scott/pio/OpenMQTTGateway/main/ZgatewayLORA.ino line 48
0x400d22a8: setup() at /home/scott/pio/OpenMQTTGateway/main/main.ino line 645
0x400db05b: loopTask(void*) at /home/scott/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp line 14
0x4008939d: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```
According to information on the espressif site for LOAD_PROHIBITED Guru meditation error, I have made the relevant part **BOLD**:

This CPU exception happens when application attempts to read from or write to an invalid memory location. The address which was written/read is found in EXCVADDR register in the register dump. If this address is zero, it usually means that application attempted to dereference a NULL pointer. **If this address is close to zero, it usually means that application attempted to access member of a structure, but the pointer to the structure was NULL**. If this address is something else (garbage value, not in 0x3fxxxxxx - 0x6xxxxxxx range), it likely means that the pointer used to access the data was either not initialized or was corrupted.

What it appeared to be doing is:
SPI.begin(LORA_SCK, LORA_MISO, LORA_MOSI, LORA_SS);
line 38 in ZgatewayLORA.ino
Those pins are defined and correct in config_LORA.h
#define LORA_SCK     5    // GPIO5  -- SX1278's SCK
#define LORA_MISO    19   // GPIO19 -- SX1278's MISO
#define LORA_MOSI    27   // GPIO27 -- SX1278's MOSI
#define LORA_SS      18   // GPIO18 -- SX1278's CS
#define LORA_RST     23   // GPIO14 -- SX1278's RESET
#define LORA_DI0     26   // GPIO26 -- SX1278's IRQ(Interrupt Request)
I have used the same config and sandeep mistry's library in another project with this board.
I tried commenting out bme280, as it seems to be not picking up the address, and was not sure of the pins and where they are defined. It still crashes.

1technophile, 2020-03-03:
Hello,

Thanks for the extended search of the issue.
Could you post the env that you are using ?

spattinson, 2020-03-03:
Hi here are the files I modified for my config, thanks for taking a look.
[scott.zip](https://github.com/1technophile/OpenMQTTGateway/files/4280940/scott.zip)

1technophile, 2020-03-03:
Hi, your env seems to miss:
`${libraries.bme280}`

If it still doesn't work, could you try without the BME280 sensor and without it into the configuration to see if it is coming from this.

spattinson, 2020-03-03:
Hi, I try with only bme280 that works and I get readings with it.
I try with only LoRa and it still gets guru meditation.
I have looked at the LoRa library, the one that gets installed with pio is version 0.5.0 I am writing the sender part and that uses version 0.7.0 and it does not crash on SPI.begin. I will try to use that library with OMG instead.

spattinson, 2020-03-03:
library versiom 0.7.0 still crashes. I ran it through the ESP Exception decoder again:
```
: LORA_SCK: Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400014dc  PS      : 0x00060c30  A0      : 0x800d9468  A1      : 0x3ffb1e30  
A2      : 0x00000005  A3      : 0x00000001  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x80088ddc  A9      : 0x3ffb1de0  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0x00060c20  A15     : 0x00000000  SAR     : 0x0000000e  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000005  LBEG    : 0x400da658  LEND    : 0x400da6ba  LCOUNT  : 0x00000002  

Backtrace: 0x400014dc:0x3ffb1e30 0x400d9465:0x3ffb1e40 0x400d953d:0x3ffb1e60 0x400d54c6:0x3ffb1e80 0x400d5529:0x3ffb1ea0 0x400d1bb2:0x3ffb1ed0 0x400d1c7a:0x3ffb1f20 0x400d1ccc:0x3ffb1f40 0x400d22a8:0x3ffb1f60 0x400db05b:0x3ffb1fb0 0x4008939d:0x3ffb1fd0

PC: 0x400014dc
EXCVADDR: 0x00000005

Decoding stack results
0x400d8571: Print::write(char const*) at /home/scott/.platformio/packages/framework-arduinoespressif32/cores/esp32/Print.h line 66
0x400d8649: Print::print(char const*) at /home/scott/.platformio/packages/framework-arduinoespressif32/cores/esp32/Print.cpp line 89
0x400d4f82: Logging::printFormat(char, __va_list_tag*) at .pio/libdeps/ttgo-lora32-v1/ArduinoLog/ArduinoLog.cpp line 188
0x400d4fe5: Logging::print(__FlashStringHelper const*, __va_list_tag) at .pio/libdeps/ttgo-lora32-v1/ArduinoLog/ArduinoLog.cpp line 99
0x400d19d2: Logging::printLevel__FlashStringHelper const*>(int, __FlashStringHelper const*, ...) at .pio/libdeps/ttgo-lora32-v1/ArduinoLog/ArduinoLog.h line 285
0x400d1c9e: setupLORA() at .pio/libdeps/ttgo-lora32-v1/ArduinoLog/ArduinoLog.h line 219
0x400d2066: setup() at /home/scott/pio/OpenMQTTGateway/main/main.ino line 645
0x400da167: loopTask(void*) at /home/scott/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp line 14
0x40088b7d: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```
I have googled the error without much luck so far.

spattinson, 2020-03-03:
I have updated the framework 1.11.2, I was one release behind.
Also changed the board type to ttgo-t-beam.
Still no luck

1technophile, 2020-03-04:
Do you have another lora board to test with like the ttgo lora32 oled?

spattinson, 2020-03-04:
I have only non lora boards ttgo t-display atm. i built a simple mqtt to lora gateway i found on github, that works but uses arduino ide, maybe i try building it with pio. I also built disaster radio with pio that works but it uses lmic library not sandeep mistry one.

spattinson, 2020-03-05:
I Have a solution to my issue using some other code that suits my needs, you can close this if you wish, or keep it open and I can retest for the next few weeks. I don't need to need to deploy my solution immediately. thanks!

spattinson, 2020-03-06:
Hi
Its https://github.com/imelekhin/LoRatoMQTT/blob/master/esp32mqtt.ino once i remove display related stuff because i dont have one. Its quite small small. The steps you make are the same. I think maybe you are hitting some incompatibility in framework or library. The other thing I saw  while googling was spi bus and running tasks on different cores?

KaVo194, 2020-03-25:
I'm experiencing exactly the same problem. I tested it with two different TTGO boards resulting in the same behavior.

`Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
N: 
************* WELCOME TO OpenMQTTGateway **************
T: Connecting to Voigtsnetwork
T: .
N: WiFi ok with manual config credentials
T: OpenMQTTGateway mac: CC:50:E3:B6:84:E8
T: OpenMQTTGateway ip: 192.168.0.124
T: Port: 1883
T: Mqtt server connection by IP: 192.168.0.3
N: LORA_SCK: Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400014dc  PS      : 0x00060a30  A0      : 0x800d8560  A1      : 0x3ffb1e50  
A2      : 0x00000005  A3      : 0x00000001  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x800885bc  A9      : 0x3ffb1e00  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0x00060a20  A15     : 0x00000000  SAR     : 0x0000000e  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000005  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400014dc:0x3ffb1e50 0x400d855d:0x3ffb1e60 0x400d8635:0x3ffb1e80 0x400d4f6e:0x3ffb1ea0 0x400d4fd1:0x3ffb1ec0 0x400d19ce:0x3ffb1ef0 0x400d1c9a:0x3ffb1f40 0x400d2062:0x3ffb1f60 0x400da153:0x3ffb1fb0 0x40088b7d:0x3ffb1fd0

Rebooting...`

KaVo194, 2020-03-25:
Here are the boards I used for testing:
[](https://de.aliexpress.com/item/32915894264.html?)
[](https://de.aliexpress.com/item/4000133691142.html?)

KaVo194, 2020-03-25:
Here are the boards I used for testing:
[https://de.aliexpress.com/item/4000133691142.html?spm=a2g0o.productlist.0.0.1fee666fF6PFoO&algo_pvid=10d4075f-192d-4f32-9f9f-19b76de4ff14&algo_expid=10d4075f-192d-4f32-9f9f-19b76de4ff14-10&btsid=0ab6d70515851486040486045e0c89&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_](url)
[https://de.aliexpress.com/item/32915894264.html?spm=a2g0o.productlist.0.0.1fee666fF6PFoO&algo_pvid=10d4075f-192d-4f32-9f9f-19b76de4ff14&algo_expid=10d4075f-192d-4f32-9f9f-19b76de4ff14-0&btsid=0ab6d70515851486040486045e0c89&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_](url)

spattinson, 2020-03-26:
Looks the same to me, its when spi is started for lora radio. Ive moved on because my needs are simple, receive one json sensor data and forward to mqtt, which i manage to write together myself.

wizzor, 2020-04-01:
Hi, I am also seeing the same issue with ttgo-lora32-oled board. 

On my MQTT broket (Home assistant based Mosquitto, running on container) I see the following:

`INFO] found USER on local database
1585768792: New client connected from 192.168.1.245 as DVES_A2A209 (p2, c1, k30, u'USER').
1585769012: Socket error on client DVES_A2A209, disconnecting.`

1technophile, 2020-04-27:
I have made a modification to the development branch, may you test it and say if it is working please?

KaVo194, 2020-05-02:
> I have made a modification to the development branch, may you test it and say if it is working please?

No crash anymore with TTGO LORA SX1278 ESP32 0.96 OLED. Thx so far, will continue testing.

1technophile, 2020-05-02:
> No crash anymore with TTGO LORA SX1278 ESP32 0.96 OLED. Thx so far, will continue testing.

Thanks for the feedbak!

closed 2020-06-11 by 1technophile