1technophile/OpenMQTTGateway #565: Compatible lora sx1276

lscosta90br, 2020-03-04:
Hello,

i get some months esp32 lora:
https://en.aliexpress.com/item/32838347451.html?spm=a2g0s.9042311.0.0.2742b90a53urcO
Can you tell if they are compatible with your firmware?

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

1technophile, 2020-03-04:
Hello, the url seems incorrect, could you correct it?

lscosta90br, 2020-03-04:
Excuse me,

Test it on the link and it's not in my purchase history

```code
Description:
Dominant Frequency: 240MHZ
Flash: 8Mbyte
Processor: Tensilica LX6 Dual Core
Master chip: ESP32
LoRa chip: SX1276
Support frequency band: 868-915MHZ
Open communication distance: 2.8KM
Computing capacity: up to 600DMIPS
Dual-mode Bluetooth: traditional Bluetooth and BLE low-power Bluetooth
Development environment: perfect support for Arduino
Operating voltage: 3.3-7V
Operating temperature range: -40-90 ℃
Receiver sensitivity: -139dbm (SD12, 125KHz)
UDP continuous throughput: 135Mbps
USB adapter chip: CP2102
Support mode: Sniffer, Station, softAP and Wi-Fi Direct
```
[image]
[image]
[image]

closed 2020-05-10 by stale[bot]

1technophile, 2022-10-25:
Now they are https://github.com/1technophile/OpenMQTTGateway/releases/tag/v1.0.0