1technophile/OpenMQTTGateway #567: LAN8720 for esp32

wouldgo, 2020-03-04:
**Is your feature request related to a problem? Please describe.**
I have a LAN8720 eth board working with an esp-wroom-32 dev kit board and I thought if it's possibile to use this configuration with OpenMQTTGateway (mainly due to proximity with an unused ethernet plug of my lan network). The main funcionalities I want are BLE + DHT sensors.

**Describe the solution you'd like**
Integrate LAN8720 board with the actual project to enable ethernet connectivity.

**Describe alternatives you've considered**
I looked at the code and `main.ino` seems the place where connection is established but I'm stuck into understanding how approach the way for achieving my goal.

Could you please point out if what I want to achieve is in or out of scope of this project?

Thanks a lot for your time,
Dario

1technophile, 2020-04-16:
Hi,

It is clearly in the scope of the project.
And in my point of view you are right on prefering Ethernet versus wifi.

Let me some time to check where the modifications has to be done.

debegr92, 2020-06-07:
Hi,
nice to read, that the LAN8720 is in the pipeline! Just wondering, why there are no ESP32 Ethernet BLE gateways. I also want to use some BLE devices in my home automation, but not with a WiFi<->BLE gateway. I want to use some LAN connected device. Are there any news about this feature?

debegr92, 2020-06-07:
We can use this board as some reference design to test the code. I try to get one of these to do some tests.
https://www.olimex.com/Products/IoT/ESP32/ESP32-GATEWAY/open-source-hardware

1technophile, 2020-06-07:
Does this one has the same module in your point of view?
https://www.olimex.com/Products/IoT/ESP32/ESP32-POE/open-source-hardware

debegr92, 2020-06-08:
**Both boards above have the LAN8710A chip**, which is very similar to the LAN8720A.
The LAN8720A has only RMII interface and is therefore a little bit smaller. The most breakout boards are using the LAN8720A.

1technophile, 2020-06-08:
Very interesting, I took contact with Olimex.
Moreover these boards proposes external antenna, which can be very interesting for the BLE range.

debegr92, 2020-06-08:
They also have some metal cases for this kind of gateway! So basically, it is possible to get a ready to use (great range) BLE gateway. Flash firmware and good to go! 😊

1technophile, 2020-06-08:
I prefer ethernet aslo as it is more secure than Wifi, I'm very enthusiast about these boards. Let's see how it goes with Olimex.

debegr92, 2020-06-08:
I will get my board hopefully next week.

1technophile, 2020-06-13:
Received,
As a first test it works out of the box with `esp32dev-ble` !

debegr92, 2020-06-19:
Yes! Got my board this week and works great! I hope to find some time to help programming and test with the LAN.

1technophile, 2020-06-19:
It would be great!

ndoo, 2020-06-23:
I [added support for Olimex ESP32-POE in another similar project](https://github.com/shmuelzon/esp32-ble2mqtt/pull/72)  and will see if I can spare time over the weekend to add Ethernet support to OMG.

If you would like reference code to work on this sooner, though, [Olimex publishes a sample sketch for Arduino framework too](https://github.com/OLIMEX/ESP32-POE/blob/master/SOFTWARE/ARDUINO/ESP32_PoE_Ethernet_Arduino/ESP32_PoE_Ethernet_Arduino.ino).

1technophile, 2020-06-23:
It would be great if you can help !  Let me take a look to provide you some guidance.

1technophile, 2020-06-26:
For info, OpenMQTTGateway can already use ethernet with a W5100 and an arduino.
You will find all the functions related to [wifi](https://github.com/1technophile/OpenMQTTGateway/blob/71f1f0c1eceeaa98c5ea6843c5ef2414e2a6f506/main/main.ino#L489)and [ethernet ](https://github.com/1technophile/OpenMQTTGateway/blob/71f1f0c1eceeaa98c5ea6843c5ef2414e2a6f506/main/main.ino#L502)in main.ino.
The choice between the appropriate setup function is made at build time by the board choice and [config options](https://github.com/1technophile/OpenMQTTGateway/blob/development/main/User_config.h).

1technophile, 2020-06-30:
Hello @ndoo,

Did you find time to work on this implementation?

ndoo, 2020-07-10:
Not just yet :(

1technophile, 2020-07-28:
Hi @ndoo ,

If you think you are not going to have time please say it and I will do it.

ndoo, 2020-07-29:
It does not look like something I’ll have time for in the next 3 weeks due to pandemic related developments at work, for which I have to apologize. 

So please have a go if you have the time, thank you.

1technophile, 2020-07-29:
Thanks for the info

1technophile, 2020-08-27:
I'm working on it for info

1technophile, 2020-08-28:
PR #747 available for test

wouldgo, 2020-09-01:
Hi I built via platformio the OLIMEX-ESP32-ETHERNET branch and burned on an Olimex ESP-POE. The build process went like a charm but as I look at the serial monitor I spotted this logs. 

If you need more info let me know and thanks a lot for your time and dedication.

```N: Ethernet Connected
N: BLE scans interval: 55555
N: BLE scans number before connect: 10
N: Publishing only BLE sensors: false
N: minrssi: 100
N: Low Power Mode: 0
N: Setup OpenMQTTGateway end
W: MQTT connection...
W: MQTT client disconnected no BLE scan
[E][WiFiClient.cpp:258] connect(): socket error on fd 55, errno: 113, "Software caused connection abort"
W: failure_number_mqtt: 1
W: failed, rc=-2
W: MQTT connection...
[E][WiFiClient.cpp:258] connect(): socket error on fd 56, errno: 113, "Software caused connection abort"
W: failure_number_mqtt: 2
W: failed, rc=-2
W: MQTT connection...
[E][WiFiClient.cpp:258] connect(): socket error on fd 57, errno: 113, "Software caused connection abort"
W: failure_number_mqtt: 3
W: failed, rc=-2
W: MQTT connection...
W: MQTT client disconnected no BLE scan
[E][WiFiClient.cpp:258] connect(): socket error on fd 58, errno: 113, "Software caused connection abort"
W: failure_number_mqtt: 4
W: failed, rc=-2
W: MQTT connection...
[E][WiFiClient.cpp:258] connect(): socket error on fd 59, errno: 113, "Software caused connection abort"
W: failure_number_mqtt: 5
W: failed, rc=-2
W: MQTT connection...
[E][WiFiClient.cpp:258] connect(): socket error on fd 60, errno: 113, "Software caused connection abort"
W: failure_number_mqtt: 6
W: failed, rc=-2
W: MQTT connection...
W: MQTT client disconnected no BLE scan
E: Ethernet Disconnected
[E][WiFiClient.cpp:258] connect(): socket error on fd 61, errno: 113, "Software caused connection abort"
W: failure_number_mqtt: 7
W: failed, rc=-2
W: Network disconnected:
W: ethernet
W: Network disconnected:
W: ethernet
W: Network disconnected:
W: ethernet
W: Network disconnected:
W: ethernet
W: Network disconnected:
W: MQTT client disconnected no BLE scan
W: ethernet
W: Network disconnected:
W: ethernet
W: Network disconnected:
W: ethernet```

1technophile, 2020-09-01:
I have not tested with   Olimex ESP-POE, but as this is the same ethernet component it should work.

Did you enter your MQTT credential in user_config.h or platformio.ini?

Alternatively, if this is not the problem could you rebuild it with the below log level (in User_config.h)
`#define LOG_LEVEL LOG_LEVEL_TRACE`

And post the whole trace

closed 2020-09-02 by 1technophile

wouldgo, 2020-09-02:
My bad, 
the issue was due to a checkout before switching branch that revert to default User_config.h

It sends data now.
Thanks a lot,

Dario

1technophile, 2020-09-02:
Glad you got it working!

1technophile, 2020-09-02:
@wouldgo, a question is the LED connected to GPIO 33 on the POE board ? Like the non POE one ?

hnaeem07, 2025-03-28:
any possiblity to RS232 to Ethernet, if so what pins i have to use, for 232 internface and ethernet pinouts. thanks in advance guys.