1technophile/OpenMQTTGateway #568: Filter 433 commands so that not all are published to MQTT.

wimpie007, 2020-03-13:
I get a lot of 'rubbisch' 433 commands on my gateway.
Polution you might call it. 
All these events are published to MQTT, wich leads to a lot of activity in my nodered/ HA-automations wich actually do nothing.
Also, for this reason, HA-autodiscovery is unusable: i used it a week, and got HUNDREDS of unused switches in HA.

**Describe the solution you'd like**
a filter-mechanism. (whitelist for example protocol-length(s). 24 would - in my case - work in almost all cases)

**Describe alternatives you've considered**
*dirty*-hack the code myself. :) You don't like that!:)

1technophile, 2020-03-21:
Indeed we could replicate the mechanism used for white & black list management of BLE devices

jumika, 2020-12-26:
+1 for this. My mqtt boker is flooded with all the unknown 433mhz device signals.

DigiH, 2021-08-02:
Hi all, I have already implemented a flag for suppressing unparsed 433/868 signals to the MQTT publish queue in NothernMan's rtl_433_ESP a while back. It's documented on https://github.com/NorthernMan54/rtl_433_ESP, but obviously didn't make it into the OMG documentation.

https://github.com/NorthernMan54/rtl_433_ESP/pull/7

While not a proper white/black-list implementation, it might help remedy your issues as well.

The build flag is

'-DSUPPRESS_UNPARSED=true'

Even with a black-list, this might still be needed, to avoid the MQTT publishing of unparsed/unknown devices.

Hope this helps.