1technophile/OpenMQTTGateway #570: Invalid device_class for Home-Assistant discovery

tht, 2020-03-19:
I've just started to use OpenMQTTGateway and use the current master branch (as of yesterday). Home-Assistant is also up to date (0.107.1). MQTT Auto-Discovery is used for a bunch of other sensors and it works perfectly fine in Home-Assistant. 

I tried to use OpenMQTTGateway to integrate a Xiaomi Scale into Home-Assistant using the Auto-Discovery feature.

The configuration for the scale gets published to MQTT but Home-Assistant was not able to integrate the sensor because of an invalid value in `device_class`. The gateway reports `device_class: weight` but there is no such device-class according to the Home-Assistant documentation: https://www.home-assistant.io/integrations/sensor/#device-class

Error logged in Home-Assistant:

 > exception in async_discover_sensor when dispatching 'mqtt_discovery_new_sensor_mqtt': ({'state_topic': 'opengw/OpenMQTTGateway_ESP32_BLE/BTtoMQTT/C847xxxxxxxx', 'name': 'MiScale-weight', 'unique_id': 'C847xxxxxxxx-MiScale-weight', 'device_class': 'weight', 'value_template': '{{ value_json.weight | is_defined }}', 'unit_of_measurement': 'kg', 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/sensor.py", line 81, in async_discover_sensor config = PLATFORM_SCHEMA(discovery_payload) File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: value is not allowed for dictionary value @ data['device_class']

As a result the Scale is not added to Home-Assistant.

It is easy to fix. The problem is on this line: https://github.com/1technophile/OpenMQTTGateway/blob/6fa40cb56da53a6c4aa3bd99bbdb00cf388c0a3e/main/ZgatewayBT.ino#L255

Replace the word `weight` by an empty string (`""`) and the Home-Assistant integration works perfectly fine.

Looks like there are quite a few other issues with invalid `device_class` values. There is no `steps`, `presence` and probably others as I haven't checked other files.

1technophile, 2020-03-20:
Hello,

Thanks for reporting this, would you be interested in submitting a Pull Request that corrects these points (from the development branch)?

lubeda, 2020-03-21:
sorry i saw this to late, i've got the same issue. but i don't know how to fix this  besides my "stupid" hack

PR #572 (2020-03-21): Fixed invalid device_class values
Removed device_class values unknown to Home-Assistant.

Fixes #570 and fixes #571.
commit 133d4c64 (2020-03-21): Fixed invalid device_class values (#572)

Removed device_class values unknown to Home-Assistant. Fixes #570 and fixes #571.
closed 2020-03-21 by 1technophile