1technophile/OpenMQTTGateway #573: Slow RF KAKU message parsing when using Pilight

schmurtzm, 2020-03-23:
**Describe the bug**
Hi, I use a Sonof RF bridge with Direct Hack, I recently buy [DIO plugs](https://chacon.com/en/remote-controlled-sockets/1013-set-of-3-on-off-sockets-and-3-channel-remote-control-dio-10-5411478548959.html) which use "newkaku" RF protocol. It doesn't work with the default firmware, so I have recompiled OpenMQTTGateway with RF2 -> not better and then with Pilight -> OK. With Pilight the signal is recognized but I'm a little disappointed by the speed : about 2 seconds between pressing the button and the parsing of the signal. Not very handful for lights in  domotic...

**To Reproduce**
Steps to reproduce the behavior:
1. Program an Sonof RF bridge with Pilight implementation
2. Send a KAKU RF message
3. Wait for the serial/MQTT message 

**Expected behavior**
It should take something like 500 ms maximum , isn't it ?

**Environment (please complete the following information):**
* OpenMQTTGateway version : Master and release v0.9.3 tested

**Additional context**
With my other RF remotes (not with KAKU protocol) it seems a little faster but it's still long compared to RF mode.

Thanks for your awesome work !

1technophile, 2020-06-07:
Maybe a solution to test would be to reduce the number of protocols built with Pilight to the ones you are using. I know how to do it on a raspberry but not on the ESP, maybe @puuu has an idea?

puuu, 2020-06-10:
@1technophile ESPilight offers the method `limitProtocols()` to limit the protocols used to parse the pulsetrain. It requires a JSON list as a string with protocols names. An empty list `"[]"` means all protocols. The file [`test_proto_limit.ino`](https://github.com/puuu/ESPiLight/blob/master/tests/test_proto_limit/test_proto_limit.ino) can be seen as an example.

I hope it helps.

1technophile, 2020-06-10:
@puuu thanks for the tip!

@schmurtzm would you be interested in testing this with my help if needed?

closed 2020-08-16 by stale[bot]