1technophile/OpenMQTTGateway #574: RGB Light Integration (PWM Output)

Tobiassss, 2020-03-24:
Don't be ashamed of your OpenMQTTGateway and hide it somewhere in an inconspicuous little box. Let it brighten the room like I did by combining it with a RGB LED.

Technically it is adding PWM Actuator functionality to the OpenMQTTGateway.

Sending `{"red":10,"green":20,"blue":30,"white":40}` to `home/OpenMQTTGateway_ESP32/commands/MQTTtoRGB` will set 4 PWM outputs to the commanded values.

My code is pretty focused on controlling RGB(W) LEDs with it, it is only running on an ESP32 and the configuration is done  via `config_RGB.h`. But of course the PWM outputs could be used for anything. The code could be extended for other uC. And the configuration could be done more flexible over MQTT. So this is just a little appetizer for the potential for such a function. 

It works fine for me and I'd like to share it with the community as little thank-you for the great work you all have done with this project.

Feel free to comment, request features, make improvements and maybe this function might be merged into the main project.

ZactuatorRGB.ino
```
/*  
  OpenMQTTGateway Addon  - ESP8266 or Arduino program for home automation 

   Act as a wifi or ethernet gateway between your 433mhz/infrared IR signal and a MQTT broker 
   Send and receiving command by MQTT
 
    Output pin defined to High or Low
  
    Copyright: (c)Florian ROBERT
    
    Contributors:
    - 1technophile
  
    This file is part of OpenMQTTGateway.
    
    OpenMQTTGateway is free software: you can redistribute it and/or modify
    it under the terms of the GNU General Public License as published by
    the Free Software Foundation, either version 3 of the License, or
    (at your option) any later version.

    OpenMQTTGateway is distributed in the hope that it will be useful,
    but WITHOUT ANY WARRANTY; without even the implied warranty of
    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
    GNU General Public License for more details.

    You should have received a copy of the GNU General Public License
    along with this program.  If not, see <http://www.gnu.org/licenses/>.
*/
#include "User_config.h"

#ifdef ZactuatorRGB

#define ACTUATOR_RGB_PWM_FREQUENCY 1200
#define ACTUATOR_RGB_PWM_RESOLUTION 8

#define ACTUATOR_RGB_CHANNEL_RED 0
#define ACTUATOR_RGB_CHANNEL_GREEN 1
#define ACTUATOR_RGB_CHANNEL_BLUE 2
#define ACTUATOR_RGB_CHANNEL_WHITE 3

void setupRGB()
{
  ledcSetup(ACTUATOR_RGB_CHANNEL_RED, ACTUATOR_RGB_PWM_FREQUENCY, ACTUATOR_RGB_PWM_RESOLUTION);
  ledcSetup(ACTUATOR_RGB_CHANNEL_GREEN, ACTUATOR_RGB_PWM_FREQUENCY, ACTUATOR_RGB_PWM_RESOLUTION);
  ledcSetup(ACTUATOR_RGB_CHANNEL_BLUE, ACTUATOR_RGB_PWM_FREQUENCY, ACTUATOR_RGB_PWM_RESOLUTION);
  ledcSetup(ACTUATOR_RGB_CHANNEL_WHITE, ACTUATOR_RGB_PWM_FREQUENCY, ACTUATOR_RGB_PWM_RESOLUTION);
  
  ledcAttachPin(ACTUATOR_RGB_RED_PIN, ACTUATOR_RGB_CHANNEL_RED);
  ledcAttachPin(ACTUATOR_RGB_GREEN_PIN, ACTUATOR_RGB_CHANNEL_GREEN); 
  ledcAttachPin(ACTUATOR_RGB_BLUE_PIN, ACTUATOR_RGB_CHANNEL_BLUE);
  ledcAttachPin(ACTUATOR_RGB_WHITE_PIN, ACTUATOR_RGB_CHANNEL_WHITE);
}

#ifdef jsonReceiving
void MQTTtoRGB(char *topicOri, JsonObject &RGBdata)
{
  if (cmpToMainTopic(topicOri, subjectMQTTtoRGB))
  {
    trc(F("MQTTtoRGB json data analysis"));

    // colors which are not mentioned are automatically set to 0 (off) 
    int redValue = RGBdata["red"] | 0;
    int greenValue = RGBdata["green"] | 0;
    int blueValue = RGBdata["blue"] | 0;
    int whiteValue = RGBdata["white"] | 0;

    trc(F("redValue"));
    trc(redValue);
    trc(F("greenValue"));
    trc(greenValue);
    trc(F("blueValue"));
    trc(blueValue);
    trc(F("whiteValue"));
    trc(whiteValue);  

    ledcWrite(ACTUATOR_RGB_CHANNEL_RED, redValue);
    ledcWrite(ACTUATOR_RGB_CHANNEL_GREEN, greenValue);
    ledcWrite(ACTUATOR_RGB_CHANNEL_BLUE, blueValue);
    ledcWrite(ACTUATOR_RGB_CHANNEL_WHITE, whiteValue);

    // we acknowledge the sending by publishing the value to an acknowledgement topic
    pub(subjectGTWRGBtoMQTT, RGBdata);
  }
}
#endif

#ifdef simpleReceiving
void MQTTtoRGB(char *topicOri, char *datacallback)
{
  if ((cmpToMainTopic(topicOri, subjectMQTTtoRGB)))
  {
    // not implemented yet
  }
}
#endif

#endif
```

config_RGB.h
```
/*  
  OpenMQTTGateway  - ESP8266 or Arduino program for home automation 

   Act as a wifi or ethernet gateway between your 433mhz/infrared IR signal  and a MQTT broker 
   Send and receiving command by MQTT
 
   This files enables to set your parameter for the ON OFF actuator
  
    Copyright: (c)Florian ROBERT
  
    This file is part of OpenMQTTGateway.
    
    OpenMQTTGateway is free software: you can redistribute it and/or modify
    it under the terms of the GNU General Public License as published by
    the Free Software Foundation, either version 3 of the License, or
    (at your option) any later version.

    OpenMQTTGateway is distributed in the hope that it will be useful,
    but WITHOUT ANY WARRANTY; without even the implied warranty of
    MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
    GNU General Public License for more details.

    You should have received a copy of the GNU General Public License
    along with this program.  If not, see <http://www.gnu.org/licenses/>.
*/
#ifndef config_RGB_h
#define config_RGB_h

extern void setupRGB();
extern void MQTTtoRGB(char * topicOri, char * datacallback);
extern void MQTTtoRGB(char * topicOri, JsonObject& RFdata);
/*----------------------------USER PARAMETERS-----------------------------*/
/*-------------DEFINE YOUR MQTT PARAMETERS BELOW----------------*/
#define subjectMQTTtoRGB  "/commands/MQTTtoRGB"
#define subjectGTWRGBtoMQTT  "/RGBtoMQTT"

/*-------------------PIN DEFINITIONS----------------------*/
// default pin, if not set into the MQTT json
#ifndef ACTUATOR_RGB_PIN
    #ifdef ESP8266
            #define ACTUATOR_RGB_PIN 15 //12 for sonoff basic relay
    #elif ESP32
        #define ACTUATOR_RGB_RED_PIN 25 
        #define ACTUATOR_RGB_GREEN_PIN 33
        #define ACTUATOR_RGB_BLUE_PIN 32
        #define ACTUATOR_RGB_WHITE_PIN 13
    #else
        #define ACTUATOR_RGB_PIN 13
    #endif
#endif

#endif
```

(Of course the other programm files have to be adjusted, too. I could provide them, but this would fit better in an individual feature branch than typing it here as a comment.)

1technophile, 2020-03-25:
hello,

Thanks for posting your findings.

Is it not something that could be achieved with fastled library?

Tobiassss, 2020-04-01:
Correct me if I'm wrong, but fastled can only drive "intelligent" LEDs like WS2812 by sending a digital data stream. My function will generate an analog-like PWM Signal for controlling "dumb" LEDs.
This is handy when you just want to control some simple status LEDs, high power LEDs with an external driver (LED drivers are mostly PWM dimmable) or every other PWM controllable device like motor drivers etc.

1technophile, 2020-04-10:
As I'm not a fastled expert I can't affirm that.

What do you think about this example:
https://github.com/marmilicious/FastLED_examples/blob/master/AnalogBasicSetup.ino

closed 2020-06-17 by stale[bot]