1technophile/OpenMQTTGateway #576: Add support for the blinds drive like AM43 Blinds drive

T-REX-XP, 2020-03-27:
Hi, it would be great, when the firmware will be extended by the support of AM43 Blinds drive.
Here is a few implementations of the controlling blinds via BLE:

- https://github.com/binsentsu/am43-ctrl
- https://github.com/TheBazeman/A-OK-AM43-Blinds-Drive

Here is the device: 
https://nl.aliexpress.com/item/4000025499519.html

1technophile, 2020-04-16:
Nice idea!

T-REX-XP, 2020-04-18:
Here is the BLE Arduino [library,](https://github.com/nkolban/ESP32_BLE_Arduino/blob/master/examples/BLE_client/BLE_client.ino) 
I try to implement the control via sketch, the client already connected and find the characteristic, but I can't understand how I should send the packet in the right format. can you help me with it??

Here are some of the packets from the worked project on none.js
Original sources: https://github.com/binsentsu/am43-ctrl/blob/master/src/am43.js

```
const AM43HANDLE = 0x000e;

const HEX_KEY_OPEN_BLINDS = "00ff00009a0d010096";
const HEX_KEY_CLOSE_BLINDS = "00ff00009a0d0164f2";
const HEX_KEY_STOP_BLINDS = "00ff00009a0a01cc5d";

const HEX_KEY_POSITION_BLINDS_PREFIX = "00ff0000";
const HEY_KEY_POSITION_BLIND_FIXED_CRC_CONTENT = "9a0d01";

const HEX_KEY_BATTERY_REQUEST = "00ff00009aa2010138";
const HEY_KEY_LIGHT_REQUEST = "00ff00009aaa010130";
const HEY_KEY_POSITION_REQUEST = "00ff00009aa701013d"; 
```

and here is the open blind function

```
    am43Open() {
        this.writeKey(AM43HANDLE, HEX_KEY_OPEN_BLINDS);
        this.lastaction = 'OPEN';
        this.state = 'OPEN';
    }

    writeKey(handle, key) {
        if (am43.busyDevice != null) {
            this.writeLog('Connection busy for other device, waiting...');
            setTimeout(() => {
                this.writeKey(handle, key)
            }, 1000);
            return;
        }

        this.performWriteKey(handle, key);
    }
```

T-REX-XP, 2020-04-18:
here is my sketch:

```
#include <Arduino.h>
/**
 * A BLE client example that is rich in capabilities.
 * There is a lot new capabilities implemented.
 * author unknown
 * updated by chegewara
 */

#include "BLEDevice.h"
//#include "BLEScan.h"
/*
const serviceUUID = '0000fe5000001000800000805f9b34fb';
const am43CharUUID = '0000fe5100001000800000805f9b34fb';
*/
// The remote service we wish to connect to.
//("4fafc201-1fb5-459e-8fcc-c5c9c331914b");
static BLEUUID serviceUUID("0000fe50-0000-1000-8000-00805f9b34fb");

// The characteristic of the remote service we are interested in.
//("beb5483e-36e1-4688-b7f5-ea07361b26a8")
static BLEUUID charUUID("0000fe51-0000-1000-8000-00805f9b34fb");
//												   prefix(0x9a)-idOfAction(0x0d)(move)-len(data)-data(0x64)-checkSum(0xf2)
static uint8_t closeBlind[9] = {0x00, 0xff, 0x00, 0x00, 0x9a, 0x0d, 0x01, 0x64, 0xf2};

static boolean doConnect = false;
static boolean connected = false;
static boolean doScan = false;
static BLERemoteCharacteristic *pRemoteCharacteristic;
static BLEAdvertisedDevice *myDevice;

static void notifyCallback(
	BLERemoteCharacteristic *pBLERemoteCharacteristic,
	uint8_t *pData,
	size_t length,
	bool isNotify)
{
	Serial.print("Notify callback for characteristic ");
	Serial.print(pBLERemoteCharacteristic->getUUID().toString().c_str());
	Serial.print(" of data length ");
	Serial.println(length);
	Serial.print("data: ");
	Serial.println((char *)pData);
}

class MyClientCallback : public BLEClientCallbacks
{
	void onConnect(BLEClient *pclient)
	{
	}

	void onDisconnect(BLEClient *pclient)
	{
		connected = false;
		Serial.println("onDisconnect");
	}
};

bool connectToServer()
{
	Serial.print("Forming a connection to ");
	Serial.println(myDevice->getAddress().toString().c_str());

	BLEClient *pClient = BLEDevice::createClient();
	Serial.println(" - Created client");

	pClient->setClientCallbacks(new MyClientCallback());

[... 130 lines cut ...]
		const unsigned int length =sizeof(msg);
		for (; (index < length); ++index)
		{
			const char c = msg[index];
			if (c == '*')
				break;
			checksum = checksum ^ c;
		}
		msg[4] = checksum;

		Serial.println("XOR checksum of Hello:");
		Serial.print(checksum,HEX);
		//Serial.print(msg);

		// Set the characteristic's value to be the array of bytes that is actually a string.
		pRemoteCharacteristic->writeValue(closeBlind, true);
	}
	else if (doScan)
	{
		BLEDevice::getScan()->start(0); // this is just eample to start scan after disconnect, most likely there is better way to do it in arduino
	}

	delay(1000); // Delay a second between loops.
} // End of loop
```

1technophile, 2020-04-18:
I started to implement a few month ago characteristic reading [here](https://github.com/1technophile/OpenMQTTGateway/blob/9989bb697c1df01d545555df69217aea9079a5c0/main/ZgatewayBT.ino#L332) for mi flora.
Maybe you could take it a look and see if it helps.

kayaozalp, 2020-10-18:
There is a ESP32 library for AM43 here: https://github.com/buxtronix/am43

Would be great to combine it into OMG