1technophile/OpenMQTTGateway #578: Sonoff RF Bridge won´t connect to unlisted WiFi after reboot

marrobHD, 2020-04-06:
Before submitting a problem please check the troubleshooting section
https://github.com/1technophile/OpenMQTTGateway/wiki/Troubleshooting

**Describe the bug**
After rebooting my Sonoff RF Bridge it won´t connect back to unlisted WiFi network.

**To Reproduce**
Steps to reproduce the behavior:
1. Configure 'WiFi and MQTT'
2. Wait until it connects to WiFi
3. Reboot
4. See error at its AP

**Expected behavior**
It should connect to my unlisted WiFi network.

**Screenshots**
[image]

**Environment (please complete the following information):**
* OpenMQTTGateway version used: v0.9.3

**Additional context**
Add any other context about the problem here.

1technophile, 2020-04-16:
I have done a bypass on the development branch related to wifi connections problem at restart.
Would it be possible to test the last development branch to see if you reproduce the issue?
Or maybe you've already tried with this modification?

marrobHD, 2020-04-17:
After few restarts it works now instantly

closed 2020-04-17 by marrobHD

1technophile, 2020-04-17:
Ok it could have be an issue with the flash memory. OMG format it automaticaly if it doesn't connect at first start.