1technophile/OpenMQTTGateway #579: Incorrect Library Names and different settings required for OMG

siparker, 2020-04-06:
Before submitting a problem please check the troubleshooting section
https://github.com/1technophile/OpenMQTTGateway/wiki/Troubleshooting

**Describe the bug**
When downloading the libraries there are naming problems such as the pub sub library being named PubSubClient_ID89 or the Adafruit Sensor library being named Adafruit Unified Sensor_ID31

When using arduino IDE this causes an error on startup for each of these libraries

https://i.imgur.com/vMQVHpE.png

Also when started there is a warning that the pubsubclient is the incorrect version please use the one from the release pages.

If you want to use a customised pubsubclient library would it not be wiser to name it something else like pubsubclient2 so that this doesn't overwrite the library i am using in other projects with a customised one?

**To Reproduce**
download the libraries from the release page and extract

**Expected behavior**
standard libraries are used or if not then customised libraries that follow naming conventions are provided.

**Screenshots**
https://i.imgur.com/vMQVHpE.png

**Environment (please complete the following information):**
* OpenMQTTGateway version 0.9.3

**Additional context**
Add any other context about the problem here.

1technophile, 2020-04-06:
Thanks for pointing this, the CI [script ](https://github.com/1technophile/OpenMQTTGateway/blob/development/scripts/prepare_deploy.sh) need to be modified to remove the ID of the libraries before packing them.
Regarding PubSubClient the best way for the moment is to modify the MQTT_MAX_PACKET_SIZE manualy.

siparker, 2020-04-07:
Hi, thanks for the quick reply. 

It might also be worth adding a note somewhere in regards to what changes are made in each of the libraries , if any, so that people dont get confused by modified libraries with other projects. Just a thought...

1technophile, 2020-04-27:
Done

closed 2020-04-27 by 1technophile