1technophile/OpenMQTTGateway #582: Support for A7129 receiver

milnovako, 2020-04-14:
Would it be possible to add A7129 RX support?

This module is used for kinetic switches.
The idea is to make a gateway if possible?
Nowhere could I find his datasheet.
[image]
[image]
[image]

1technophile, 2020-04-14:
Hi,

If it is a rare module I'm not sure it is worth to spend time on it.
Did you try with a classic module like the ones proposed on the docs?

milnovako, 2020-04-14:
I tried everything but nothing. This module is increasingly being used. But the manufacturer only supplies a switch to a single receiver. When rf bridge could be made, because the switches are without batteries, they use kinetic energy, I think it would be good to integrate this.
[image]

1technophile, 2020-04-14:
The kinetic switches use a shorter impulse waveform compared to powered ones.
One test you could try is using a basic receiver with a decreased [separation limit ](https://github.com/sui77/rc-switch/blob/7fec3904a2f6f567766d0afb7d1bb4d0abf990c7/RCSwitch.cpp#L101)(2000 or less)

milnovako, 2020-04-19:
I tried everything but not successfully :)

milnovako, 2020-04-19:
Anyway, thanks !

1technophile, 2020-05-11:
Let me think about it, I will go back to you

closed 2020-07-18 by stale[bot]