1technophile/OpenMQTTGateway #584: send value without JSON: -DsimpleReceiving=true does not work ?

tparvais, 2020-04-21:
I compiled OMG for my ESP 32 with platformio for esp32dev-ble

I update platformio.ini with this:

```
[com-esp]
lib_deps =
  ${env.lib_deps}
  ${libraries.wifimanager}
build_flags =
  ${env.build_flags}
  -DMQTT_MAX_PACKET_SIZE=1024
  '-DsimpleReceiving=true'
  '-DsimplePublishing=true'
  '-DTRACE=1'
  ;'-DCORE_DEBUG_LEVEL=4'
upload_port = COM[4]
upload_speed = 512000 
```
It compiles properly and runs well on my ESP32.
But it keeps sending JSON value to MQTT instead of old fashoi per value sytem I need here, despite the two flags 

  '-DsimpleReceiving=true'
  '-DsimplePublishing=true'

I also tried to change User_config.h:

`#define jsonPublishing true //comment if you don't want to use Json  publishing  (one topic for all the parameters)
//example home/OpenMQTTGateway_ESP32_DEVKIT/BTtoMQTT/4XXXXXXXXXX4 {"rssi":-63,"servicedata":"fe0000000000000000000000000000000000000000"}
#define jsonReceiving true //comment if you don't want to use Json  reception analysis

//#define simplePublishing true //comment if you don't want to use simple publishing (one topic for one parameter)
//example 
// home/OpenMQTTGateway_ESP32_DEVKIT/BTtoMQTT/4XXXXXXXXXX4/rssi -63.0
// home/OpenMQTTGateway_ESP32_DEVKIT/BTtoMQTT/4XXXXXXXXXX4/servicedata fe0000000000000000000000000000000000000000
//#define simpleReceiving true //comment if you don't want to use old way reception analysis`

to

`//#define jsonPublishing true //comment if you don't want to use Json  publishing  (one topic for all the parameters)
//example home/OpenMQTTGateway_ESP32_DEVKIT/BTtoMQTT/4XXXXXXXXXX4 {"rssi":-63,"servicedata":"fe0000000000000000000000000000000000000000"}
//#define jsonReceiving true //comment if you don't want to use Json  reception analysis

#define simplePublishing true //comment if you don't want to use simple publishing (one topic for one parameter)
//example 
// home/OpenMQTTGateway_ESP32_DEVKIT/BTtoMQTT/4XXXXXXXXXX4/rssi -63.0
// home/OpenMQTTGateway_ESP32_DEVKIT/BTtoMQTT/4XXXXXXXXXX4/servicedata fe0000000000000000000000000000000000000000
#define simpleReceiving true //comment if you don't want to use old way reception analysis`

No effect neither.

What can I do ?

1technophile, 2020-04-21:
Hi,

>But it keeps sending JSON value to MQTT instead of old fashion per value system I need here, despite the two flags

Could you post a sample of the value published, are you talking about system valies or BLE ones ?

tparvais, 2020-04-22:
Hello

I restart from a clean source folder. I just add the following:

[com-esp]
lib_deps =
  ${env.lib_deps}
  ${libraries.wifimanager}
build_flags =
  ${env.build_flags}
  -DMQTT_MAX_PACKET_SIZE=1024
  '-DTRACE=1'
  '-DBase_Topic="ESP_32_Mezza/"'
  '-Dwifi_ssid="Orange-70a"'
  '-Dwifi_password="ophain1421"'
  '-DGateway_Name="OpenMQTTGateway_BLE"'
  '-DsimplePublishing="true"'
  '-DsimpleReceiving="true"'

upload_port = COM[4]
upload_speed = 512000 

do clean, erase flash and recompile.

I had to configure wifi & MQTT from the webpage

I use "ESP32_Mezza/" for MQTT_topic and OpenMQTTGateway_BLE as gateway name.

On Mqtt , I got these topics which I don't understand why they are not all below "ESP32_Mezza" home topic and why there are 3 different branches instead of 1 ? 

+ still JSON values...

Thank you

[image]

tparvais, 2020-04-22:
After having restarted MQTT , it seems better, strangely:
only wrong part is home / home_presence which is strange for me ? Is it possible to desactivate this ?

[image]

1technophile, 2020-04-25:
Yes, you can deactivate it by commenting [this line](https://github.com/1technophile/OpenMQTTGateway/blob/ac743ed350d45f437bfeabc8697036ac0f6c59d9/main/config_BT.h#L46)

1technophile, 2020-06-07:
I'm closing the issue, if you have other comments feel free to reopen it.

closed 2020-06-07 by 1technophile