1technophile/OpenMQTTGateway #587: MQTT configuration lost

alexmc1510, 2020-04-27:
Before submitting a problem please check the troubleshooting section
https://github.com/1technophile/OpenMQTTGateway/wiki/Troubleshooting

**Describe the bug**
MQTT configuration is lost sometimes (days, weeks...) and I need to switch off my router to be able to access configuration webpage and re-enter MQTT parameters (server, user and pass). Also, MQTT LWT remains online.
As patch solution, would it be possible not to disable configuration webpage once configuration done?

**To Reproduce**
Steps to reproduce the behavior:
Impossible to define. I happens randomly, every some days, maybe weeks.

**Expected behavior**
MQTT configuration should not be lost.

**Screenshots**
.
[image]

**Environment (please complete the following information):**
* OpenMQTTGateway version used (V0.9.3)
* Library version related to the problem you have (MQTT but not sure)

**Additional context**
Using OpenMQTTGateway with Sonoff RF retrofitted with IR capability. Connected to Home Assistant MQTT server.

1technophile, 2020-04-27:
> As patch solution, would it be possible not to disable configuration webpage once configuration done?

The first possibility is to use the [ESPWifiManualconfig](https://github.com/1technophile/OpenMQTTGateway/blob/c88084519fa5830cf6db7d8b39a68bf20da938e5/main/User_config.h#L72) and to set your credentials into [user_config.h](https://github.com/1technophile/OpenMQTTGateway/blob/c88084519fa5830cf6db7d8b39a68bf20da938e5/main/User_config.h)

On the current development branch the reconnection mechanism has been improved and you should have a better stability around weeks.

I take note of your suggestion to see how it can be integrated.

alexmc1510, 2020-04-27:
Thank you so much.
Just last question. It would be possible to modify user_config.h without cables, I mean through OTA?
I programmed it with platformio.

1technophile, 2020-04-27:
> Just last question. It would be possible to modify user_config.h without cables, I mean through OTA?Yes you can do that

alexmc1510, 2020-04-28:
Would you be so kind to send me some info regarding OTA update? I am not able to upload it with platformio.

1technophile, 2020-04-28:
Here is an example:
``` ini
[env:esp32dev-ble]
platform = ${com.esp32_platform}
board = esp32dev
board_build.partitions = min_spiffs.csv
lib_deps =
  ${com-esp.lib_deps}
  ${libraries.ble}
build_flags = 
  ${com-esp.build_flags}
  '-DZgatewayBT="BT"'
  '-DGateway_Name="OpenMQTTGateway_ESP32"'
upload_protocol = espota
upload_port = 192.168.1.11
upload_flags =
  --auth=OTAPASSWORD
  --port=8266
upload_speed = 512000
```

1technophile, 2020-06-22:
With the last version your MQTT configuration shouldn't be lost. I have removed the automatic erase of the flash in case of connect failure first time.

closed 2020-06-22 by 1technophile