1technophile/OpenMQTTGateway #589: OMG floods my brooker with messages

opl-git, 2020-04-28:
Before submitting a problem please check the troubleshooting section
https://github.com/1technophile/OpenMQTTGateway/wiki/Troubleshooting

**Describe the bug**
The OMG floods my MQTT Broker so much, it actually goes down.
Here an output of the Serial Connection:
_MQTT connection...
Connected to broker
Subscription OK to the subjects_
This mesage comes alsmost 5x every second
I really can't the problem. This is my second gateway with OMG, the first runs without any problems. 
All I did was change the Base_Topic... (first OMG uses "og/", the second "wg/" 

**To Reproduce**
Steps to reproduce the behavior:
I flashed the OMG again for several times

**Expected behavior**
A clear and concise description of what you expected to happen.
A normal Login to the brooker

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
* OpenMQTTGateway version 0.93
Using the Arduino way to flash OMG on my ESP32 Dev Kit

**Additional context**
Add any other context about the problem here.

1technophile, 2020-04-28:
This can be due to the fact that you are using the same gateway name on both boards, did you changed the second one?

opl-git, 2020-04-28:
Thank you so much! Your awesome!
That was the problem!
I thought I only had to change the basic topic.

closed 2020-04-28 by opl-git

1technophile, 2020-04-28:
You're welcome, if you would like to add a description of the issue/solution in the [troubleshooting ](https://github.com/1technophile/OpenMQTTGateway/blob/development/docs/upload/troubleshoot.md) page don't hesitate