1technophile/OpenMQTTGateway #590: ESP32 GPIO

pmknowles, 2020-04-28:
**Is your feature request related to a problem? Please describe.**
It isn't clear which bin to use as there doesn't appear to be a description as to what each bin does. I flashed the ble version which saw the beacons I have, I then wanted to turn a GPIO on. It's not clear what is meant by a 'pin'. Is it the 'D' number or the GPIO number? I was seeing the MQTT message on the broker but nothing was happening at the ESP32 end.
I thought perhaps I needed to use the 'all; bin.
This just stopped with this message in ESPlorer

> Bosch BME280 Initialized - Result of .begin(): 0x0
> No TSL2561 detected
> TSL2561 Initialized. Printing detials now.
> ------------------------------------
> Sensor:       TSL2561
> Driver Ver:   1
> Unique ID:    12345
> Max Value:    17000.00 lux
> Min Value:    1.00 lux
> Resolution:   1.00 lux
> ------------------------------------
> 
> IR_EMITTER_PIN 
> 14
> IR_RECEIVER_PIN 
> 26
> ZgatewayIR setup done 
> ZgatewayLORA setup failed!
It could be that I've missed something obvious.

**Describe the solution you'd like**
A better clarification of what the bins are for and clearer instructions about the terminology

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

jmw6773, 2020-05-01:
Each pre-compiled bin file enables a subset of the modules. The -all bin's are mainly used to test compiling to verify that all of the modules compile correctly, but can be used if you have a LORA module connected. (As far as I know, only the LORA and the 2G modules block the OpenMQTTGateway if they aren't connected, but there may be others.)

If you're able, it's probably better to clone the git and compile the firmware with only the modules you need. This also allows you to change which pins are used and to make other setting changes.

As far as I know, there isn't a way to control multiple GPIO pins. However if you just want to switch one pin ON/OFF, you could probably use the ZactuatorONOFF module.

As for the pin numbering, it really depends on which board you are compiling for. Many Arduino boards use A0-Ax for analog and D0 - Dx for digital to make it easier. But with all of the different boards that OpenMQTTGateway can be run on, it's probably better to just use the GPIO pin number for the chip you're using.
So on a WeMos D1 Mini, instead of D4 you would put 2. (You can Google 'Wemos D1 Mini pinout' for additional pins.)

Docs are never fun to write, but they are getting better. @1technophile and the other contributors have done a tremendous job on this project. But I'm sure they would welcome any help you can give to make the docs better.

I hope this helps a little.

pmknowles, 2020-05-01:
I feel your pain about doc writing. Knowing that the pin is the GPIO rather than the D number helps as does the explanation of the bins.
I was hoping for a Tasmota style ability to allocate pins as input and output (not necessarily a graphical UI - a console like ability would be useful).

Thank you
Phil

1technophile, 2020-05-01:
We could replace the PIN mention by GPIO on the program logs if it helps and add some help on the docs related to bin files.

pmknowles, 2020-05-01:
Thank you. From experience something that I think makes perfect sense when I write often is missing something for the less experienced reader.

Regards

Phil

closed 2020-06-14 by 1technophile