1technophile/OpenMQTTGateway #591: v0.9.4beta Weather Station library missing in all-libraries

Alomon31, 2020-04-29:
Hi,

Title is explicit, it seeme like WeatherStationDataRx_ID6800 folder is missing in esp32dev-all-libraries.zip (at least, I didn't check for the other hardware).

Thanks for support.

1technophile, 2020-05-01:
I confirm, for whom interested don't hesitate to propose a pull request with a modified platformio.ini.
The goal is to add the library and the gateway to esp32dev-all and novemcuv2-all env.

Alomon31, 2020-05-04:
Issue closed as it was corrected in last commit.
However, the contributor has added `weatherstationrx = WeatherStationDataRx@0.3.3` in the libraries although `rfWeatherStation = WeatherStationDataRx@0.3.1` was already declared. Shouldn't it be merged?
I've seen that Pilight is missing as well in esp32-all, I will try and add it.

closed 2020-05-04 by Alomon31

jmw6773, 2020-05-04:
@Alomon31, you're right. Good catch, I didn't see the previous declaration for the library. Do you know if version 0.3.3 of the library is okay to use, or should 0.3.1 be kept?

I can make a pull request to remove one of them and update the build definitions as needed.

Alomon31, 2020-05-04:
Sorry but I don't know the libraries, just saw that when having a first look at the file (I'm quite new here).

jmw6773, 2020-05-04:
I'm new here too. Jumped in last week to try to help out a little.

I didn't see many changes between the versions when looking at the libraries github page, but I'll keep the original version as the initial push had it and let the sensors' contributor update it if needed.

I'll made a Pull Request tomorrow morning (JST) to fix the platformio.ini file definitions.