1technophile/OpenMQTTGateway #596: Separate MQTT topic for each device

besynnerlig, 2020-05-01:
I'm on version v0.9.4beta running on a Sonoff RF Bridge + Pilight

I currently only have 10 door/window sensors GS-WDS07

My problem is that all sensor are published on a single MQTT topic. Example with 2 different sensor devices:
```
rf433gwOpenMQTTGateway/PilighttoMQTT {"message":{"unit":20903,"battery":1,"state":"closed"},"protocol":"iwds07","length":"","repeats":2,"status":2}
rf433gwOpenMQTTGateway/PilighttoMQTT {"message":{"unit":5639,"battery":1,"state":"opened"},"protocol":"iwds07","length":"","repeats":2,"status":2}
```

I don't mind that the payload is JSON. That's perfectly fine but I'd really like to have a separate MQTT Topic for each device. E.g. something like 
```
rf433gwOpenMQTTGateway/PilighttoMQTT/20903 {"message":{"unit":20903,"battery":1,"state":"closed"},"protocol":"iwds07","length":"","repeats":2,"status":2}
rf433gwOpenMQTTGateway/PilighttoMQTT/5639 {"message":{"unit":5639,"battery":1,"state":"opened"},"protocol":"iwds07","length":"","repeats":2,"status":2}
```

As seen in this [documentaion](https://docs.openmqttgateway.com/integrate/openhab2.html#mqtt-2-manual-setup-openhab2-4) where I assume that _C47C9999D1B8_ is a specific device. 

Am I missing something here?

(Ohhh... btw the ugly **rf433gwOpenMQTTGateway** topic should actually be **rf433gw/OpenMQTTGateway** but even though I'm entering that trailing slash after _rf433gw_ while configuring OMG in the web GUI for first time doesn't really make any difference. )

1technophile, 2020-05-01:
This feature is called [ValueAsasubject ](https://github.com/1technophile/OpenMQTTGateway/blob/470790b42e9ea1a8774239e7a11745b8fc283da7/main/User_config.h#L272).
It will work with all the gateways returning a "value" key, ZgatewayRF for example
Unfortunately as Pilight return a "unit" key it is not taken into account on the current code.

besynnerlig, 2020-05-01:
Thanks @1technophile, I really appreciate your kind help.

I think though that this is a dealbreaker for me so I'll probably try to revert the Pilight hardware hack and restore the Sonoff RF Bridge to original. Would you happen to know if my existing door/window sensors GS-WDS07 will work using any other environment choise that supports the valueAsASubject feature? If so please what environment should I choose? Thanks!!!

1technophile, 2020-05-01:
GS-WDS07 is supported by ZgatewayRF, you can upload this gateway to the Sonoff RFBridge and you should be able to have the value concatenated to the subject.

besynnerlig, 2020-05-01:
Thanks @1technophile :smile: 

I'm actually a noob but I assume that means that I just need to

`#define ZgatewaySRFB`

In user_config and keep the rest to default.

I'll give it a try :sunglasses:

1technophile, 2020-05-01:
You re welcome. As you have done the direct hack you can use ZgatewayRF instead of SRFB. And yes uncomment it in user_config.h set your different parameters and upload it.

besynnerlig, 2020-05-01:
@1technophile Thanks a lot Florian for your kind help. 

I got a little confused by your answer though. I was thinking to revert my direct hack. Do you mean that I can keep it and just switch from SRFB to ZgatewayRF???

1technophile, 2020-05-01:
Let me give me you more explanation on it.
The direct hack enables to bypass the dedicated chip used for decoding and encoding the signal of the ESP, while this chip enable to relieve the ESP load it is limited in number of protocols supported.
By doing the hardware hack you enable the ESP to handle the decoding and encoding. By the way it is easier to load all the Arduino RF libraries available on github.
The best libraries available are in my point of view:
-RCswitch (ZgatewayRF)
-ESPilight (ZgatewayPilight)
-NewremoteSwitch (ZgatewayRF2)

ZgatewaySRFB will not work with the direct hack as it will try to talk to the dedicated chip and you bypassed it.

closed 2020-05-02 by besynnerlig