1technophile/OpenMQTTGateway #598: OneWiretoMQTT support HA discovery

phdelodder, 2020-05-02:
**Is your feature request related to a problem? Please describe.**
OneWiretoMQTT doesn't use HA discovery

**Describe the solution you'd like**
OneWiretoMQTT should support HA discovery

**Describe alternatives you've considered**
Manual Configuration

**Additional context**

Version 0.9.4-Beta

jmw6773, 2020-05-27:
By OneWire, do you mean the DS1820 temperature sensor?

jmw6773, 2020-05-27:
@1technophile, if this functionality is something you'd like added, I think the DS1820 sensor could be made HA Discovery compatible by making each DS1820 sensor write to a sub-MQTT topic using it's unique identifier.

In the device setup, for each sensor detected, it would publish the HA Discovery information. (Like how BT gateway does it, but only once during the DS1820 setup.)

ZsensorDS1820.ino: L172
`pub(OW_TOPIC + String("/") + ds1820_addr[i], DS1820data);`

The createDiscoveryFromList would need to be re-written, or a new function created, to allow other topics.
https://github.com/1technophile/OpenMQTTGateway/blob/0afcb61717f5d8bd98afb951e1a108bfe8327af1/main/ZmqttDiscovery.ino#L53

1technophile, 2020-06-02:
Hello,

Yes we could add it, PR welcome!

jmw6773, 2020-06-05:
@phdelodder , could you pull the dev version to see if it resolves the HADiscovery for OneWire issue? If you can't build it, I can compile it for you if you send me which configuration you would like to use.

phdelodder, 2020-06-08:
Did some testing but the sensor can't be found any more, tried other firmware and it was discovered. Don't know what changed but it's unusable now.

jmw6773, 2020-06-08:
Do you see any sensors under the MQTT integration for the device?

HA --> Configuration --> Integrations --> MQTT
   Under MQTT, click the `# devices` link
   Find and click on your OMG device

phdelodder, 2020-06-08:
No nothing, the device isn't even found physically

jmw6773, 2020-06-08:
Could you let me know what hardware you are using (ESP8266, ESP32, etc), which pin you have the DS18b20 attached to, and the config you have for your platform.ini? I'll try to duplicate your setup and see if I can replicate the issue.

phdelodder, 2020-06-08:
I was using a ESP32, on pin 2. 

```
[env:esp32dev-ble]
platform = ${com.esp32_platform}
board = esp32dev
board_build.partitions = min_spiffs.csv
lib_deps =
  ${com-esp.lib_deps}
  ${libraries.ble}
build_flags =
  ${com-esp.build_flags}
  '-DZgatewayBT="BT"'
  '-DLOW_POWER_LED=2'
  '-DLOW_POWER_LED_OFF=1'
  '-DGateway_Name="OMG_Garage"'
```

jmw6773, 2020-06-08:
Using that platform.ini, I was able to get the OMG to show in the MQTT devices. I'm not sure why you don't see the OMG in HA.
However, you don't have the DS1820 sensor enabled in that platform.ini config. 

You also might want to be cautious using GPIO2 with BLE as if you have LOW_POWER_MODE enabled, it could pull the LOW_POWER_LED (GPIO2 in the default build) down to LOW. 
I've also found that I can't flash my ESP32 (MH-ET LIVE MiniKit board) when the DS18b20's are plugged into GPIO2.

Try building it with:

```
[env:esp32dev-ble]
platform = ${com.esp32_platform}
board = esp32dev
board_build.partitions = min_spiffs.csv
lib_deps =
  ${com-esp.lib_deps}
  ${libraries.ble}
  ${libraries.onewire}
  ${libraries.dallastemperature}
build_flags =
  ${com-esp.build_flags}
  '-DZgatewayBT="BT"'
  '-DZsensorDS1820="DS1820"'
  '-DLOW_POWER_LED=2'
  '-DLOW_POWER_LED_OFF=1'
  '-DGateway_Name="OMG_Garage"'
```

phdelodder, 2020-06-08:
I enable the sensor using the user config. Well GPIO2 is defined per default, so if it indeed a caution-point perhaps assigning a different one for ESP32. 

I have now ordered 2 [new](https://www.aliexpress.com/item/4000296897646.html?spm=a2g0s.9042311.0.0.c2ab4c4diHHNGu) ESP32, the chipset is ESP32-WROOM-32U to add an antenna. 

When does arrive I can further test this. My test ESP32 isn't giving a com-port after flashing my latest OMG build.

closed 2020-08-15 by stale[bot]