1technophile/OpenMQTTGateway #599: HA discovery should group the entities created by the OMG device

phdelodder, 2020-05-02:
**Is your feature request related to a problem? Please describe.**
Using HA discovery all the entities are scared, not group by the device that creates the entries. 

**Describe the solution you'd like**
Group the entries per device

**Describe alternatives you've considered**
Search for the entry one by one.

version 0.9.4-beta

jmw6773, 2020-05-02:
Might be doable by concatenating OMG's Gateway_Short_Name (or Gateway_Name) with the sensors' name in the createDiscovery method.

https://github.com/1technophile/OpenMQTTGateway/blob/470790b42e9ea1a8774239e7a11745b8fc283da7/main/ZmqttDiscovery.ino#L85

The Gateway Name is configurable in User_config.h or through the platformio.ini file.

Making this change would make everything show in the form 'sensor.OMG_adc' instead of 'sensor.adc'

@1technophile - If this is useful, I can submit a Pull Request.

1technophile, 2020-05-02:
The context we have to care about is when using multiple OMG gateways, we must avoid duplicating entities into HA and I think using the Gateway_Short_Name will do the job, contrary to the Gateway_Name that needs to be changed when using several boards.

Yes you're PR is welcome!

phdelodder, 2020-05-02:
Might I add that for example, I have several mi flora using one gateway i can'( distint one from another there for I would suggest: sensor.'Gateway_Short_Name'_'MiFlora_MAC'_name

1technophile, 2020-05-02:
> i can't distint one from another

Even with the mac adress?

phdelodder, 2020-05-03:
With the mac adress of the gateway or the BLE sensor? If the it's the BLE sensor then it would be sufficient.

1technophile, 2020-06-07:
I propose to continue the discussion [here ](https://community.openmqttgateway.com/t/notes-on-issues-with-ha-discovery/953)if you don't mind

closed 2020-06-07 by 1technophile