1technophile/OpenMQTTGateway #601: HA discovery config for gatewaybt sensor isn't working

phdelodder, 2020-05-03:
Before submitting a problem please check the troubleshooting section
https://docs.openmqttgateway.com/upload/troubleshoot.html

**Describe the bug**

Every X seconds there is a publish of these information about the OMG: 
```
home/OMG_Garage/SYStoMQTT {"uptime":120,"version":"v0.9.4beta","freeMem":53216,"rssi":-24,"SSID":"DelodderVanTorre","ip":"192.168.0.X","mac":"30:AE:A4:XX:XX:XX","wifiPrt":0,"low_power_mode":0,"modules":"DS1820BTHADiscovery"}
```
The config for HA discovery specifies the value of the sensor as ID, as there are is no ID in the publish, the sensor is useless. 
```
home/OMG_Garage/SYStoMQTT {"uptime":120,"version":"v0.9.4beta","freeMem":53216,"rssi":-24,"SSID":"DelodderVanTorre","ip":"192.168.0.109","mac":"30:AE:A4:73:6D:34","wifiPrt":0,"low_power_mode":0,"modules":"DS1820BTHADiscovery"}
```

**To Reproduce**
Steps to reproduce the behavior:
1. Use HA discovery

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
* OpenMQTTGateway version used  0.9.4-beta
* Library version related to the problem you have (if you have troubles with RF provide the version of RCSwitch library) : HADiscovery

jmw6773, 2020-05-03:
It looks like the status topic for the gatewayBT should be:
home/OMG_Garage/BTtoMQTT/+

Try publishing the following to your MQTT server and see if the sensor starts working in Home Assistant. (The sensor value will keep updating with the last scanned BT device's MAC.)
I tried to match your settings (ESP MAC and OMG Name), but please double check.

**Topic:**
`homeassistant/sensor/30AEA4736D34gatewayBT/config`

**Packet:**
`{"stat_t":"home/OMG_Garage/BTtoMQTT/+","name":"OMG_gatewayBT","uniq_id":"30AEA4736D34gatewayBT","val_tpl":"{{ value_json.id }}","device":{"name":"OMG_Garage","manufacturer":"OMG_community","sw_version":"v0.9.4beta","identifiers":["30AEA4736D34"]}}`

phdelodder, 2020-05-03:
After testing it seems it's working like you showed it, now I have my doubts about the sensor. For example when these messages a published, the last entry is without ID so the sensor is empty: 
```
home/home_presence/OMG_Garage {"id":"80:ea:ca:89:02:74","name":"Flower care","rssi":-92,"distance":27.75274}
home/OMG_Garage/BTtoMQTT/80EACA890274 {"id":"80:ea:ca:89:02:74","name":"Flower care","rssi":-92,"distance":27.75274,"servicedata":"7120980004740289caea800d091002a002","servicedatauuid":"0000fe95-0000-1000-8000-00805f9b34fb"}
home/OMG_Garage/BTtoMQTT/80EACA890274 {"fer":672}
```

I propose to do the following for the config: 
[image]`
**Payload:**
```
{
	"stat_t": "home/OMG_Garage/SYStoMQTT",
	"avty_t": "home/OMG_Garage/LWT",
	"pl_avail": "online",
	"pl_not_avail": "offline",
	"name": "OMG_gatewayBT",
	"uniq_id": "30AEA4736D34gatewayBT",
	"val_tpl": "{{ value_json['rssi'] }}",
	"device": {
		"name": "OMG_Garage",
		"manufacturer": "OMG_community",
		"sw_version": "v0.9.4beta",
		"identifiers": [
			"30AEA4736D34"
		]
	}
``

jmw6773, 2020-05-03:
The RSSI value in the SYStoMQTT topic is the RSSI value of the WiFi connection for the ESP. It doesn't have any relation to the Bluetooth. 

It would be better to try to implement putting the value for the sensor into the main payload. I'm not sure how feasible that is to implement though.

phdelodder, 2020-05-03:
For me it was only an example, I want to show that all the info is put in toSystoMQTT has more info and can be used by HA. 

Only the uptime has no relevance on the ESP32.

xHirscHx, 2020-05-23:
I cant get ha discover to work,all good in serial and shows up in MQTT-Explorer Ha sees the gateway but not any devices.  what the manual config file for a bt device look like for hassio yaml  can't find an example

1technophile, 2020-05-23:
> I cant get ha discover to work,all good in serial and shows up in MQTT-Explorer Ha sees the gateway but not any devices

Do you see on MQTT explorer the device discovery messages into` homeassistant/sensor` topic?

> what the manual config file for a bt device look like for hassio yaml can't find an example

Did you checked the [docs](https://docs.openmqttgateway.com/integrate/home_assistant.html#manual-integration-examples)?

xHirscHx, 2020-05-23:
In mqtt explorer homeassistant/sensor  the only sensor is for the gatewayBT .
Looking for a tracking beacon sensor BT room detection with watch or phone

joselito11, 2020-05-31:
My gateway sensor is alway unknown. If I publish `homeassistant/sensor/2462ABF34504gatewayBT/config` sensor disapear from HA.

jmw6773, 2020-06-01:
New sensors aren't created for every Bluetooth device detected as it would create a sensor for every bluetooth device that has ever been in range of the OpenMQTTGateway.

Only the devices [described in the docs](https://docs.openmqttgateway.com/use/ble.html#receiving-signals-mi-flora-mi-jia-device-lywds02-cleargrass-or-mi-scale) have sensors created using the discovery process.

Other Bluetooth devices have an MQTT topic created, but aren't registered using HADiscovery. You can [manually add the topic](https://docs.openmqttgateway.com/integrate/home_assistant.html#manual-integration-examples) for your BLE beacon to HomeAssistant as a sensor depending on which information you are looking for. (Probably RSSI)

jmw6773, 2020-06-01:
@joselito11, most HADiscovery configs are only sent when the OpenMQTTDGateway is powered on or restarted. Some sensors, [described here](https://docs.openmqttgateway.com/use/ble.html#receiving-signals-mi-flora-mi-jia-device-lywds02-cleargrass-or-mi-scale), have their configs sent the first time they are detected.

If you overwrite the /config topic with a blank data, the sensor will be removed from Home Assistant and won't re-appear until you restart the Gateway. (Restarting HA will not make it re-appear.)

Sensor definition for gatewayBT:
https://github.com/1technophile/OpenMQTTGateway/blob/0afcb61717f5d8bd98afb951e1a108bfe8327af1/main/ZmqttDiscovery.ino#L462-L472

joselito11, 2020-06-01:
Thank you for explanation, have them added manualy, working fine! Only gatewayBT sensor do not report, but I dont need him.
Great work!

closed 2020-08-08 by stale[bot]