1technophile/OpenMQTTGateway #605: ESP32 EVB from olimex : add default pinout for IR tx and RX

ouinouin, 2020-05-05:
**Is your feature request related to a problem? Please describe.**
the ESP32-EVB from olimex already has integrated aTX and RX IR LED,  TX uses GPIO12 and RX use GPIO39 , maybe it could be added to the platformio configuration and could also be build and released ? 

regards.

1technophile, 2020-05-05:
Good suggestion !  Do you want to propose a pull request ?

ouinouin, 2020-05-06:
Hi, 
i boubt being technically fit for that unfortunately. if you miss time for that i ll have a look, i suppose looking at the adaptations made for a specific platform like m5stick or m5stack would be a good step to know how to do (plenty of #define I suppose).
I ll also propose a one line linux command to flash your firmware to add into the documentation (which is actually quite clear , this is rare enough to mention it :-D ).

1technophile, 2020-05-11:
> i boubt being technically fit for that unfortunately. if you miss time for that i ll have a look, i suppose looking at the adaptations made for a specific platform like m5stick or m5stack would be a good step to know how to do (plenty of #define I suppose).

You may propose a first draft here and I will comment it if you want

> I ll also propose a one line linux command to flash your firmware to add into the documentation (which is actually quite clear , this is rare enough to mention it :-D ).

Don't hesitate to share

> somthing I think since a while would be to use OMG to create an open codes database with the help of the users, but the codes would be sent automatically, not manually, an dthe user could list all the codes through a web iface for example and mark the model of the remote ctrl...

It's a nice idea, we may link it also with some model database

> the IR receiver embedded on this board has a 36khz filter, it might limit the applications unfortunately :

I think so, most of the device run on 38Khz I think