1technophile/OpenMQTTGateway #607: MQTT client ID

valdiks, 2020-05-06:
If there are two or more OMGs with same config but different mqtt topic.
It seems that OMG MQTT client ID is generated from build_flags = ...  -DGateway_Name

MQTT server: Eclipse Mosquitto
Log information: Client OpenMQTTGateway_ESP32_BLE already connected, closing old connection.
This log entry repeats every single second for each OMG.
So log file is flooded with this entry.

OMG version: 0.9.4beta

MQTT client ID should be generated from Gateway_Name which is defined in User_config or from ESP32 (or other board) MAC address/Hardware ID.

1technophile, 2020-05-06:
It is defined in user_config.h but you can overload it with platformio.ini
Depending on which IDE you are using.

valdiks, 2020-05-06:
I know but it doesn't work for me. It only change mqtt topic but not mqtt client id.
I use platform.io.

1technophile, 2020-05-06:
Please post your .ini file and a serial monitor extract of the OMG start in LOG_LEVEL_VERBOSE mode (line 291 of user_config.h)

valdiks, 2020-05-07:
Not sure what you need but please look here [https://gist.github.com/valdiks/83605fbed8b3bcc1f489c5d6ca743406](url)

I had to change OpenMQTTGateway_ESP32_BLE to Home_OMG in platform.ini because mqtt server reports about 
> Client OpenMQTTGateway_ESP32_BLE already connected, closing old connection.

1technophile, 2020-05-07:
> I had to change OpenMQTTGateway_ESP32_BLE to Home_OMG in platform.ini because mqtt server reports about

Ok, and if working now what are you expecting  ?

valdiks, 2020-05-07:
I think that for every board the mqtt client id should be unique and generated automatically.
Or should be set from the User_config.h Gateway_Name.

Or you could add information to the documentation that if you need more than one board with the same firmware then -DGateway_Name should be changed.

jmw6773, 2020-05-13:
@valdiks, it would be easy to make the MQTT client id unique per board by concatenating the MAC address to the gateway_name for ESP devices as below. However, each OpenMQTTGateway should have a unique Gateway_Name defined. The Gateway_Name is also used for the MQTT topic, HA Discovery, and other functions.

The default Gateway_Name can be changed at compile time by defining it as you have shown and also in the [WiFi and MQTT configuration portal](https://docs.openmqttgateway.com/upload/portal.html) during initial setup. Does changing the Gateway Name here not change the clientID for MQTT? It looks like it should.

```
#if defined(ESP8266) || defined(ESP32)
    String clientId = String(gateway_name) + "-" + (ESP.getChipId() & 0x1fff);
#else
    String clientId = gateway_name;
#endif
```

1technophile, 2023-02-04:
Could be done with this macro
https://github.com/1technophile/OpenMQTTGateway/blob/2ab603e1bebf35552896307d9b1bae817387fd55/main/User_config.h#L55

closed 2023-02-04 by 1technophile