1technophile/OpenMQTTGateway #609: Enable infrared symfony protocol

ouinouin, 2020-05-09:
Hi,
I use this protocol wich looks mainly related to fans /ceiling fans, but there is no support in OMG.
I added the support by adding : 
in ZgatewayIR.ino : 

#ifdef IR_SYMPHONY
  if (strstr(protocol_name, "SYMPHONY") != NULL)
  {
    if (valueBITS == 0)
      valueBITS = kSymphonyBits;
    if (valueRPT == repeatIRwNumber)
      valueRPT = std::max(valueRPT, kSymphonyDefaultRepeat);
    irsend.sendSymphony(data, valueBITS, valueRPT);
    return true;
  }
#endif

and in config_IR.h

#define IR_SYMPHONY

I guess more protocols from the IRremoteESP8266 may be missing, is there a simple way for the lib and the .ino supported protocols to be in sync ? 
ps : still experimenting on ESP32 EVB.... for the instance im unable to transmit with integrated LED.

1technophile, 2020-06-07:
> I guess more protocols from the IRremoteESP8266 may be missing, is there a simple way for the lib and the .ino supported protocols to be in sync ?

I didn't found it for the moment but I agree it is difficult to maintain in sync.

1technophile, 2020-06-19:
done

closed 2020-06-19 by 1technophile