1technophile/OpenMQTTGateway #611: IR led reversed output on M5stick-C

ouinouin, 2020-05-10:
hi, 
While compiling with the M5stick-C , the infrared emitter  is always switched on by default.
I digged a bit and found the led has to be driven reverse (they dont use a transistor to drive this small IR led)
fortunately this can be reversed by using : IRsend irsend(IR_EMITTER_PIN,1);
so i changed this line 42 of ZgatewayIR.ino and then managed to transmit some signals well decoded by IRrecvdumpV2 sketch running on another platform.

whats the best way to do to make it configurable in your code ? should it be and option for each platform defined in platformio.ini with a default value ?

1technophile, 2020-06-07:
Thanks for pointing that.

> whats the best way to do to make it configurable in your code ? should it be and option for each platform defined in platformio.ini with a default value ?

We may add it as a config parameter at build time into config_IR.h, and set it into the corresponding platformio.ini env.

Are you interested in proposing a PR?

1technophile, 2020-08-18:
@ouinouin could you test the linked PR please?