1technophile/OpenMQTTGateway #616: Increase Bluetooth range, what are the best modules to use

csigabit, 2020-05-15:
I'd like to use the xiaomi/aqara bt temp sensors with omg. I've tested in 3 residences using HM10 or ESP32 built-in bluetooth and the optimal maximum distance was 5-7 meters, 1 wall between sensor and omg. While my phoneBT or the new bt mesh xiaomi gateway could get sensor data over 10m and 2 walls between the sensor and phone/gw.

Maybe there are modules out there, that could do better than HM10 CC2541 which I have tested with different versions of firmware. What other modules can be used? Flash files and instructions too if necessary and available.

1technophile, 2020-05-16:
That's a good question, I would first recommend to try an ESP32 with an external antenna. But I didn't get the chance to test one yet.
The only ESP32 board with antenna that I have seen have a camera:
https://www.banggood.com/LILYGO-TTGO-T-Journal-ESP32-Camera-Module-Development-Board-OV2640-SMA-WiFi-3dbi-Antenna-Fisheye-Lens-p-1418432.html?rmmds=search&cur_warehouse=CN

1technophile, 2020-06-07:
If you want to discuss more about this I suggest this [thread ](https://community.openmqttgateway.com/t/esp32-ble-rang-extension/938)on the community

closed 2020-06-07 by 1technophile