1technophile/OpenMQTTGateway #620: Core dump after 2 days of error free operation.

danobot, 2020-06-01:
Before submitting a problem please check the troubleshooting section
https://docs.openmqttgateway.com/upload/troubleshoot.html

**Describe the bug**
I configured this on a new ESP32 board using `v0.9.4beta`. It was running fine for 2 days. Last night it stopped working altogether. WHen i plugged in to check serial logs the device had reset itself and lost the wifi configuration.

Core dump printed
```
N: Wifi connected
{"mqtt_server":"10.1.1.2","mqtt_port":"1883","mqtt_user":"omg","mqtt_pass":"","mqtt_topic":"home/","gateway_name":"gw"}
N: RF_EMITTER_PIN: 12                                                                                                           N: RF_RECEIVER_PIN: 27                                                                                                                               N: Setup OpenMQTTGateway end                                                                                                                                                           N: Erasing ESP Config, restarting                                                                                                                                                                                                                                      W: ESP32: Forcing to wifi 0
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Core 1 register dump:
PC      : 0x40080f81  PS      : 0x00060034  A0      : 0x8008115d  A1      : 0x3ffbe750  
A2      : 0x00000001  A3      : 0x0000000b  A4      : 0x00000000  A5      : 0x3ffbe760  
A6      : 0x000000ba  A7      : 0x00000000  A8      : 0x800814aa  A9      : 0x3f4053c2  
A10     : 0x00000000  A11     : 0x000000ad  A12     : 0x00000000  A13     : 0x3f4053c2  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x0000001e  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
Core 1 was running in ISR context:
EPC1    : 0x40086ce3  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40080f81

Backtrace: 0x40080f81:0x3ffbe750 0x4008115a:0x3ffbe780 0x4008123d:0x3ffbe7b0 0x40084a31:0x3ffbe7d0 0x40086ce0:0x3ffba190 0x40082e4b:0x3ffba1b0 0x40088e3d:0x3ffba1d0

Rebooting...
```

**To Reproduce**
Not sure how to reproduce it because it was working fine for 2 days, then stopped working and now I am unable to even set it up on ESP32 as it keeps 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
* OpenMQTTGateway version used (V0.9.3, 0.8, development)
* Library version related to the problem you have (if you have troubles with RF provide the version of RCSwitch library)

**Additional context**

```
*WM: [3] -> www.msftconnecttest.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [2] Portal Timeout In 61 seconds
*WM: [2] <- HTTP WiFi save  
*WM: [3] Method: GET
*WM: [2] Parameters 
*WM: [2] -------------------- 
*WM: [2] server: 
*WM: [2] port: 
*WM: [2] user: 
*WM: [2] pass: 
*WM: [2] name: 
*WM: [2] topic: 
*WM: [2] -------------------- 
*WM: [3] Sent wifi save page 
*WM: [2] process connect 
*WM: [2] No ssid, skipping wifi 
*WM: [3] WIFI station disconnect 
*WM: [3] WiFi station enable 
*WM: [2] Disabling STA 
*WM: [2] <- HTTP Root 
*WM: [3] -> 192.168.4.1 
*WM: [3] lastconxresult: Unknown
*WM: [2] WiFi Scan ASYNC started 
*WM: [2] <- HTTP Wifi 
*WM: [0] [ERROR] scan waiting 
*WM: [0] . 
[... 112 lines cut ...]
*WM: [1] STA static IP:
*WM: [2] setSTAConfig static ip not set 
*WM: [3] WIFI station disconnect 
*WM: [1] No saved credentials, skipping wifi 
*WM: [2] Connection result: WL_NO_SSID_AVAIL
*WM: [3] lastconxresult: WL_NO_SSID_AVAIL
*WM: [1] AutoConnect: FAILED 
*WM: [2] AccessPoint set password is VALID 
*WM: [1] your_password 
*WM: [3] WIFI station disconnect 
*WM: [3] WiFi station enable 
*WM: [2] Disabling STA 
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  OpenMQTTGateway_ESP32_RF
*WM: [1] AP IP address: 192.168.4.1
*WM: [3] setupConfigPortal 
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip:  192.168.4.1
*WM: [2] HTTP server started 
*WM: [2] WiFi Scan ASYNC started 
*WM: [2] Config Portal Running, blocking, waiting for clients... 
*WM: [2] WiFi Scan ASYNC completed in 2310 ms
*WM: [2] WiFi Scan ASYNC found: 8

```

1technophile, 2020-06-01:
> Not sure how to reproduce it because it was working fine for 2 days, then stopped working and now I am unable to even set it up on ESP32 as it keeps coredumping due to Meditation Error

You may try to erase the flash before a new upload to see if it is still core dumping.

Could you post your configuration used (env definition from platformio.ini) and user_config.h if you modified it?

Could you indicate your router model and eventually try with your smartphone as an internet AP?

danobot, 2020-06-02:
* erased flash, cleaned build and reuploaded - same

**platformio.ini:** No customisation other than uncommenting the `esp32dev-rf` profile
```
default_envs = esp32dev-rf
```

**User_config.h**
Updated MQTT server details by removing credentials and updating IP address:
[image]

and changed the time_avoid_duplicate value to 150ms:
[image]

Router Model: UniFi AP-nanoHD (small business router with strong,reliable Wifi connection)

Version: Release 0.9.4beta [Github Pre Release](https://github.com/1technophile/OpenMQTTGateway/tree/v0.9.4beta)
Wifi Manager setup:
[image]

**ESP32 Logs with RF module attached (core dump happens):**
```
*WM: [3] Sent wifi save page 
*WM: [2] process connect 
*WM: [2] Connecting as wifi client... 
*WM: [1] STA static IP:
*WM: [2] setSTAConfig static ip not set 
*WM: [3] WIFI station disconnect 
*WM: [1] Connecting to new AP: unifi_not
*WM: [3] Using Password: ********
*WM: [3] WiFi station enable 
*WM: [1] connectTimeout not set, ESP waitForConnectResult... 
*WM: [2] Connection result: WL_CONNECTED
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [1] Connect to new AP [SUCCESS] 
*WM: [1] Got IP Address: 
*WM: [1] 10.1.1.96 
*WM: [2] disconnect configportal 
*WM: [2] restoring usermode STA
*WM: [2] wifi status: WL_CONNECTED
*WM: [2] wifi mode: STA
*WM: [1] config portal exiting 
N: Wifi connected
                 {"mqtt_server":"10.1.1.2","mqtt_port":"1883","mqtt_user":"","mqtt_pass":"","mqtt_topic":"home/","gateway_name":"gw"}N: RF_EMITTER_PIN: 12 
                                                                                                                                                           N: RF_RECEIVER_PIN: 27 
                                                                                                                                                                                  N: Setup OpenMQTTGateway end
                                                                                                                                                                                                              N: Erasing ESP Config, restarting
                                                                                                                                                                                                                                               W: ESP32: Forcing to wifi 0
                                                                                                                                                                                                                                                                          Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Core 1 register dump:
PC      : 0x40080f7f  PS      : 0x00060034  A0      : 0x8008115d  A1      : 0x3ffbe750  
A2      : 0x00000001  A3      : 0x0000001a  A4      : 0x0000001a  A5      : 0x00000031  
A6      : 0x000000ba  A7      : 0x00000000  A8      : 0x800814aa  A9      : 0x3f4053c2  
A10     : 0x00000000  A11     : 0x000000ad  A12     : 0x8008a633  A13     : 0x3f4053c2  
A14     : 0x00000008  A15     : 0x00000001  SAR     : 0x00000014  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff  
Core 1 was running in ISR context:
EPC1    : 0x400621b0  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40080f7f

Backtrace: 0x40080f7f:0x3ffbe750 0x4008115a:0x3ffbe780 0x4008123d:0x3ffbe7b0 0x40084a31:0x3ffbe7d0 0x400621ad:0x3ffb1c70 0x40092672:0x3ffb1ca0 0x40092711:0x3ffb1cd0 0x40092ba6:0x3ffb1cf0 0x40087091:0x3ffb1d10 0x40145d96:0x3ffb1d30 0x400f28d1:0x3ffb1d50 0x400f418a:0x3ffb1d80 0x400f1bca:0x3ffb1db0 0x400f14d6:0x3ffb1dd0 0x400e22f0:0x3ffb1e20 0x400d2138:0x3ffb1e40 0x400d48f6:0x3ffb1e60 0x400d5285:0x3ffb1f70 0x400d538b:0x3ffb1f90 0x400e5b75:0x3ffb1fb0 0x40088e3d:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
[... 151 lines cut ...]
*WM: [2] Connecting as wifi client... 
*WM: [1] STA static IP:
*WM: [2] setSTAConfig static ip not set 
*WM: [3] WIFI station disconnect 
*WM: [1] No saved credentials, skipping wifi 
*WM: [2] Connection result: WL_NO_SSID_AVAIL
*WM: [3] lastconxresult: WL_NO_SSID_AVAIL
*WM: [1] AutoConnect: FAILED 
*WM: [2] AccessPoint set password is VALID 
*WM: [1] your_password 
*WM: [3] WIFI station disconnect 
*WM: [3] WiFi station enable 
*WM: [2] Disabling STA 
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  OpenMQTTGateway_ESP32_RF
*WM: [1] AP IP address: 192.168.4.1
*WM: [3] setupConfigPortal 
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip:  192.168.4.1
*WM: [2] HTTP server started 
*WM: [2] WiFi Scan ASYNC started 
*WM: [2] Config Portal Running, blocking, waiting for clients... 
*WM: [2] WiFi Scan ASYNC completed in 2911 ms
*WM: [2] WiFi Scan ASYNC found: 9
```

1technophile, 2020-06-02:
> Erasing ESP Config, restarting

This message appears when `TRIGGER_PIN` is set to low and generate the reset of the ESP.

Do you have something connected to `TRIGGER_PIN` gpio 0 per default on ESP32 ?

danobot, 2020-06-03:
good point but not that I am aware.. I have:
* power connected via USB to ESP32. 
* Power connected to RF module and
* D27 connected to Data pin of RF module.

No special setup, this is complete stock as per documentation. Figured I'd give that a go before customising it. (usb connection not shown below)

[image]

I have checked the connections. The RF module works perfectly on Raspberry Pi.
Are there specific models for ESP32 that are recommended? Mine is ESP32 DEVKIT V1.
Any known issues with USB power in the long run?

1technophile, 2020-06-03:
> Are there specific models for ESP32 that are recommended? Mine is ESP32 DEVKIT V1.

You can see that in the [compatibility list](https://docs.google.com/spreadsheets/d/1_5fQjAixzRtepkykmL-3uN3G5bLfQ0zMajM9OBZ1bx0/edit#gid=1323184277).

> Any known issues with USB power in the long run?

This can be of course an issue, I have the case of if connected to an USB Hub the ESP reset continuously.

>No special setup, this is complete stock as per documentation. Figured I'd give that a go before customising it. (usb connection not shown below)

Try to change the value of TRIGGER_PIN to 26 to see if you get the same behaviour.

1technophile, 2020-06-07:
@danobot did you got the opportunity to test with another `TRIGGER_PIN `?

danobot, 2020-06-08:
Yes, changing the TRIGGER_PIN allowed me to set up the ESP32 and had it run for 5 days. Last night, it crashed again. Havent checked the logs why it crashed this time yet.

danobot, 2020-06-17:
Thanks for the PR and fix.

1technophile, 2020-06-21:
So as to reproduce:
* with esp32dev-rf either with wifimanager or with manual config
* let the gateway connect to the wifi and mqtt server
* deactivate the wifi on your access point
* after some seconds the following error appears:
```
T: jsonPublishing
W: MQTT connection...
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Core 1 register dump:
PC      : 0x40080fe9  PS      : 0x00060034  A0      : 0x800811ca  A1      : 0x3ffbe750
A2      : 0x00000001  A3      : 0x0000001c  A4      : 0x00000000  A5      : 0x00000031
A6      : 0x0000000b  A7      : 0x00000000  A8      : 0x8008151a  A9      : 0x3f402458
A10     : 0x00000000  A11     : 0x000000d0  A12     : 0x00000000  A13     : 0x3f402458
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x0000001c  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000
Core 1 was running in ISR context:
EPC1    : 0x40086d5f  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40080fe9

Backtrace: 0x40080fe9:0x3ffbe750 0x400811c7:0x3ffbe780 0x400812ad:0x3ffbe7b0 0x40084aa1:0x3ffbe7d0 0x40086d5c:0x3ffba190 0x40082ebb:0x3ffba1b0 0x40088ead:0x3ffba1d0

Rebooting...
ets Jun  8 2016 00:22:57
```

The interesting sentence is C`ore 1 was running in ISR context:`, after some researches I found [this](https://github.com/espressif/arduino-esp32/issues/855) and [this](https://github.com/espressif/arduino-esp32/issues/1907) .

The issue occurs only when using rcswitch library, irremoteESP8266 doesn't generate this problem.

1technophile, 2020-06-21:
Correction pushed into RCSwitch repositories

closed 2020-06-21 by 1technophile