1technophile/OpenMQTTGateway #624: nRF24 / MySensors Support?

TRSx80, 2020-06-07:
I searched around, but was somewhat surprised to only find [this issue](https://github.com/1technophile/OpenMQTTGateway/issues/32) discussing possibility of nRf24 / MySensors support, where it seems you decided to only support RFM69.

I just did some [experiments](https://forum.mysensors.org/topic/11200/finally-progress-evidence-based-radio-testing-method) recently that have increased my confidence that I will be able to get nRF24L01+ working, which I had problems with before.

As I am already using OMG for my 433mhz stuff [1], of course it is only natural to look around for MySensors support, before getting out the soldering iron to assemble a second, largely redundant gateway.

I am not sure how you feel about it, however I can say this about the history of MySensors, which I have been studying and (trying to) use for a few years now.  It seems to me like the serial and even regular plain Ethernet gateways are going by the wayside in favor of Ethernet with MQTT for MySensors.  I know that now I am building a new gateway, this is the way I will go.

I am still pretty new to developing on Arduino, however I gathered from the RFM69 discussion I linked, that we may be pushing up against some size constraints if we include too much stuff (in "one file"(?))?  Perhaps some modular architecture might be helpful?  Forgive my ignorance / naivete if that is a foolish idea.

In the short term, I will probably just throw together another gateway, as Nano clones, nRF radios, and even Ethernet ports are cheap (and I need to put this 24 port switch I bought into service).  :smile:  But longer term, I hoped to start a discussion, or at least make a place to track such discussion, even if it might be far off on the horizon (or completely shot down) in case someone else comes along looking for same information.

[1] And it has run absolutely flawless so far!  Thank you very much not only for your work, but perhaps even moreso for sharing it like a civilized person!  :beers:

1technophile, 2020-06-07:
> As I am already using OMG for my 433mhz stuff [1], of course it is only natural to look around for MySensors support, before getting out the soldering iron to assemble a second, largely redundant gateway.

Of course, this is one of the first goals of OMG, reducing the number of gateways...

> I am still pretty new to developing on Arduino, however I gathered from the RFM69 discussion I linked, that we may be pushing up against some size constraints if we include too much stuff (in "one [...]

The nRF24 would be done on a separate file, ZgatewayNRF24.ino for example, I don't think there will be majors concerns about architecture.

> I hoped to start a discussion, or at least make a place to track such discussion, even if it might be far off on the horizon (or completely shot down) in case someone else comes along looking for same information.

Good idea, to complete the discussion I'm open on the NRF24 gateway adding, if someone is interested in proposing a PR, feel free to ask questions on this issue or in the [forum ](https://community.openmqttgateway.com/)

>[1] And it has run absolutely flawless so far! Thank you very much not only for your work, but perhaps even moreso for sharing it like a civilized person! It is greatly appreciated! 🍻

Thanks for this, it's nice to have feedback!

1technophile, 2023-02-04:
PR welcome

closed 2023-02-04 by 1technophile