1technophile/OpenMQTTGateway #630: IR SONY does not work on last development branch

Minims, 2020-06-12:
Before submitting a problem please check the troubleshooting section
https://docs.openmqttgateway.com/upload/troubleshoot.html

**Describe the bug**
mosquitto_pub  -t home/OpenMQTTGateway/commands/MQTTtoIR -m '{"hex":"0x490","protocol_name":"SONY","repeat":3}'

Works on master branch, not on current dev branch 0.9.4beta

Protocol NEC is working in both version.

**To Reproduce**
Steps to reproduce the behavior:
Compile latest master branch  0.9.3
Compile latest dev branch 0.9.4beta

**Expected behavior**
mosquitto_pub  -t home/OpenMQTTGateway/commands/MQTTtoIR -m '{"hex":"0x490","protocol_name":"SONY","repeat":3}'

TV react to the ir code

1technophile, 2020-06-18:
Hi,

Thanks for pointing this, giving a look now

1technophile, 2020-06-18:
Could you try with the last development branch not the v0.9.4beta. I have made [corrections ](https://github.com/1technophile/OpenMQTTGateway/pull/619) since the v0.9.4beta.

Minims, 2020-06-18:
Ok, i will try in few minutes :-)

Minims, 2020-06-18:
Same issue with last dev branch.

LED is blinking 3 times as expected, but the TV does not understand : 

```
home/OpenMQTTGateway/commands/MQTTtoIR {"hex":"0x490","protocol_name":"SONY", repeat:3}
home/OpenMQTTGateway/IRtoMQTT {"hex":"0x490","protocol_name":"SONY","repeat":3}

```

Minims, 2020-06-18:
found it , here is the fix : 

```
[MINIMS]$ git diff main/ZgatewayIR.ino
diff --git a/main/ZgatewayIR.ino b/main/ZgatewayIR.ino
index ee2623e..6c09aff 100644
--- a/main/ZgatewayIR.ino
+++ b/main/ZgatewayIR.ino
@@ -322,7 +322,7 @@ bool sendIdentifiedProtocol(const char* protocol_name, unsigned long long data,
   }
 #  endif
 #  ifdef IR_SONY
-  if (strcmp(protocol_name, "SONY") == 1) {
+  if (strstr(protocol_name, "SONY") != NULL) {
     Log.notice(F("Sending IR signal with %s"), protocol_name);
     if (valueBITS == 0)
       valueBITS = SONY_12_BITS;
```

1technophile, 2020-06-18:
Thanks, it should be:
`if (strcmp(protocol_name, "SONY") == 0) {`

1technophile, 2020-06-18:
Dev branch updated

Minims, 2020-06-18:
Thanks it's working great ! :-)

closed 2020-06-18 by Minims