1technophile/OpenMQTTGateway #635: 433MHz Wireless Outdoor Waterproof Infrared Beam Detector compatibility.

duyuelryel, 2020-06-15:
Greetings! Does anyone tryed this to work with OMG? Seller describes it as 433mhz.

https://a.aliexpress.com/_d6ayT9G

1technophile, 2020-07-05:
Hello,

Your link doesn't work

1technophile, 2020-07-05:
I didn't know that this kind of device exist. I will be interested by your feedback if your buy it. 
When I see the electronic and the different other components there is a big change that it will be compatible.

duyuelryel, 2020-07-05:
> I didn't know that this kind of device exist. I will be interested by your feedback if your buy it.
> When I see the electronic and the different other components there is a big change that it will be compatible.

Item description say, it sends a command to security systems by 433. I have askes seller and he said, it works with different security centers too. So i ordered it. It will arrive in the next weeks and will send infos about it. Photos etc.

duyuelryel, 2020-08-07:
So, the sensor just arrived. I dry tested it and it seems very good.
My OMG device is set to pilight, but my tasmotised sonof 433 rf bridge sees the code flawlessly. So it must be OMG compatible too. If you want, i can make a review just tell me where.

1technophile, 2020-08-07:
> So, the sensor just arrived. I dry tested it and it seems very good.

nice !

> If you want, i can make a review just tell me where.

It would be cool, you can write it on the [forum](https://community.openmqttgateway.com) and I will add a link to it on the compatible devices list.

duyuelryel, 2020-08-07:
> It would be cool, you can write it on the [forum](https://community.openmqttgateway.com) and I will add a link to it on the compatible devices list.

All right. I'll do it in the next days!

1technophile, 2020-12-11:
[Added ](https://compatible.openmqttgateway.com/index.php/product/gsmwond-wired-433mhz-wireless-barrier-2-beams-detector/)to the list of compatible devices

closed 2020-12-11 by 1technophile