1technophile/OpenMQTTGateway #640: WiFi connection Troubles

ajcollett, 2020-06-19:
**Describe the bug**
I have an Openwrt wifi device that I am trying to connect to. 

When connecting initially with the setup page, all goes well, and communication works to the broker. 

But after a reboot, the auto-connect fails, and the AP is put back up again. 

**To Reproduce**
Steps to reproduce the behaviour:
1. Flash
2. Connect to wireless successfully
3. reboot
4. Autoconnect wifi fails and AP is setup again for config. 

**Environment (please complete the following information):**
* OpenMQTTGateway version used 0.9.3 released now and tried on latest 0.8 too
* Base flash images, no custom code. 
* Openwrt wifi AP.

**Additional context**
Log:
```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00\00\00À0\00\00\00ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10044
load:0x40080400,len:5900
entry 0x400806ac
mounting FS...
mounted file system
reading config file
opened config file
{"mqtt_server":"<snip>","mqtt_port":"1883","mqtt_user":"home-mqtt","mqtt_pass":"<snip>","mqtt_topic":"home/","gateway_name":"IRgateway-tv-room"}
parsed json
*WM: [3] allocating params bytes: 20
*WM: [2] Added Parameter: server
*WM: [2] Added Parameter: port
*WM: [2] Added Parameter: user
*WM: [2] Added Parameter: pass
*WM: [2] Added Parameter: name
*WM: [3] Updated _max_params: 10
*WM: [3] re-allocating params bytes: 40
*WM: [2] Added Parameter: topic
*WM: [1] AutoConnect 
*WM: [2] ESP32 event handler enabled 
*WM: [2] Connecting as wifi client... 
*WM: [1] STA static IP:
*WM: [2] setSTAConfig static ip not set 
*WM: [3] WIFI station disconnect 
*WM: [1] Connecting to saved AP: Awe
*WM: [3] Using Password: <snip>
*WM: [3] WiFi station enable 
*WM: [2] 5000 ms connectTimeout set
*WM: [2] 5000 ms timeout, waiting for connect...
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] . 
*WM: [2] Connection result: WL_IDLE_STATUS
*WM: [3] lastconxresult: WL_IDLE_STATUS
*WM: [1] AutoConnect: FAILED 
*WM: [2] AccessPoint set password is VALID 
*WM: [1] your_password 
*WM: [3] WIFI station disconnect 
*WM: [3] WiFi station enable *WM: [2] [EVENT] WIFI_REASON: 8

*WM: [2] Disabling STA 
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  OpenMQTTGateway_ESP32_IR
*WM: [1] AP IP address: 192.168.4.1
*WM: [3] setupConfigPortal 
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip:  192.168.4.1
*WM: [2] HTTP server started 
*WM: [2] WiFi Scan ASYNC started 
*WM: [2] Config Portal Running, blocking, waiting for clients... 
*WM: [2] WiFi Scan ASYNC completed in 2307 ms
*WM: [2] WiFi Scan ASYNC found: 3

Brownout detector was triggered

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x 0@ @\00\00\00
```

1technophile, 2020-06-19:
Hello, Could you try with the last development branch?

ajcollett, 2020-06-19:
Sure, do you mean 0.9.4 beta? I don't have a dev env setup to build the images yet.

ajcollett, 2020-06-19:
With 0.9.4 beta I still get the same issue. Are there known WIFI setups that struggle? Or is my DHCP too slow?

1technophile, 2020-06-19:
I mean with the dev branch (more advanced compared to v0.9.4beta), are you able to do a build from it?

ajcollett, 2020-06-19:
OK! We have progress. 

Dev branch works. I set it up with all the details so I didn't need to set them in wifi manager. 

It connects to the one station. However, when I move it to the other side of the house, it doesn't "roam" to the other AP with the same SSID. It still seems to be trying to connect to the first AP, even though the SSID's are the same. 

If I turn off the original AP, it still won't connect to the other station. 

Is this a known thing?

Should I get log output for this?

As to this issue, it's probaly fixed in dev, so we can probably close this?

1technophile, 2020-06-19:
Thanks for the feedback, yes you can close it and open another regarding roaming (that I have never tested)

ajcollett, 2020-06-19:
Ok I got it to connect to the other AP. It was on another channel. When both AP's are on the same channel it seems to be okay to switch. Still, it's weird. Okay, I will open another after checking some details about EPS32. Maybe it's not a Openmqtt issue.

closed 2020-06-19 by ajcollett