1technophile/OpenMQTTGateway #688: Add LYWSD03 to BLE devices

michapr, 2020-07-04:
Is there a chance to add the LYWSD03 device?
I have searched how I could add it in code, but didn't found the right way.

If it helps - here is the python integration for RPIEasy:
https://github.com/enesbcs/rpieasy/blob/master/_P517_BLELYWSD03.py

BTW: is there any document where I can find how to integrate new device?

Thanks!
Michael

michapr, 2020-07-04:
Additional info:
Service Data UUID: 0xFE95
Data (as sample): 0x30585B0501B2F66B38C1A4280100
complete local name: LYWSD03MMC
(23.9°C, 53.2% rH)

1technophile, 2020-07-05:
Hello;

you can find an open issue for this subject below:
#540 

The difficulty with this one is the data encryption, @Floyddotnet was working on it, maybe he could say if there is some progress.
I close this issue as it is a duplicate.

closed 2020-07-05 by 1technophile