1technophile/OpenMQTTGateway #691: Multiple 433 MHz frames

rikouest, 2020-07-07:
Hi,

I flashed a Wemos D1 with OpenMQTTGateway. Using a 433 MHz wall switch, when it is pressed, multiple frames are sent, too many of them, bringing soooo many Mqtt frames to be sent : impossible to use this 433 MHz solution as on/off switch !

Is there a way to make the receiver wait 2 seconds after the 1st frame has been received and sent to MQTT, before receiving more frames ?
I tried to add _delay(2000)_ before :
_boolean result = client.publish(subjectRFtoMQTT,(char *)value.c_str());_
in ZgatewayRF, but it brings nothing more, received frames are all sent again just after the pause.

Any idea how to avoid this "radio bounce" ?

1technophile, 2020-07-07:
Hi,

>multiple frames are sent, too many of them, bringing soooo many Mqtt frames to be sent : impossible to use this 433 MHz solution as on/off switch !

Are you receiving the same values or different values? If they are the same you may increase the [time ](https://github.com/1technophile/OpenMQTTGateway/blob/0e257a58c137124c81967086970353640ec411c8/main/User_config.h#L273)between duplicates.
If they are different maybe there is something wrong with your setup, could you try another receiver PIN?

rikouest, 2020-07-10:
I tried #define time_avoid_duplicate 5000, same problem, not working

1technophile, 2020-07-10:
Could you post a serial monitor extract when the signal is received, I'm asking myself also why the issue doesn't have the  information requested on the template...

rikouest, 2020-07-12:
One single press and I get two frames sent to MQTT :

Rcv. RF
isAduplicate?
Adv data RFtoMQTT
Sending RFtoMQTT
2468196
RFtoMQTT OK
Rcv. RF
isAduplicate?
Adv data RFtoMQTT
Sending RFtoMQTT
2468196
RFtoMQTT OK

1technophile, 2020-07-12:
Could you give the information requested in the template please?
It could avoid iterating on the issue.

1technophile, 2020-07-24:
I'm closing the issue, when you have the information feel free to reopen it.

closed 2020-07-24 by 1technophile