1technophile/OpenMQTTGateway #692: ArduinoJson 6.x compatibility issues

michapr, 2020-07-07:
There are some compatibility issues if you are updating used ArduinoJson from 5.x to 6.x

Compilation failed with remark to https://arduinojson.org/v6/doc/upgrade/

Michael

1technophile, 2020-07-07:
Yes, so as to understand your need, despite the fact it would be better to support 6.x do you miss some functions on 5.x that are addressed in 6.x?

michapr, 2020-07-07:
There is no real "need", but if user have installed actual version of ArduinoJson he wiill get an error while compiling (like me after updating ;) ...) Compile the project is possible only with older 5.x version.

Not sure what is the best way to solve it,....

1technophile, 2020-07-07:
> Not sure what is the best way to solve it,....

We need to adapt the code to the new version and make extensive test to verify that there is no impact.

Legion2, 2020-07-12:
If we make our code compatible with ArduinoJson 6.x it will no longer be compatible with ArduinoJson 5.x or older versions.
@michapr the real problem here is the arduino ecosystem, which only allows you to install one library version globally, so that every sketch uses it. Platformio solved this by letting the sketch specify the libraries with version it needs and installs them automatically for each build of the project.

michapr, 2020-07-12:
>the real problem here is the arduino ecosystem, 

Yes, that's why I thought "Not sure what is the best way to solve it,...." ... ;)

Anyway I think this is the right solution now - Thanks! 

Michael

closed 2020-07-12 by michapr