1technophile/OpenMQTTGateway #693: Send 40-bit rf signal with some bits encoded incorrectly

sleepwalkera, 2020-07-07:
**Describe the bug**
Use MQTTto433 to send a 40-bit rf signal, but some of the bits in the actual signal are encoded incorrectly.

**To Reproduce**
Steps to reproduce the behavior:
1.  Publish to `mosquitto_pub -h localhost -u mqtt -P mqtt -t "home/OpenMQTTGateway/commands/MQTTto433" -m '{"value":"1001010110101100101101010000010100010001","protocol":1,"length":40,"repeat":4}'`
2. In serial monitor `N: Received json : {"value":"1001010110101100101101010000010100010001","protocol":1,"length":40,"repeat":4}`
3. The signal actual send is “0000000010010101010101101110110000010001“.

**Expected behavior**
The signal actual send should be "1001010110101100101101010000010100010001".

**Screenshots**
[image]

**Environment (please complete the following information):**
* OpenMQTTGateway version used (OpenMQTTGateway v0.9.4)
* Library version related to the problem you have (nodemcuv2-rf-libraries.zip comes with OpenMQTTGateway v0.9.4 release)
* Have tested with the basic examples SendDemo given with rc-switch, send 40-bit signal correctly.

**Additional context**
Also tried send "1111111111111111111111111111111111111111" and "0000000011111111111111111111111111111111", the signal actual send is “0000000011000111000111000111000111000111“.
[image]

sleepwalkera, 2020-07-07:
When I use decimal code
`mosquitto_pub -h localhost -u mqtt -P mqtt -t "home/OpenMQTTGateway/commands/MQTTto433" -m '{"value":642847671569,"protocol":1,"length":40,"repeat":4}'
Decimal 642847671569 which is binary 1001010110101100101101010000010100010001
The signal actual send is “0000000010010101010101101110110000010001“
It is more similar to the original code. Only the highest 8 bits are inconsistent.
[image]

1technophile, 2020-07-07:
Hello,

RCswitch fork used doesn't support 40bits signals, only up to [32 bits](https://github.com/1technophile/rc-switch/blob/8fc6d06c6e030b9f3b6fa99d56daccf3e33a50f6/src/RCSwitch.h#L60)

You may try this [fork ](https://github.com/sui77/rc-switch/pull/221/files) if you want 64bits support. I would be interested to know if working.

For my curiosity what kind of device need a 40bits signal?

sleepwalkera, 2020-07-08:
Thanks for reply
I have tested with the basic examples SendDemo given by nodemcuv2-rf-libraries.zip comes with OpenMQTTGateway v0.9.4 release.It can send 40-bit signal correctly.This shows that this version of rc-switch can already support 40-bit signals.But using MQTTto433 fails.

The wireless emitter of my motorized clothes rack use these 40-bit signal. May also be used on some electric curtain tracks.
[https://www.alibaba.com/product-detail/Dooya-Control-System-Single-Channel-433_60316301013.html](url)

1technophile, 2020-07-08:
> I have tested with the basic examples SendDemo given by nodemcuv2-rf-libraries.zip comes with OpenMQTTGateway v0.9.4 release.It can send 40-bit signal correctly.

Interesting

> But using MQTTto433 fails.

Due to the fact we are using `unsigned long` into the code, you may try to replace it with `unsigned long long` to handle 64 bits code.

sleepwalkera, 2020-07-08:
> Interesting
>
>
> Due to the fact we are using `unsigned long` into the code, you may try to replace it with `unsigned long long` to handle 64 bits code.

Thanks
According to your prompt, I modified the code, and now it can send 40-bit signals normally. 
Can you add this feature in the next release?

1technophile, 2020-07-08:
Great, I will add it!

1technophile, 2020-07-10:
Added to RCSwitch [fork](https://github.com/1technophile/rc-switch/commit/385a7e065f225c159bcd0d1b6278495423a1bd23)

1technophile, 2020-07-11:
@sleepwalkera would it be possible to test this PR :  https://github.com/1technophile/OpenMQTTGateway/pull/698

sleepwalkera, 2020-07-15:
@1technophile Have tested this PR. Works like a charm👍

1technophile, 2020-07-15:
Thanks for the feedback !

closed 2020-07-18 by 1technophile