1technophile/OpenMQTTGateway #705: Hi, i buy a AC Meter AT3010. work thru BLE

rdaguerre, 2020-07-21:
hi, how can i add support for this?
excuse me for my english
https://www.aliexpress.com/item/32957814319.html 
thanks

1technophile, 2020-07-21:
If you have the device could you download[ beacon simulator app ](https://play.google.com/store/apps/details?id=net.alea.beaconsimulator)

and see if your device advertise service data like below
[image]

rdaguerre, 2020-07-21:
[image] [image]

i want add a new device unknown.
thanks for your interesting.

1technophile, 2020-07-21:
and could you post the corresponding data seen on your BLE device?
The next step is to search a correspondance between the service data and the displayed ones.

1technophile, 2020-08-06:
I'm closing this one as I have no news, if you still need help do not hesitate to reopen it. Or better ask a question in the [forum](community.openmqttgateway.com).

1technophile, 2020-08-24:
@septs thanks

closed 2020-11-26 by stale[bot]