1technophile/OpenMQTTGateway #708: Sonoff RF Bridge - Installed OMG but no AP is visible - Bridge is no longer flashable.

gaussey, 2020-07-24:
Solder connection broke on the board after flashing so is no longer able to flash, no AP is visible but the device is flashing when I send a command from an RF device. Am I screwed? I’ve ordered another Sonoff RF in the meantime.

1technophile, 2020-07-25:
> Solder connection broke on the board after flashing so is no longer able to flash

You can try to scratch the circuit path with a cutter and solder there.

> no AP is visible but the device is flashing when I send a command from an RF device. 
A flashing led just mean that the internal RF decoder receive a signal (not linked with ESP8285)

1technophile, 2020-08-26:
I close the issue, if you have other questions, do not hesitate to ask on the [forum](https://community.openmqttgateway.com/)

closed 2020-08-26 by 1technophile