1technophile/OpenMQTTGateway #711: IR Repeat when using Raw

ajcollett, 2020-07-26:
Hello!

So I have been getting somewhere with my setup. 

I had to enable Raw signals from IR, since the power on signal on my Sony Bravia isn't in the DB it seems. 

So, when I try send RAW data, I can get it to work, but only if I send the code 3 times in a row. 

Now on the other buttons, I have set repeat to 3. In this case, it doesn't matter what I do, I can't get it to work without send the mqtt payload 3 times in rapid succession. 

Any ideas? Seems to me that repeat doesn't work on Raw?

Version: git master from about 1 or 2 weeks ago. I can try latest?

IR code I receive when pressing power on the remote:
```
{
    "value": 4045713590,
    "bits": 13,
    "protocol": -1,
    "hex": "0xF124B0B6",
    "protocol_name": "UNKNOWN",
    "raw": "2448,558,1256,574,632,562,1230,574,636,564,1226,578,630,566,630,568,1224,580,628,572,628,568,628,570,626"
}
```
What I have to send to get any response, publish x3 times:
```
{
    "bits": 13,
    "protocol_name": "Raw",
    "raw": "2448,558,1256,574,632,562,1230,574,636,564,1226,578,630,566,630,568,1224,580,628,572,628,568,628,570,626"
}
```

1technophile, 2020-07-27:
Hello,

> I had to enable Raw signals from IR, since the power on signal on my Sony Bravia isn't in the DB it seems.

You mean that you can't find it in [global cache DB](https://irdb.globalcache.com/)?

> Any ideas? Seems to me that repeat doesn't work on Raw?

Yes, I confirm that you can't repeat raw signals. on the current version.
I can add it to the next one if you are interested.

> Version: git master from about 1 or 2 weeks ago. I can try latest?

ajcollett, 2020-07-30:
Hey!

I didn't try very hard to look in the global cache DB. I struggled to get things working so went with what I knew, i.e. the raw Data I could get. I am not sure why things fail sometimes. I resorted to trying sending things, and assuming it wrong if I didn't get a repeat when listening to a topic. 

I think I'd like to help clarify the docs. When is what accepted? If I send a hex and RAW data, it seems to get ignored, but just raw data, then it works if I use "Raw" proto name. 

Anyway, clarity aside, it would be helpful to be able to repeat the Raw proto like these others. However, just docs on what is and isn't possible in each proto would be helpful. 

I think I would be keen to try help out with this project, since I am somewhat invested in it for my own home infra. But I do have 1 Q. What is the biggest difference you'd say, between this and [ESPHome](https://esphome.io/)?

1technophile, 2020-08-01:
> I think I'd like to help clarify the docs. When is what accepted?

Nice to propose, feel free to modify it and submit a PR, you have an edit button on the bottom of each doc page.

> What is the biggest difference you'd say, between this and ESPHome?

I have never used ESPHome so difficult to give you feedback.

What I can say is that OMG is focused on gateway use cases and we try to be creative and clear things:
* 2016 [433mhz to MQTT](https://1technophile.blogspot.com/2016/09/433tomqttto433-bidirectional-esp8266.html) gateway
* 2017 [Mi Flora integration](https://1technophile.blogspot.com/2017/11/mi-flora-integration-to-openmqttgateway.html) by BLE on an arduino with an HM10
* 2017 [IR to MQTT](https://1technophile.blogspot.com/2016/11/433nIRtomqttto433nIR-bidirectional-esp8266.html) gateway
* 2018 [Mi Flora integration by BLE on an ESP32](https://github.com/nkolban/esp32-snippets/issues/282)

So you may ask what will be out in 2020, surprise the best is in front of us :-)

closed 2020-10-12 by stale[bot]