1technophile/OpenMQTTGateway #714: Some BLE Advertising Packets lost

sleepwalkera, 2020-07-30:
Mijia smart locks use miBeacon (BLE advertising packet) to report door opening events. Although it is encrypted, I found that when a door opening event occurs, the service data always starts with the frame control bit "4840".

Normal service data: "5840fc03c12d3837a7fafc3ea5b482c0336a086cb5c2"

|data| name |
| ------ | ------ |
| 5840 | Frame Control |
| fc03 | Product ID |
| c1 | Frame Counter |
| 2d3837a7fafc | MAC Address |
| 3ea5b482c0336a086cb5c2  | encrypt data |

Door opening service data: "4840fc03a6eac3e7a81407f0ed151451d6167be5d50828ce"

|data| name |
| ------ | ------ |
| 4840 | Frame Control |
| fc03 | Product ID |
| a6eac3e7a81407f0ed151451d6167be5d50828ce  | encrypt data |

Therefore, I can simply use the frame control bit to identify the door open event.
I verified this with a simple pyblue script.
However, when I tried to capture ble packets using OMG, I found that not every BLE advertising packet was captured by BTtoMQTT.
Although I set TimeBtw_Read to 500, the scanning speed is still very slow, and I think I missed some event packets.

I want to disable other features such as device discovery to speed up scanning. But I am not familiar with the code. Can you give me some debugging guidance?

[https://iot.mi.com/new/doc/embedded-development/ble/ble-mibeacon.html](miBeacon)

1technophile, 2020-07-30:
Hello,

Nice finding!

You should try to play with [setInterval](https://github.com/eos1d3/ESP32_BLE/blob/9d6c1edd86ce330e16a57d44e555f56f6fd9e256/src/BLEScan.h#L55) and [setWindow ](https://github.com/eos1d3/ESP32_BLE/blob/9d6c1edd86ce330e16a57d44e555f56f6fd9e256/src/BLEScan.h#L56 )parameters.
By default they are set to [100](https://github.com/eos1d3/ESP32_BLE/blob/9d6c1edd86ce330e16a57d44e555f56f6fd9e256/src/BLEScan.cpp#L31), try maybe to go down the window to 50.

1technophile, 2020-07-30:
> I want to disable other features such as device discovery to speed up scanning. But I am not familiar with the code. Can you give me some debugging guidance?

Just comment ZmqttDiscovery if you are using Arduino IDE or remove this [line](https://github.com/1technophile/OpenMQTTGateway/blob/60d173ff5117aaaf5cf805f6a2146dca06fe0e93/platformio.ini#L127) if using platformio

1technophile, 2020-07-30:
Note that I will add the possibility to change the interval and window parameter in config_BT.h

sleepwalkera, 2020-07-30:
Thanks for your reply and debugging guidance
I have tried to set the setInterval and setWindow parameters to 50, or even 10, but still losing data packets.
This is my simple bluepy script. It runs perfectly on my laptop, and never lose any event packet.
[https://pastebin.com/03NG61nk](url)
I only have one esp32-dev, and I don't know what kind of information is helpful to locate the problem.

sleepwalkera, 2020-07-30:
It seems that before each line of "Found xxx devices, scan end deinit controller", there will be a pause of about 8 to 10 seconds. During this period, the device seems to be frozen.

1technophile, 2020-07-31:
Could you try to set the `Scan_duration` to 0 so as to see how the gateway behave?

sleepwalkera, 2020-07-31:
In the end, I tried setting Scan_duration to 3, which perfectly solved the problem and no longer lost the ble advertising packet. Now I can get the door opening event in homeassistant, thank you very much, I am willing to assist in any further testing

1technophile, 2020-07-31:
> In the end, I tried setting Scan_duration to 3, which perfectly solved the problem and no longer lost the ble advertising packet. Now I can get the door opening event in homeassistant, thank you very much, I am willing to assist in any further testing

Interesting,
Did you changed also `TimeBtw_Read `, `setInterval `and `setWindow `parameters?

sleepwalkera, 2020-08-01:
I set TimeBtw_Read to 500, and keep the setInterval and setWindow parameters as default values. It seems that Scan_duration is the key parameter. As long as Scan_duration is reduced, the packet capture rate will be greatly improved. But after hours of testing, I found that as stated in the comments, changing the value of Scan_duration will lead esp32 to instability (crashes and restarts approximately every ten minutes).

11:42:27.386 -> N: Device detected: F4FEFB808731
11:42:29.893 -> N: Found 2 devices, scan end deinit controller
11:42:30.384 -> N: Scan begin
11:42:30.802 -> N: Device detected: F4FEFB808731
11:42:30.910 -> N: Device detected: FCFAA737382D
11:42:30.910 -> N: Subject: /BTtoMQTT/FCFAA737382D
11:42:30.910 -> N: Received json: {"id":"FC:FA:A7:37:38:2D","rssi":-62,"servicedata":"5840fc03a62d3837a7fafc327596a3e2ad05b6556f40"}
11:42:33.595 -> N: Found 2 devices, scan end deinit controller
11:42:34.093 -> N: Scan begin
11:42:34.441 -> N: Device detected: F4FEFB808731
11:42:34.510 -> N: Device detected: FCFAA737382D
11:42:34.510 -> N: Subject: /BTtoMQTT/FCFAA737382D
11:42:34.510 -> N: Received json: {"id":"FC:FA:A7:37:38:2D","rssi":-67,"servicedata":"5840fc03a62d3837a7fafc327596a3e2ad05b6556f40"}
11:42:35.961 -> abort() was called at PC 0x401d13df on core 1
11:42:35.961 ->
11: 42: 35.996 -> Backtrace: 0x400926d0: 0x3ffd0480 0x40092901: 0x3ffd04a0 0x401d13df: 0x3ffd04c0 0x401d1426: 0x3ffd04e0 0x401d14d3: 0x3ffd0500 0x401d1556: 0x3ffd0520 0x401d156d: 0x3ffd0540 0x400da383: 0x3ffd0560 0x400d82e2: 0x3ffd05a0 0x400d59b8: 0x3ffd05c0 0x400ef0dd: 0x3ffd05e0 0x4008ee19: 0x3ffd0600
11:42:35.996 ->
11:42:35.996 -> Rebooting...
11:42:35.996 -> ets Jun 8 2016 00:22:57

At the same time, I would like to know if there is a more efficient way to capture a single specific event in addition to high-frequency scanning, as long as the data is not lost, and a slight delay is acceptable.

Thank you

1technophile, 2020-08-01:
Now that you have found an interesting `Scan_duration` I would try to change the `setWindow` parameter by putting it down and going up with the `setInterval`.
If you search on the web you may find also some guidance.
In all the case you have my encouragements, this is something interesting and that can help many others.

sleepwalkera, 2020-08-02:
After two days of testing and running, esp32 seems to no longer work through the openmqttgateway firmware. No matter I re-upload the program or erase the flash, the esp32 enters a repeated cycle and restarts, and it should be damaged. It is not yet known whether it is related to high-frequency scanning, but I suggest adjusting this parameter carefully.

00:19:41.803 -> N: 
00:19:41.803 -> ************* WELCOME TO OpenMQTTGateway **************
00:19:41.909 -> 
00:19:41.909 -> Brownout detector was triggered
00:19:41.909 -> 
00:19:41.909 -> ets Jun  8 2016 00:22:57
00:19:41.909 -> 
00:19:41.909 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
00:19:41.909 -> configsip: 0, SPIWP:0xee
00:19:41.909 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
00:19:41.909 -> mode:DIO, clock div:2
00:19:41.909 -> load:0x3fff0018,len:4
00:19:41.909 -> load:0x3fff001c,len:1216
00:19:41.943 -> ho 0 tail 12 room 4
00:19:41.943 -> load:0x40078000,len:9720
00:19:41.943 -> ho 0 tail 12 room 4
00:19:41.943 -> load:0x40080400,len:6364
00:19:41.943 -> entry 0x400806b8
00:19:42.505 -> N: 
00:19:42.505 -> ************* WELCOME TO OpenMQTTGateway **************
00:19:42.607 -> 
00:19:42.607 -> Brownout detector was triggered
00:19:42.607 -> 
00:19:42.607 -> ets Jun  8 2016 00:22:57
00:19:42.607 -> 
00:19:42.607 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
00:19:42.607 -> configsip: 0, SPIWP:0xee
00:19:42.643 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
00:19:42.643 -> mode:DIO, clock div:2
00:19:42.643 -> load:0x3fff0018,len:4
00:19:42.643 -> load:0x3fff001c,len:1216
00:19:42.643 -> ho 0 tail 12 room 4
00:19:42.643 -> load:0x40078000,len:9720
00:19:42.643 -> ho 0 tail 12 room 4
00:19:42.643 -> load:0x40080400,len:6364
00:19:42.643 -> entry 0x400806b8
00:19:43.247 -> N: 
00:19:43.247 -> ************* WELCOME TO OpenMQTTGateway **************
00:19:43.318 -> 
00:19:43.318 -> Brownout detector was triggered
00:19:43.318 -> 
00:19:43.318 -> ets Jun  8 2016 00:22:57
00:19:43.318 -> 
00:19:43.318 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
00:19:43.318 -> configsip: 0, SPIWP:0xee
00:19:43.355 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
00:19:43.355 -> mode:DIO, clock div:2
00:19:43.355 -> load:0x3fff0018,len:4
00:19:43.355 -> load:0x3fff001c,len:1216
00:19:43.355 -> ho 0 tail 12 room 4
00:19:43.355 -> load:0x40078000,len:9720
00:19:43.355 -> ho 0 tail 12 room 4
00:19:43.355 -> load:0x40080400,len:6364
00:19:43.355 -> entry 0x400806b8
00:19:43.956 -> N: 
00:19:43.956 -> ************* WELCOME TO OpenMQTTGateway **************
00:19:44.025 -> 
00:19:44.025 -> Brownout detector was triggered
00:19:44.025 -> 
00:19:44.025 -> ets Jun  8 2016 00:22:57
00:19:44.025 -> 
00:19:44.025 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
00:19:44.063 -> configsip: 0, SPIWP:0xee
00:19:44.063 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
00:19:44.063 -> mode:DIO, clock div:2
00:19:44.063 -> load:0x3fff0018,len:4
00:19:44.063 -> load:0x3fff001c,len:1216
00:19:44.063 -> ho 0 tail 12 room 4
00:19:44.063 -> load:0x40078000,len:9720
00:19:44.063 -> ho 0 tail 12 room 4
00:19:44.063 -> load:0x40080400,len:6364
00:19:44.063 -> entry 0x400806b8
00:19:44.655 -> N: 
00:19:44.655 -> ************* WELCOME TO OpenMQTTGateway **************
00:19:44.759 -> 
00:19:44.793 -> Brownout detector was triggered
00:19:44.793 -> 
00:19:44.793 -> ets Jun  8 2016 00:22:57
00:19:44.793 -> 
00:19:44.793 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
00:19:44.793 -> configsip: 0, SPIWP:0xee
00:19:44.793 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
00:19:44.793 -> mode:DIO, clock div:2
00:19:44.793 -> load:0x3fff0018,len:4
00:19:44.793 -> load:0x3fff001c,len:1216
00:19:44.793 -> ho 0 tail 12 room 4
00:19:44.793 -> load:0x40078000,len:9720
00:19:44.793 -> ho 0 tail 12 room 4
00:19:44.793 -> load:0x40080400,len:6364
00:19:44.828 -> entry 0x400806b8
00:19:45.391 -> N: 
00:19:45.391 -> ************* WELCOME TO OpenMQTTGateway **************
00:19:45.501 -> 
00:19:45.501 -> Brownout detector was triggered
00:19:45.501 -> 
00:19:45.501 -> ets Jun  8 2016 00:22:57
00:19:45.501 -> 
00:19:45.501 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
00:19:45.501 -> configsip: 0, SPIWP:0xee
00:19:45.501 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
00:19:45.501 -> mode:DIO, clock div:2
00:19:45.501 -> load:0x3fff0018,len:4
00:19:45.501 -> load:0x3fff001c,len:1216
00:19:45.501 -> ho 0 tail 12 room 4
00:19:45.501 -> load:0x40078000,len:9720
00:19:45.501 -> ho 0 tail 12 room 4
00:19:45.539 -> load:0x40080400,len:6364
00:19:45.539 -> entry 0x400806b8
00:19:46.109 -> N: 
00:19:46.109 -> ************* WELCOME TO OpenMQTTGateway **************
00:19:46.211 -> 
00:19:46.211 -> Brownout detector was triggered
00:19:46.211 -> 
00:19:46.211 -> ets Jun  8 2016 00:22:57
00:19:46.211 -> 
00:19:46.211 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
00:19:46.211 -> configsip: 0, SPIWP:0xee
00:19:46.211 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
00:19:46.211 -> mode:DIO, clock div:2
00:19:46.211 -> load:0x3fff0018,len:4
00:19:46.211 -> load:0x3fff001c,len:1216
00:19:46.211 -> ho 0 tail 12 room 4
00:19:46.211 -> load:0x40078000,len:9720
00:19:46.246 -> ho 0 tail 12 room 4
00:19:46.246 -> load:0x40080400,len:6364
00:19:46.246 -> entry 0x400806b8

1technophile, 2020-08-03:
I will submit in the following days a branch with a new BLE library, I would be interested by your feedback on it, if you have some time to test it.

sleepwalkera, 2020-08-03:
Great, I am very willing to test it

1technophile, 2020-08-18:
Hi @sleepwalkera,

Could you test with the last development version, please?

sleepwalkera, 2020-08-30:
Hi @1technophile ,
Thank you very much for your efforts. After several days of testing, it was found that the current version has amazing and significant improvements in stability. Even after several days of high-frequency scanning, it no longer crashes.

09:12:10.996 -> N: Scan begin
09:12:11.069 -> N: Device detected: F4:FE:FB:80:87:31
09:12:11.069 -> N: Subject: /BTtoMQTT/F4FEFB808731
09:12:11.069 -> N: Received json : {"id":"F4:FE:FB:80:87:31","rssi":-71}
09:12:11.388 -> N: Device detected: FC:FA:A7:37:38:2D
09:12:11.421 -> N: Subject: /BTtoMQTT/FCFAA737382D
09:12:11.421 -> N: Received json : {"id":"FC:FA:A7:37:38:2D","rssi":-74,"servicedata":"5840fc03ad2d3837a7fafc73dc36d98c1f5725cc83dc"}
09:12:12.020 -> N: Found 2 devices, scan number 2 end deinit controller
09:12:12.020 -> N: BLE Connect begin
09:12:12.269 -> N: BLE Connect end
09:12:12.590 -> N: Scan begin
09:12:12.660 -> N: Device detected: F4:FE:FB:80:87:31
09:12:12.660 -> N: Subject: /BTtoMQTT/F4FEFB808731
09:12:12.660 -> N: Received json : {"id":"F4:FE:FB:80:87:31","rssi":-70}
09:12:13.022 -> N: Device detected: FC:FA:A7:37:38:2D
09:12:13.022 -> N: Subject: /BTtoMQTT/FCFAA737382D
09:12:13.022 -> N: Received json : {"id":"FC:FA:A7:37:38:2D","rssi":-77,"servicedata":"5840fc03ad2d3837a7fafc73dc36d98c1f5725cc83dc"}
09:12:13.599 -> N: Found 2 devices, scan number 3 end deinit controller
09:12:13.599 -> N: BLE Connect begin
09:12:13.888 -> N: BLE Connect end
09:12:14.172 -> N: Scan begin
09:12:14.208 -> N: Device detected: F4:FE:FB:80:87:31
09:12:14.208 -> N: Subject: /BTtoMQTT/F4FEFB808731
09:12:14.208 -> N: Received json : {"id":"F4:FE:FB:80:87:31","rssi":-72}
09:12:14.635 -> N: Device detected: FC:FA:A7:37:38:2D
09:12:14.635 -> N: Subject: /BTtoMQTT/FCFAA737382D
09:12:14.635 -> N: Received json : {"id":"FC:FA:A7:37:38:2D","rssi":-76,"servicedata":"5840fc03ad2d3837a7fafc73dc36d98c1f5725cc83dc"}
09:12:15.169 -> N: Found 2 devices, scan number 4 end deinit controller
09:12:15.202 -> N: BLE Connect begin
09:12:15.475 -> N: BLE Connect end
09:12:15.766 -> N: Scan begin
09:12:15.906 -> N: Device detected: F4:FE:FB:80:87:31
09:12:15.906 -> N: Subject: /BTtoMQTT/F4FEFB808731
09:12:15.940 -> N: Received json : {"id":"F4:FE:FB:80:87:31","rssi":-71}
09:12:16.255 -> N: Device detected: FC:FA:A7:37:38:2D
09:12:16.255 -> N: Subject: /BTtoMQTT/FCFAA737382D
09:12:16.255 -> N: Received json : {"id":"FC:FA:A7:37:38:2D","rssi":-77,"servicedata":"5840fc03ad2d3837a7fafc73dc36d98c1f5725cc83dc"}
09:12:16.773 -> N: Found 2 devices, scan number 5 end deinit controller
09:12:16.773 -> N: BLE Connect begin
09:12:17.057 -> N: BLE Connect end
09:12:17.342 -> N: Scan begin
09:12:17.448 -> N: Device detected: F4:FE:FB:80:87:31
09:12:17.448 -> N: Subject: /BTtoMQTT/F4FEFB808731
09:12:17.448 -> N: Received json : {"id":"F4:FE:FB:80:87:31","rssi":-70}
09:12:17.870 -> N: Device detected: FC:FA:A7:37:38:2D
09:12:17.870 -> N: Subject: /BTtoMQTT/FCFAA737382D
09:12:17.870 -> N: Received json : {"id":"FC:FA:A7:37:38:2D","rssi":-77,"servicedata":"5840fc03ad2d3837a7fafc73dc36d98c1f5725cc83dc"}
09:12:18.334 -> N: Found 2 devices, scan number 6 end deinit controller
09:12:18.369 -> N: BLE Connect begin
09:12:18.619 -> N: BLE Connect end
09:12:18.936 -> N: Scan begin
09:12:18.972 -> N: Device detected: F4:FE:FB:80:87:31
09:12:18.972 -> N: Subject: /BTtoMQTT/F4FEFB808731
09:12:18.972 -> N: Received json : {"id":"F4:FE:FB:80:87:31","rssi":-73}
09:12:19.481 -> N: Device detected: FC:FA:A7:37:38:2D
09:12:19.481 -> N: Subject: /BTtoMQTT/FCFAA737382D
09:12:19.481 -> N: Received json : {"id":"FC:FA:A7:37:38:2D","rssi":-77,"servicedata":"5840fc03ad2d3837a7fafc73dc36d98c1f5725cc83dc"}
09:12:19.943 -> N: Found 2 devices, scan number 7 end deinit controller
09:12:19.943 -> N: BLE Connect begin
09:12:20.222 -> N: BLE Connect end
09:12:20.540 -> N: Scan begin
09:12:20.998 -> N: Device detected: F4:FE:FB:80:87:31
09:12:20.998 -> N: Subject: /BTtoMQTT/F4FEFB808731
09:12:21.034 -> N: Received json : {"id":"F4:FE:FB:80:87:31","rssi":-72}
09:12:21.432 -> N: Device detected: FC:FA:A7:37:38:2D
09:12:21.432 -> N: Subject: /BTtoMQTT/FCFAA737382D
09:12:21.432 -> N: Received json : {"id":"FC:FA:A7:37:38:2D","rssi":-77,"servicedata":"5840fc03ad2d3837a7fafc73dc36d98c1f5725cc83dc"}
[... 113 lines cut ...]
00:57:28.932 -> N: Received json : {"uptime":6846,"version":"version_tag","freemem":65212,"rssi":-70,"SSID":"OpenWrt","ip":"192.168.0.191","mac":"80:7D:3A:CA:84:6C","wifiprt":0,"lowpowermode":0,"interval":300,"scanbcnct":1,"modules":"RFBT"}
00:57:28.932 -> N: Subject: /SYStoMQTT
00:57:48.964 -> N: Received json : {"uptime":6966,"version":"version_tag","freemem":65212,"rssi":-69,"SSID":"OpenWrt","ip":"192.168.0.191","mac":"80:7D:3A:CA:84:6C","wifiprt":0,"lowpowermode":0,"interval":300,"scanbcnct":1,"modules":"RFBT"}
00:57:48.964 -> N: Subject: /SYStoMQTT
00:57:48.964 -> N: Received json : {"uptime":7086,"version":"version_tag","freemem":65212,"rssi":-63,"SSID":"OpenWrt","ip":"192.168.0.191","mac":"80:7D:3A:CA:84:6C","wifiprt":0,"lowpowermode":0,"interval":300,"scanbcnct":1,"modules":"RFBT"}
00:57:48.964 -> N: Subject: /SYStoMQTT
00:57:48.964 -> N: Received json : {"uptime":7206,"version":"version_tag","freemem":65212,"rssi":-64,"SSID":"OpenWrt","ip":"192.168.0.191","mac":"80:7D:3A:CA:84:6C","wifiprt":0,"lowpowermode":0,"interval":300,"scanbcnct":1,"modules":"RFBT"}
00:57:48.964 -> N: Subject: /SYStoMQTT
[... cut ...]
01:00:09.192 -> N: Received json : {"uptime":8046,"version":"version_tag","freemem":65212,"rssi":-69,"SSID":"OpenWrt","ip":"192.168.0.191","mac":"80:7D:3A:CA:84:6C","wifiprt":0,"lowpowermode":0,"interval":300,"scanbcnct":1,"modules":"RFBT"}
01:02:09.388 -> N: Subject: /SYStoMQTT
01:02:09.388 -> N: Received json : {"uptime":8166,"version":"version_tag","freemem":65212,"rssi":-69,"SSID":"OpenWrt","ip":"192.168.0.191","mac":"80:7D:3A:CA:84:6C","wifiprt":0,"lowpowermode":0,"interval":300,"scanbcnct":1,"modules":"RFBT"}

Everything returned to normal after manual restart.

1technophile, 2020-08-30:
Thanks for the feedback!

I will try to reproduce with your parameters.

sleepwalkera, 2020-08-30:
The timing of failure is unpredictable, sometimes it will occur within an hour or two, and sometimes it will take several days.

1technophile, 2020-08-30:
Could you try to reproduce with TRACE log level instead of NOTICE please. We will have better chance to catch the problem if we do it both.

sleepwalkera, 2020-09-01:
14:50:51.814 -> N: Scan begin
14:50:51.814 -> T: Creating BLE buffer
14:50:51.814 -> N: Device detected: 56:CD:30:84:13:B5
14:50:51.814 -> T: getDeviceByMac 56:CD:30:84:13:B5
14:50:51.814 -> T: Manufacturer Data: 4c0010051b18a4c299
14:50:51.814 -> T: getDeviceByMac 56:CD:30:84:13:B5
14:50:51.814 -> T: manufacturerdata 4c0010051b18a4c299
14:50:51.814 -> T: Remove manufacturer data
14:50:51.814 -> N: Subject: /BTtoMQTT/56CD308413B5
14:50:51.814 -> N: Received json : {"id":"56:CD:30:84:13:B5","rssi":-69,"txpower":12}
14:50:51.814 -> T: jsonPublishing
14:50:51.814 -> N: Found 1 devices, scan number 5195 end deinit controller
14:50:51.814 -> N: BLE Connect begin
14:50:51.814 -> N: BLE Connect end
14:50:51.814 -> T: BT Task running on core: 0
14:50:51.814 -> N: Scan begin
14:50:51.814 -> T: Creating BLE buffer
14:50:51.814 -> N: Device detected: 56:CD:30:84:13:B5
14:50:51.814 -> T: getDeviceByMac 56:CD:30:84:13:B5
14:50:51.814 -> T: Manufacturer Data: 4c0010051b18a4c299
14:51:11.848 -> T: getDeviceByMac 56:CD:30:84:13:B5
14:51:11.848 -> T: manufacturerdata 4c0010051b18a4c299
14:51:11.848 -> T: Remove manufacturer data
14:51:11.848 -> N: Subject: /BTtoMQTT/56CD308413B5
14:51:11.848 -> N: Received json : {"id":"56:CD:30:84:13:B5","rssi":-69,"txpower":12}
14:51:11.848 -> T: jsonPublishing
14:51:11.848 -> N: FouT: BT Task running on core: 0
14:51:11.848 -> N: Scan begin
14:51:11.848 -> T: Creating BLE buffer
14:51:11.848 -> N: Device detected: 56:CD:30:84:13:B5
14:51:11.848 -> T: getDeviceByMac 56:CD:30:84:13:B5
14:51:11.848 -> T: Manufacturer Data: 4c0010051b18a4c299
14:51:11.848 -> T: getDeviceByMac 56:CD:30:84:13:B5
14:51:11.848 -> T: manufacturerdata 4c0010051b18a4c299
14:51:11.848 -> T: Remove manufacturer data
14:51:11.848 -> N: Subject: /BTtoMQTT/56CD308413B5
14:51:11.848 -> N: Received json : {"id":"56:CD:30:84:13:B5","rssi":-78,"txpower":12}
14:51:11.848 -> T: jsonPublishing
14:51:11.848 -> N: Found 1 devices, scan number 5207 end deinit controller
14:51:11.848 -> N: BLE Connect begin
14:51:11.848 -> N: BLE Connect end
14:51:11.848 -> T: BT Task running on core: 0
14:51:11.848 -> N: Scan begin
14:51:11.848 -> T: Creating BLE buffer
14:51:11.848 -> N: Device detected: 56:CD:30:84:13:B5
14:51:11.848 -> T: getDeviceByMac 56:CD:30:84:13:B5
14:51:11.848 -> T: Manufacturer Data: 4c0010051b18a4c299
14:51:11.848 -> T: getDeviceByMac 56:CD:30:84:13:B5
14:51:11.848 -> T: manufacturerdata 4c0010051b18a4c299
14:51:31.881 -> T: Remove manufacturer data
14:51:31.881 -> N: Subject: /BTtoMQTT/56CD308413B5
14:51:31.881 -> N: Received json : {"id":"56:CD:30:84:13:B5","rssi":-78,"txpower":12}
14:51:31.881 -> T: jsonPublishing
14:51:31.881 -> N: Found 1 devices, scan number 5208 end deinit controller
14:51:31.881 -> N: BLE Connect begin
14:51:31.881 -> N: BLE Connect end
14:51:31.881 -> T: BT N: Found 1 devices, scan number 5219 end deinit controller
14:51:31.881 -> N: BLE Connect begin
14:51:31.881 -> N: BLE Connect end
14:51:31.881 -> T: BT Task running on core: 0
14:51:31.881 -> N: Scan begin
14:51:31.881 -> T: Creating BLE buffer
14:51:31.881 -> N: Device detected: 70:D6:7A:03:77:8F
14:51:31.881 -> T: getDeviceByMac 70:D6:7A:03:77:8F
14:51:31.881 -> T: Manufacturer Data: 4c00100503188a36ce
14:51:31.881 -> T: Low rssi, device filtered
14:51:31.881 -> T: Creating BLE buffer
14:51:31.881 -> N: Device detected: 54:AF:A6:43:E4:F1
14:51:31.881 -> T: getDeviceByMac 54:AF:A6:43:E4:F1
14:51:31.881 -> T: Manufacturer Data: 4c0010051a1c0e88f0
[... 568 lines cut ...]
17:17:22.589 -> T: jsonPublishing
17:19:22.792 -> T: retrieving value of system characteristics Uptime (s):18603
17:19:22.792 -> N: Subject: /SYStoMQTT
17:19:22.792 -> N: Received json : {"uptime":18603,"version":"version_tag","freemem":41344,"rssi":-47,"SSID":"Openwrt","ip":"192.168.2.6","mac":"80:7D:3A:CA:84:6C","wifiprt":0,"lowpowermode":0,"interval":300,"scanbcnct":1,"modules":"RFBT"}
17:19:22.792 -> T: jsonPublishing
17:20:53.184 -> T: retrieving value of system characteristics Uptime (s):18723
17:20:53.184 -> N: Subject: /SYStoMQTT
17:20:53.184 -> N: Received json : {"uptime":18723,"version":"version_tag","freemem":41344,"rssi":-42,"SSID":"Openwrt","ip":"192.168.2.6","mac":"80:7D:3A:CA:84:6C","wifiprt":0,"lowpowermode":0,"interval":300,"scanbcnct":1,"modules":"RFBT"}
17:20:53.217 -> T: jsonPublishing
17:22:53.187 -> T: retrieving value of system characteristics Uptime (s):18843
17:22:53.187 -> N: Subject: /SYStoMQTT
17:22:53.225 -> N: Received json : {"uptime":18843,"version":"version_tag","freemem":41344,"rssi":-36,"SSID":"Openwrt","ip":"192.168.2.6","mac":"80:7D:3A:CA:84:6C","wifiprt":0,"lowpowermode":0,"interval":300,"scanbcnct":1,"modules":"RFBT"}
17:22:53.225 -> T: jsonPublishing
17:24:53.192 -> T: retrieving value of system characteristics Uptime (s):18963
17:24:53.225 -> N: Subject: /SYStoMQTT
17:24:53.225 -> N: Received json : {"uptime":18963,"version":"version_tag","freemem":41344,"rssi":-44,"SSID":"Openwrt","ip":"192.168.2.6","mac":"80:7D:3A:CA:84:6C","wifiprt":0,"lowpowermode":0,"interval":300,"scanbcnct":1,"modules":"RFBT"}
17:24:53.225 -> T: jsonPublishing
17:26:53.199 -> T: retrieving value of system characteristics Uptime (s):19083
17:26:53.199 -> N: Subject: /SYStoMQTT
17:26:53.232 -> N: Received json : {"uptime":19083,"version":"version_tag","freemem":41344,"rssi":-44,"SSID":"Openwrt","ip":"192.168.2.6","mac":"80:7D:3A:CA:84:6C","wifiprt":0,"lowpowermode":0,"interval":300,"scanbcnct":1,"modules":"RFBT"}
17:26:53.232 -> T: jsonPublishing

--------------------------------------------------------------------------------------

Scanning will not resume unless manually restarted

sleepwalkera, 2020-09-01:
20:53:31.322 -> N: BLE Connect begin
20:53:31.322 -> N: BLE Connect end
20:53:31.322 -> T: BT Task running on core: 0
20:53:31.322 -> N: Scan begin
20:53:31.322 -> T: Creating BLE buffer
20:53:31.322 -> N: Device detected: FC:FA:A7:37:38:2D
20:53:31.322 -> T: getDeviceByMac FC:FA:A7:37:38:2D
20:53:31.322 -> T: Manufacturer Data: 8f03
20:53:31.322 -> T: Get services data number: 1
20:53:31.322 -> T: Service data: 5840fc035d2d3837a7fafc26db7edfa263354345ee55
20:53:31.322 -> T: Service data UUID: 0xfe95
20:53:31.322 -> T: getDeviceByMac FC:FA:A7:37:38:2D
20:53:31.322 -> T: Checking BLE service data validity
20:53:31.322 -> T: Searching BLE device data 5840fc035d2d3837a7fafc26db7edfa263354345ee55 size 44
20:53:31.322 -> T: Is it a mi flora ?
20:53:31.322 -> T: Is it a vegtrug ?
20:53:31.322 -> T: Is it a LYWSDCGQ?
20:53:31.322 -> T: Is it a JQJCY01YM?
20:53:31.322 -> T: Is it a LYWSD02?
20:53:31.322 -> T: Is it a CGG1?
20:53:31.322 -> T: Is it a MUE4094RT?
20:53:31.322 -> T: Is it a CGP1W?
20:53:31.322 -> T: Is it a CGG1
20:53:31.322 -> T: Is it a CGD1?
20:53:31.322 -> T: Is it a MHO_C401?
20:53:31.322 -> T: Is it a LYWSD03MMC?
20:53:31.322 -> T: servicedatauuid 0xfe95
20:53:31.322 -> T: Is it a MiBand?
20:53:31.322 -> T: Is it a XMTZC04HM?
20:53:31.322 -> T: Is it a XMTZC05HM?
20:53:31.322 -> T: manufacturerdata 8f03
20:53:31.322 -> T: Remove manufacturer data
20:53:31.322 -> T: Removing servicedatauuid
20:53:31.322 -> N: Subject: /BTtoMQTT/FCFAA737382D
20:53:31.322 -> N: Received json : {"id":"FC:FN: BLE Connect end
20:53:31.322 -> T: BT Task running on core: 0
20:53:31.322 -> N: Scan begin
20:53:31.322 -> T: Creating BLE buffer
20:53:31.322 -> N: Device detected: FC:FA:A7:37:38:2D
20:53:31.322 -> T: getDeviceByMac FC:FA:A7:37:38:2D
20:53:31.322 -> T: Manufacturer Data: 8f03
20:53:31.322 -> T: Get services data number: 1
20:53:31.322 -> T: Service data: 5840fc035d2d3837a7fafc26db7edfa263354345ee55
20:53:31.322 -> T: Service data UUID: 0xfe95
20:53:31.322 -> T: getDeviceByMac FC:FA:A7:37:38:2D
20:53:31.322 -> T: Checking BLE service data validity
20:53:31.322 -> T: Searching BLE device data 5840fc035d2d3837a7fafc26db7edfa263354345ee55 size 44
20:53:31.322 -> T: Is it a mi flora ?
20:53:31.322 -> T: Is it a vegtrug ?
20:53:31.322 -> T: Is it a LYWSDCGQ?
20:53:31.322 -> T: Is it a JQJCY01YM?
20:53:31.322 -> T: Is it a LYWSD02?
20:53:31.322 -> T: Is it a CGG1?
20:53:31.322 -> T: Is it a MUE4094RT?
20:53:31.322 -> T: Is it a CGP1W?
20:53:31.322 -> T: Is it a CGG1
20:53:31.322 -> T: Is it a CGD1?
20:53:31.322 -> T: Is it a MHO_C401?
20:53:31.322 -> T: Is it a LYWSD03MMC?
20:53:31.322 -> T: servicedatauuid 0xfe95
20:53:31.322 -> T: Is it a MiBand?
20:53:31.322 -> T: Is it a XMTZC04HM?
20:53:31.322 -> T: Is it a XMTZC05HM?
20:53:31.322 -> T: manufacturerdata 8f03
20:53:31.322 -> T: Remove manufacturer data
20:53:31.322 -> T: Removing servicedatauuid
20:53:31.322 -> N: Subject: /BTtoMQTT/FCFAA737382D
20:53:31.322 -> N: Received json : {"id":"FC:FA:A7:37:38:2D","rssi":-72,"servicedata":"5840fc035d2d3837a7fafc26db7edfa263354345ee55"}
20:53:31.322 -> T: jsonPublishing
20:53:31.322 -> N: Found 2 devices, scan number 4641 end deinit controller
[... 80 lines cut ...]
20:53:31.322 -> T: jsonPublishing
20:53:31.322 -> T: retrieving value of system characteristics Uptime (s):8096
20:53:31.322 -> N: Subject: /SYStoMQTT
20:53:31.322 -> N: Received json : {"uptime":8096,"version":"version_tag","freemem":49700,"rssi":-76,"SSID":"OpenWrt","ip":"192.168.0.191","mac":"80:7D:3A:CA:84:6C","wifiprt":0,"lowpowermode":0,"interval":300,"scanbcnct":1,"modules":"RFBT"}
20:53:31.322 -> T: jsonPublishing
20:53:31.322 -> T: retrieving value of system characteristics Uptime (s):8216
20:53:31.322 -> N: Subject: /SYStoMQTT
20:53:31.322 -> N: Received json : {"uptime":8216,"version":"version_tag","freemem":49700,"rssi":-82,"SSID":"OpenWrt","ip":"192.168.0.191","mac":"80:7D:3A:CA:84:6C","wifiprt":0,"lowpowermode":0,"interval":300,"scanbcnct":1,"modules":"RFBT"}
20:53:31.322 -> T: jsonPublishing
20:53:31.322 -> T: retrieving value of system characteristics Uptime (s):8336
20:53:31.322 -> N: Subject: /SYStoMQTT
20:53:31.322 -> N: Received json : {"uptime":8336,"version":"version_tag","freemem":49700,"rssi":-80,"SSID":"OpenWrt","ip":"192.168.0.191","mac":"80:7D:3A:CA:84:6C","wifiprt":0,"lowpowermode":0,"interval":300,"scanbcnct":1,"modules":"RFBT"}
20:53:31.322 -> T: jsonPublishing
20:53:31.322 -> T: retrieving value of system characteristics Uptime (s):8456
20:53:31.322 -> N: Subject: /SYStoMQTT
20:53:31.322 -> N: Received json : {"uptime":8456,"version":"version_tag","freemem":49700,"rssi":-75,"SSID":"OpenWrt","ip":"192.168.0.191","mac":"80:7D:3A:CA:84:6C","wifiprt":0,"lowpowermode":0,"interval":300,"scanbcnct":1,"modules":"RFBT"}
20:53:31.322 -> T: jsonPublishing
20:54:11.497 -> T: retrieving value of system characteristics Uptime (s):8576
20:54:11.497 -> N: Subject: /SYStoMQTT
20:54:11.497 -> N: Received json : {"uptime":8576,"version":"version_tag","freemem":49700,"rssi":-80,"SSID":"OpenWrt","ip":"192.168.0.191","mac":"80:7D:3A:CA:84:6C","wifiprt":0,"lowpowermode":0,"interval":300,"scanbcnct":1,"modules":"RFBT"}
20:54:11.530 -> T: jsonPublishing

---------------------------------------------------------------------------------------------------------------

another try

1technophile, 2020-09-01:
> Is there a higher log level? I am very willing to assist in further testing

Yes but this one is sufficient. I did not reproduce exactly your issue and my ESP32 is getting quite hot. 
I will dig into this.

sleepwalkera, 2020-09-01:
Thank you.  Should I try to reduce the scanning speed?  Perhaps this has exceeded the hardware capabilities of esp32?  Or is it a software issue?

1technophile, 2020-09-01:
Yes maybe try to tune the settings, I don't know if the board is intended to be used like that.

1technophile, 2020-09-02:
Maybe we may go to another level and do the test with the [BLE library ](https://github.com/h2zero/NimBLE-Arduino) directly, try maybe first to reproduce the issue with the [BLEscan ](https://github.com/h2zero/NimBLE-Arduino/tree/master/examples/Refactored_original_examples/BLE_scan) example and if reproduced we could ask the library owner its advice.
Are you interested in this approach?

sleepwalkera, 2020-09-12:
Thanks for your advice. I tested the BLEscan example for a few days. Although the scan did not stop, the scan speed did slow down after running for a few days. I will continue to see if the issue can be reproduced.

sleepwalkera, 2020-09-21:
@1technophile I tested the BLEscan example for more than 10 days. The scan did not stop and the issue could not be reproduced.

sleepwalkera, 2020-12-14:
@1technophile I have tested with the BLE library directly. Looks like no problem at the library level. Would you go further for this?

1technophile, 2020-12-14:
Hi,

Are you using the BLEScan example without any modifications?

sleepwalkera, 2020-12-14:
@1technophile Yes, the BLEScan example running stably.

1technophile, 2021-01-30:
@sleepwalkera I suppose that you are using [this example](https://github.com/h2zero/NimBLE-Arduino/blob/master/examples/Refactored_original_examples/BLE_scan/BLE_scan.ino) and remove the delay line only?

1technophile, 2021-02-26:
@sleepwalkera could you test the development branch, it now has a lockless queue mechanism that could help with your issue.

sleepwalkera, 2021-03-16:
@1technophile After days of testing, it was found that the stability of the lockless queue version has been greatly improved. Although downtime still exists, it is much less likely to happen. Thank you for your hard work.
The following is the log
---------------------------------------------------------------------------------------------------
22:27:11.316 -> T: Get services data number: 1
22:27:11.316 -> T: Converted service data (22) to 5840fc035f2d3837a7fafc41a1a6384ff81994cf082f
22:27:11.316 -> T: Service data: 5840fc035f2d3837a7fafc41a1a6384ff81994cf082f
22:27:11.316 -> T: Service data UUID: 0xfe95
22:27:11.316 -> T: getDeviceByMac FC:FA:A7:37:38:2D
22:27:11.316 -> T: Checking BLE service data validity
22:27:11.316 -> T: Searching BLE device data 5840fc035f2d3837a7fafc41a1a6384ff81994cf082f size 44
22:27:11.316 -> T: Is it a mi flora ?
22:27:11.316 -> T: Is it a vegtrug ?
22:27:11.316 -> T: Is it a LYWSDCGQ?
22:27:11.316 -> T: Is it a JQJCY01YM?
22:27:11.316 -> T: Is it a LYWSD02?
22:27:11.316 -> T: Is it a MUE4094RT?
22:27:11.316 -> T: Is it a CGP1W?
22:27:11.316 -> T: Is it a CGG1
22:27:11.316 -> T: Is it a CGD1?
22:27:11.316 -> T: Is it a MHO_C401?
22:27:11.316 -> T: Is it a LYWSD03MMC?
22:27:11.316 -> T: Is it a custom LYWSD03MMC
22:27:11.316 -> T: servicedatauuid 0xfe95
22:27:11.316 -> T: Is it a MiBand?
22:27:11.316 -> T: Is it a XMTZC04HM?
22:27:11.316 -> T: Is it a XMTZC05HM?
22:27:11.316 -> T: manufacturerdata 8f03
22:27:11.316 -> T: Is it a iNode Energy Meter?
22:27:11.316 -> T: Remove manufacturer data
22:27:11.316 -> T: Removing servicedatauuid
22:27:11.316 -> N: Subject: /BTtoMQTT/FCFAA737382D
22:27:11.316 -> N: Received json : {"id":"FC:FA:A7:37:38:2D","rssi":-74,"servicedata":"5840fc035f2d3837a7fafc41a1a6384ff81994cf082f"}
22:27:11.316 -> T: jsonPublishing
22:27:11.316 -> T: Creating BLE buffer
22:27:11.316 -> N: Device detected: 4D:AA:48:42:FB:63
22:27:11.316 -> T: getDeviceByMac 4D:AA:48:42:FB:63
22:27:11.316 -> T: Manufacturer Data: 4c0010052a18211858
22:27:11.316 -> T: Low rssi, device filtered
22:27:11.316 -> T: Creating BLE buffer
22:27:11.316 -> N: Device detected: F4:FE:FB:80:87:31
22:27:11.316 -> T: getDeviceByMac F4:FE:FB:80:87:31
22:27:11.316 -> T: Manufacturer Data: 75004204012067200d0002012b01010001000000000000000000
22:27:11.316 -> T: getDeviceByMac F4:FE:FB:80:87:31
22:27:11.316 -> T: manufacturerdata 75004204012067200d0002012b01010001000000000000000000
22:27:11.316 -> T: Is it a iNode Energy Meter?
22:27:11.316 -> T: Remove manufacturer data
22:27:11.316 -> T: Creating BLE buffer
22:27:11.316 -> N: Subject: /BTtoMQTT/F4FEFB808731
22:27:11.316 -> N: Device detected: 56:56:B8:7B:9C:99
22:27:11.316 -> T: getDeviceByMac 56:56:B8:7B:9C:99
22:27:11.316 -> N: Received json : {"id":"F4:FE:FB:80:87:31","rssi":-78}
22:27:11.316 -> T: Manufacturer Data: 4c000719010f2021888f0100042b04ae4e0dcc83bc779d87e27f8ff176
22:27:11.316 -> T: jsoT: Low rssi, device filtered
22:27:11.316 -> nPublishing
22:27:11.316 -> N: Found 5 devices, scan number 29849 end deinit controller
22:27:11.316 -> N: BLE Connect begin
22:27:11.316 -> N: BLE Connect end

1technophile, 2021-03-26:
Much of the credits goes to @csiki2

closed 2021-03-26 by 1technophile