1technophile/OpenMQTTGateway #715: Received RF values are stored twice

jmw6773, 2020-08-02:
**Describe the bug**
Values received by ZgatewayRF (and most likely other gateways) are being stored in the `receivedSignal` buffer twice. (Once for the receiving of the signal and once for the MQTT callback to the `subjectMultiGTWKey` topic.

**To Reproduce**
Run the Zgateway RF with `LOG_LEVEL` set to `LOG_LEVEL_TRACE`.

When a RF signal is received:
1. the code is stored (if not duplicated)
2. OMG transmits to 433toMQTT
3. OMG receives callback from 433toMQTT (to prevent duplicate signals from other OMG devices)
4. the code is stored again (no check for duplicate value)

**Expected behavior**
Code should only be stored once. Duplicate values should always be checked.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
* Current development

**Additional context**
I believe this should be fixable by changing L1388, but don't know if it will have other adverse effects.
`    if (data != 0) {`
to
`    if (data != 0 && !isAduplicateSignal(data)) {`

https://github.com/1technophile/OpenMQTTGateway/blob/60d173ff5117aaaf5cf805f6a2146dca06fe0e93/main/main.ino#L1379-L1390

1technophile, 2020-08-02:
I think it's a good suggestion, feel free to submit a PR. 
And at first view no impact except the processing time.

jmw6773, 2020-08-03:
Looking closer, the test for `if (strstr(topicOri, subjectMultiGTWKey) != NULL)` (L1383) looks as though it can be removed as it is already performed in the `callback` function (L482). This should improve performance for all Gateways since it was outside the `#define`.

1technophile, 2020-08-03:
When we detect ` (strstr(topic, subjectGTWSendKey) != NULL)` in L482, (an MQTT command) we must not pass on the L1383 block so as to avoid storing the MQTT commands for deduplication.
Maybe we should change:
`  if (strstr(topicOri, subjectMultiGTWKey) != NULL) `
by
`  if ((strstr(topic, subjectGTWSendKey) == NULL) `
If you think it is more explicit.

jmw6773, 2020-08-03:
Oh, I understand what it's doing now. Yeah, it's fine as it is then. But it should probably be inside the `#define` as Gateways that aren't using RF/IR don't need to run the comparison, right?

```
#if defined(ZgatewayRF) || defined(ZgatewayIR) || defined(ZgatewaySRFB) || defined(ZgatewaySRFB) || defined(ZgatewayWeatherStation)
  if (strstr(topicOri, subjectMultiGTWKey) != NULL) { // storing received value so as to avoid publishing this value if it has been already sent by this or another OpenMQTTGateway 
    SIGNAL_SIZE_UL_ULL data = jsondata.success() ? jsondata["value"] : STRTO_UL_ULL(datacallback, NULL, 10);
    if (data != 0 && !isAduplicateSignal(data)) {
      storeSignalValue(data);
    }
  }
#endif
```

1technophile, 2020-08-03:
> But it should probably be inside the #define as Gateways that aren't using RF/IR don't need to run the comparison, right?

Yes you are right, well seen!

closed 2020-08-04 by 1technophile