1technophile/OpenMQTTGateway #716: MQTT user name

Pyromane, 2020-08-02:
**Describe the bug**
The MQTT user name entered in the web interface is not used when establishing a connection if you change it

**To Reproduce**
Here are the serial debug outputs when changing the user name:
The variable "mqtt_user": "" appears to be empty after the restart and the connection attempts
```
*WM: [2] <- HTTP WiFi save  <\r><\n>
*WM: [3] Method: POST<\r><\n>
*WM: [2] Parameters <\r><\n>
*WM: [2] -------------------- <\r><\n>
*WM: [2] server: 192.168.0.108<\r><\n>
*WM: [2] port: 1883<\r><\n>
*WM: [2] user: mqsensor<\r><\n>
*WM: [2] pass: PASSWORD<\r><\n>
*WM: [2] name: OpenMQTTGateway_ESP32_BLE<\r><\n>
*WM: [2] topic: home/<\r><\n>
*WM: [2] -------------------- 

*WM: [1] config portal exiting <\r><\n>
{"mqtt_server":"192.168.0.108","mqtt_port":"1883","mqtt_user":"","mqtt_pass":"PASSWORD","mqtt_topic":"home/","gateway_name":"OpenMQTTGateway_ESP32_BLE"}N: BLEinterval: 55555<\n>
```
Here with standard values:
```
*WM: [2] <- HTTP WiFi save  <\r><\n>
*WM: [3] Method: POST<\r><\n>
*WM: [2] Parameters <\r><\n>
*WM: [2] -------------------- <\r><\n>
*WM: [2] server: 192.168.0.108<\r><\n>
*WM: [2] port: 1883<\r><\n>
*WM: [2] user: your_username<\r><\n>
*WM: [2] pass: your_password<\r><\n>
*WM: [2] name: OpenMQTTGateway_ESP32_BLE<\r><\n>
*WM: [2] topic: home/<\r><\n>

*WM: [1] config portal exiting <\r><\n>
{"mqtt_server":"192.168.0.108","mqtt_port":"1883","mqtt_user":"your_username","mqtt_pass":"your_password","mqtt_topic":"home/","gateway_name":"OpenMQTTGateway_ESP32_BLE"}N: BLEinterval: 55555<\n>
N: Minrssi: -100<\n>
```

**Environment (please complete the following information):**
* OpenMQTTGateway v0.9.4 + OpenMQTTGateway v0.9.3
* esp32dev-ble-firmware.bin
* esp32dev-ble-partitions.bin

1technophile, 2020-08-02:
> when establishing a connection if you change it

Could you indicate how do you change it please (steps, method)?

Pyromane, 2020-08-02:
I flash the files mentioned above, restart the ESP32 and connect to the AP "OpenMQTTGateway_ESP32_BLE". In the browser I open the page 192.168.4.1 and choose "configure WiFi". The other settings can be seen in the screenshoot: https://imgur.com/a/cedfTO2

1technophile, 2020-08-02:
Try to erase the flash before uploading it should help

Pyromane, 2020-08-02:
I've already tried.

1technophile, 2020-08-02:
I suppose that you flash also the boot_app and the bootloader as it boots.

What is your esp32 board?

Pyromane, 2020-08-02:
Yes, I followed the instructions https://docs.openmqttgateway.com/upload/binaries.html#esp32
see Screenshoot: https://imgur.com/a/GcNiTNi

ESP32 DEVKIT V1 with an ESP-WROOM-32 chip

1technophile, 2020-08-02:
Could you try with DOUT instead of DIO ?

1technophile, 2020-08-02:
Alternatively here is the topic about [flashing with ESP32](https://community.openmqttgateway.com/t/flashing-pre-built-esp32-binary/579/19)

1technophile, 2020-08-02:
Another question, is it only related to `mqtt_user` or other variables set on the config portal are impacted too?

Pyromane, 2020-08-02:
> Could you try with DOUT instead of DIO ?

Select "SPI Mode" DOUT and use https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/bin/bootloader_dout_80m.bin ?
I'll try tomorrow

> Another question, is it only related to `mqtt_user` or other variables set on the config portal are impacted too?

The SSID and and the associated password work because the device connects to the WLAN and receives an IP.
I can test all other parameters tomorrow.

1technophile, 2020-08-03:
Seems like an issue related to the additionnal parameters storage in the flash memory. Most of the time this is resolved by the flash erasing. It can be due also to problems with the partition schema.

Legion2, 2020-08-03:
@Pyromane how long is your user name, so how many characters does it contain?

Pyromane, 2020-08-03:
> > Could you try with DOUT instead of DIO ?
> 
> Select "SPI Mode" DOUT and use https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/bin/bootloader_dout_80m.bin ?
> I'll try tomorrow

Unchanged error message

> Most of the time this is resolved by the flash erasing.

I use the "erase" function of the flash tool utility from espressif

> @Pyromane how long is your user name, so how many characters does it contain?

Eight characters: mqsensor

Pyromane, 2020-08-03:
> @Pyromane how long is your user name, so how many characters does it contain?

The hint went in the right direction!
If the mqtt_pass has 20 characters, the mqtt_user is not saved. With 19 characters, it works fine.

20 characters:
```
*WM: [2] Parameters <\r><\n>
*WM: [2] -------------------- <\r><\n>
*WM: [2] server: 192.168.1.17<\r><\n>
*WM: [2] port: 1883<\r><\n>
*WM: [2] user: your_username<\r><\n>
*WM: [2] pass: 12345678901234567890<\r><\n>
*WM: [2] name: OpenMQTTGateway_ESP32_BLE<\r><\n>
*WM: [2] topic: home/<\r><\n>
*WM: [2] -------------------- <\r><\n>
*WM: [3] Sent wifi save page <\r><\n>

*WM: [1] config portal exiting <\r><\n>
{"mqtt_server":"192.168.1.17","mqtt_port":"1883","mqtt_user":"","mqtt_pass":"12345678901234567890","mqtt_topic":"home/","gateway_name":"OpenMQTTGateway_ESP32_BLE"}N: BLEinterval: 55555<\n>
```

19 characters:
```
*WM: [2] Parameters <\r><\n>
*WM: [2] -------------------- <\r><\n>
*WM: [2] server: 192.168.1.17<\r><\n>
*WM: [2] port: 1883<\r><\n>
*WM: [2] user: your_username<\r><\n>
*WM: [2] pass: 1234567890123456789<\r><\n>
*WM: [2] name: OpenMQTTGateway_ESP32_BLE<\r><\n>
*WM: [2] topic: home/<\r><\n>

*WM: [1] config portal exiting <\r><\n>
{"mqtt_server":"192.168.1.17","mqtt_port":"1883","mqtt_user":"your_username","mqtt_pass":"1234567890123456789","mqtt_topic":"home/","gateway_name":"OpenMQTTGateway_ESP32_BLE"}N: BLEinterval: 55555<\n>
```

1technophile, 2020-08-04:
@Legion2 thanks for the help, so you proposal on extending the password length to 30 won a bonus point

Legion2, 2020-08-04:
Maybe we can allow dynamic length using `String` type and increase the length of the WiFiManagerParameter? And show a warning in the UI.

We have a buffer overflow when using the WiFiManager, that explains why user_name is empty. The null character of the passwort overflowed into the user_name array:
https://github.com/1technophile/OpenMQTTGateway/blob/4f84e08219498102471e17493e728e4a6fdbb5c8/main/main.ino#L1019-L1024
because the length of the returned char array is +1 longer, they don't count the null character at the end.
https://github.com/tzapu/WiFiManager/blob/master/WiFiManager.cpp#L36

1technophile, 2020-08-04:
> Maybe we can allow dynamic length using String type and increase the length of the WiFiManagerParameter? And show a warning in the UI.

As wifimanager is only used in ESP with more resources compared to Arduino yes we can do that, would not it be better to use std::string ?

Legion2, 2020-08-04:
I never used std::string because it's not available for Arduino boards.

1technophile, 2020-08-06:
> I never used std::string because it's not available for Arduino boards.

ok

Legion2, 2020-08-09:
@Pyromane did you test what happens if you try to use a password with 21 characters? I'm interested if the UI will display an error or something when the input is too long.

1technophile, 2020-08-31:
As a first workaround, I have increased the size of the MQTT password, I don't want to bother too much with Wifi Manager as I would like to test and try [Autoconnect](https://github.com/Hieromon/AutoConnect)

closed 2020-09-02 by 1technophile