1technophile/OpenMQTTGateway #725: Memory leakage in getting manufacture data and the fix.

l8l8l, 2020-08-08:
BLE Scan runs out of memory when processing manufacture data, 'freemem'  reduced very quickly and leads frequent crash.

---------ZgatewayBT.ino-------------

      if (advertisedDevice.haveManufacturerData()) {
        char* manufacturerdata = BLEUtils::buildHexData(NULL, (uint8_t*)advertisedDevice.getManufacturerData().data(), advertisedDevice.getManufacturerData().length());
        Log.trace(F("Manufacturer Data: %s" CR), manufacturerdata);
        BLEdata.set("manufacturerdata", manufacturerdata);
        free(manufacturerdata); **<-  Add the pointer free here.**
      }
-------------------------------------
Tested with default_envs = esp32dev-ble, pubUnknownBLEServiceData, pubBLEManufacturerData, pubBLEServiceUUID are enabled to scan BLE Beacons. freemem stays at 50KB.

1technophile, 2020-08-08:
Thanks for pointing that

jmw6773, 2020-08-14:
@l8l8l , great find. This seems to have resolved the random crashing I was seeing on my ESP32 device.

closed 2020-08-18 by 1technophile