1technophile/OpenMQTTGateway #726: Not all parameters allow prod_env.ini configuration

mpember, 2020-08-09:
**Is your feature request related to a problem? Please describe.**
The User_config.h has some parameters that are unable to be changed by prod_env.ini values.

e.g. OMG_VERSION and ota_* parameters cannot be set in prod_end.ini

**Describe the solution you'd like**
wrapping the User_config.h definition lines with #ifndef and #endif lines makes it possible to override the parameters with values in prod_env.ini

e.g.
```
#ifndef ota_password
#  define ota_password "OTAPASSWORD"
#endif
```

1technophile, 2020-08-09:
good catch, if you are interested do not hesitate to submit a PR

mpember, 2020-08-09:
> good catch, if you are interested do not hesitate to submit a PR

I tend to use master releases for my builds.  I'll try rebase my fork and have a look.

1technophile, 2020-08-09:
> I'll try rebase my fork and have a look.

Do it from the `development` branch please

mpember, 2020-08-09:
> Do it from the `development` branch please

PR created.  It is 1:30am and I didn't do too many tests apart from a trial build on for of the boards.  If there are any issues, just shout.

1technophile, 2020-08-09:
> It is 1:30am and I didn't do too many tests apart

Do not worry no rush ;-)

closed 2020-08-10 by 1technophile