1technophile/OpenMQTTGateway #733: Moneual IR- RAW Problem

Trainerr, 2020-08-16:
After i flash the new 0.9.4 Version, i can't control my Moneual Vakuum Roboter via IR. 

in the last 0.9.3 Version, everything works. 

This command i receive:
```
N: Received json : {"value":1282166610,"protocol":-1,"bits":9,"hex":"0x4C6C4F52","protocol_name":"UNKNOWN","raw":"826,418,824,1260,400,422,402,420,1666,414,828,416,1220,854,404,834,402"}
N: Received json : {"value":1282166610,"protocol":-1,"bits":9,"hex":"0x4C6C4F52","protocol_name":"UNKNOWN","raw":"826,418,824,1260,400,422,402,420,1666,414,828,416,1220,854,404,834,402"}
```

and this one i will send before:
```N: Received json : {"protocol_name":"IR_Raw","raw":"862,380,860,1222,412,412,436,384,1698,386,860,380,1256,820,438,798,436"}```

how can i Controller my moneual?

1technophile, 2020-08-16:
Like that : `{"protocol_name":"Raw","raw":"862,380,860,1222,412,412,436,384,1698,386,860,380,1256,820,438,798,436"}`

Trainerr, 2020-08-16:
oh thank you. now i change the protokol_name to Raw and it work again.

closed 2020-08-16 by Trainerr