1technophile/OpenMQTTGateway #734: Feature syslog option

michapr, 2020-08-16:
What do you think about the option to add the syslog.h to the project?

In such a case you could debug the device if it is not more connected via serial.
Updating via OTA is working fine - a log via network could be helpful.

I have added it in most of my projects and think it is useful.

Thanks!
Michael

1technophile, 2020-08-16:
You mean this library?
https://github.com/arcao/Syslog

michapr, 2020-08-17:
yes, this one. sorry,... forgot the link ;)

michapr, 2020-10-18:
I really still think it would be a good idea to have this syslog feature.
If enabled then send the same information additional to syslog server with same level.

In this time I add it after every update the needed output lines...
But I think it can be done by a new function, maybe.

Thanks

DavidStacer, 2020-12-28:
In many of my projects, I've been using a ESP-01 (ESP8266) with some code to send Syslog messages. Connected the TX pin on the ESP32 to the RX of the ESP-01.  ESP-01's job is only to send any and all text it receives over the serial port from the ESP32 to a syslog server.

michapr, 2021-01-08:
It's not big deal to add it (technical) - it is working, if define extra places with extra lines (code) for syslog output.
(I use it in my BLE gateways)

But I had no idea how to solve it "elegantly" with a function, so that syslog can use the same procedure as standard logging.

1technophile, 2021-01-26:
> But I had no idea how to solve it "elegantly" with a function, so that syslog can use the same procedure as standard logging.

Should maybe implemented at [ArduinoLog](https://github.com/thijse/Arduino-Log) level ?

Nikolay-Ch, 2021-08-19:
This feature is very useful to collect all logs in one place

h2zero, 2021-08-19:
A similar idea that could be used more directly with arduinolog would be a telnet server. This is what I use to monitor the logs of my devices when not connected with serial port.

Nikolay-Ch, 2021-08-19:
> A similar idea that could be used more directly with arduinolog would be a telnet server. This is what I use to monitor the logs of my devices when not connected with serial port.
Hm.... In my own network, I use FluentD to collect all logs (from switches, firewalls, and other devices) in one location... So, it is useful to use one service to collect all logs... And also SysLog format is standard, so it can be used with different log-collectors in different OSs (windows, linux etc)...

tablatronix, 2021-09-19:
I also want syslog capability, I use syslogging for all my iot devices, tasmota has it build in, and I add it to all my projects. 
I have not looked at tasmotas impelmentation, but I use streamutils and arcao/Syslog.

Is there any way to make plugins for this, or any existing logging mechanism to hook into ? I will take a look and see if there is an easy way to code it in, or a PR to expose existing logging.

If not I can insert streamutils and redirect the serial stream if its harcoded.

michapr, 2022-12-15:
any progress here? 
Would be really nice to have this option to check the configured messages wireless on syslog server...

ssalonen, 2024-03-05:
This would be useful to notice any errors and such

Did someone have code based on ArduinoLog?

ssalonen, 2024-03-05:
Further comment

I found this library https://github.com/dirkx/tee-log that would make the implementation rather straightforward

It basically implements the Print interface, and therefore is pluggable with ArduinoLog used by openmqttgateway. 

Tee-log logs to both syslog and optionally also to serial. Only caveat seems to be the fact that it is hardcoded to use [wireless network interface (WifiUdp)](https://github.com/dirkx/tee-log/blob/225afed6041594b10f20ffe1dc87733333c843d9/src/SyslogStream.cpp#L41)...probably not a large effort to upstream/fork ethernet support

Integration would then basically consist of introducing configuration parameters for the 

1. syslog destination ip and port config parameters
2. Flag to enable/disable syslogging functionality. If enabled, instead logging to Serial, we would use SyslogStream from tee-log instead.

tablatronix, 2024-03-06:
I have an implementation I use in my own libs that works. I use a redirector similar to tee and a syslog library.
I can see if I can bring it over in a fork or plugin

tablatronix, 2024-03-06:
I am using
```
/**
 * Local logging solutions, using steamutils and syslog
 * StreamUtils
 * https://github.com/Chris--A/PrintEx
 * Allows for stream buffering, redirection, mirroring etc.
 *
 * syslog
 * https://github.com/arcao/Syslog.git
 * An Arduino library for logging to Syslog server via `UDP` protocol in 
 * [IETF (RFC 5424)] and [BSD (RFC 3164)] message format
 */
```

rct, 2024-12-29:
Syslog would be a nice to have to capture all log output. 

Otherwise what options are there -- use a serial connection?  some sort of client for the web console?

rct, 2024-12-30:
I probably should have mentioned, one of the reasons I'm looking into logging options is I've just gotten started with this but noticed that my Lilygo board gets into periods where it keeps rebooting. I've found a number of similar but closed issues. I'll open a new issue once I actually have some info to go on.

[image]