1technophile/OpenMQTTGateway #749: BLEAdvertisedDeviceCallbacks is not invoked for all found devices

nemoo2580, 2020-08-30:
**Describe the bug**
Hello. I am working on integrating the iNode Energy Meter sensor to your gateway, but I noticed that sometimes method `onResult` from `MyAdvertisedDeviceCallbacks` is not invoked for all found devices during Bluetooth scan. It may be a bug in the NimBLE library, but can be mitigated with devices returned from scan method:
`BLEScanResults foundDevices = pBLEScan->start(Scan_duration / 1000, false);`.

**To Reproduce**
1. Compile `development` branch with profile `esp32dev-ble` and `trace` log level.
2. Run BLE device scanning,
3. See that log "Creating BLE buffer" (first log of `onResult` method) was displayed less time than number of found devices.

**Expected behavior**
Method `onResult` from `MyAdvertisedDeviceCallbacks` should be invoked for all found devices.

**Logs**
Logs with "TEST" phrase are from iterating over the found devices.
For invalid behavior (only one log "Creating BLE buffer", but 2 devices found):
> N: Scan begin
T: Creating BLE buffer
N: Device detected: 21:E5:8A:BC:EA:44
T: getDeviceByMac 21:E5:8A:BC:EA:44
T: Manufacturer Data: 0600010920026f7ecbc9c6f47153ac13a8e24f15fc09bbb30f8b078fc4
T: BLErssi -36
T: TxPower: -59
T: Ble distance 0.01
T: Pub json :{"id":"21:E5:8A:BC:EA:44","manufacturerdata":"0600010920026f7ecbc9c6f47153ac13a8e24f15fc09bbb30f8b078fc4","rssi":-36,"distance":0.007153} into custom topic: home/home_presence/OpenMQTTGateway_ESP32_BLE
T: getDeviceByMac 21:E5:8A:BC:EA:44
T: manufacturerdata 0600010920026f7ecbc9c6f47153ac13a8e24f15fc09bbb30f8b078fc4
T: Is it a iNode Energy Meter?
T: Remove manufacturer data
N: Subject: /BTtoMQTT/21E58ABCEA44
N: Received json : {"id":"21:E5:8A:BC:EA:44","rssi":-36,"distance":0.007153}
T: jsonPublishing
N: Found 2 devices, scan number 4 end deinit controller
N: TEST Device detected: 21:E5:8A:BC:EA:44
T: TEST Manufacturer Data: 0600010920026f7ecbc9c6f47153ac13a8e24f15fc09bbb30f8b078fc4
T: TEST RRSRI: -35
N: TEST Device detected: D0:F0:18:43:DB:9B
T: TEST Manufacturer Data: 90820f0072cc0000c409a00080
T: TEST RRSRI: -80
T: BT Task running on core: 0
N: Scan begin

For valid behavior:
> N: Scan begin
T: Creating BLE buffer
N: Device detected: 1D:BE:5F:18:49:47
T: getDeviceByMac 1D:BE:5F:18:49:47
T: Manufacturer Data: 06000109200243ae4950e701526a3b0764bcaeeca76051f682dfc3e47c
T: BLErssi -39
T: TxPower: -59
T: Ble distance 0.02
T: Pub json :{"id":"1D:BE:5F:18:49:47","manufacturerdata":"06000109200243ae4950e701526a3b0764bcaeeca76051f682dfc3e47c","rssi":-39,"distance":0.015927} into custom topic: home/home_presence/OpenMQTTGateway_ESP32_BLE
T: getDeviceByMac 1D:BE:5F:18:49:47
T: manufacturerdata 06000109200243ae4950e701526a3b0764bcaeeca76051f682dfc3e47c
T: Is it a iNode Energy Meter?
T: Remove manufacturer data
N: Subject: /BTtoMQTT/1DBE5F184947
N: Received json : {"id":"1D:BE:5F:18:49:47","rssi":-39,"distance":0.015927}
T: jsonPublishing
T: Creating BLE buffer
N: Device detected: D0:F0:18:43:DB:9B
T: getDeviceByMac D0:F0:18:43:DB:9B
T: Manufacturer Data: 90826300f0cf0000c409a20080
T: BLErssi -78
T: TxPower: -59
T: Ble distance 7.85
T: Pub json :{"id":"D0:F0:18:43:DB:9B","name":"iNode-43DB9B","manufacturerdata":"90826300f0cf0000c409a20080","rssi":-78,"txpower":8,"distance":7.85288} into custom topic: home/home_presence/OpenMQTTGateway_ESP32_BLE
T: getDeviceByMac D0:F0:18:43:DB:9B
T: manufacturerdata 90826300f0cf0000c409a20080
T: name iNode-43DB9B
T: Is it a INKBIRD?
T: Is it a iNode Energy Meter?
T: iNode Energy Meter data reading
T: getDeviceByMac D0:F0:18:43:DB:9B
T: add D0:F0:18:43:DB:9B
T: value 2500.00
T: value 99.00
T: value 53232.00
T: value 162.00
T: power: 2376.00
T: energy: 21.29
T: battAndLight: 162
T: battPercent: 80
T: battVoltage: 2.76
N: Subject: /BTtoMQTT/D0F01843DB9B
N: Received json : {"id":"D0:F0:18:43:DB:9B","name":"iNode-43DB9B","manufacturerdata":"90826300f0cf0000c409a20080","rssi":-78,"txpower":8,"distance":7.85288,"model":"INODE_EM","battery":80,"power":2376,"energy":21.2928}
T: jsonPublishing
N: Found 2 devices, scan number 1 end deinit controller
N: TEST Device detected: 1D:BE:5F:18:49:47
T: TEST Manufacturer Data: 06000109200243ae4950e701526a3b0764bcaeeca76051f682dfc3e47c
T: TEST RRSRI: -36
N: TEST Device detected: D0:F0:18:43:DB:9B
T: TEST Manufacturer Data: 90826300f0cf0000c409a20080
T: TEST RRSRI: -79
T: TEST CB
N: BLE Connect begin
N: BLE Connect end
T: Launching discovery of D0:F0:18:43:DB:9B
T: INodeEMDiscovery
T: CreateDiscoverySensor INodeEM-battery
T: Pub json :{"stat_t":"home/OpenMQTTGateway_ESP32_BLE/BTtoMQTT/D0F01843DB9B","name":"INodeEM-battery","uniq_id":"D0F01843DB9B-INodeEM-battery","dev_cla":"battery","val_tpl":"{{ value_json.batt | is_defined }}","unit_of_meas":"%"} into custom topic: homeassistant/sensor/D0F01843DB9B-INodeEM-battery/config
T: CreateDiscoverySensor INodeEM-power
T: Pub json :{"stat_t":"home/OpenMQTTGateway_ESP32_BLE/BTtoMQTT/D0F01843DB9B","name":"INodeEM-power","uniq_id":"D0F01843DB9B-INodeEM-power","dev_cla":"power","val_tpl":"{{ value_json.power | is_defined }}","unit_of_meas":"W"} into custom topic: homeassistant/sensor/D0F01843DB9B-INodeEM-power/config
T: CreateDiscoverySensor INodeEM-energy
T: Pub json :{"stat_t":"home/OpenMQTTGateway_ESP32_BLE/BTtoMQTT/D0F01843DB9B","name":"INodeEM-energy","uniq_id":"D0F01843DB9B-INodeEM-energy","val_tpl":"{{ value_json.energy | is_defined }}","unit_of_meas":"kWh"} into custom 
topic: homeassistant/sensor/D0F01843DB9B-INodeEM-energy/config
T: getDeviceByMac D0:F0:18:43:DB:9B
T: update D0:F0:18:43:DB:9B
T: macAdr D0:F0:18:43:DB:9B
T: isDisc 1
T: isWhtL 0
T: isBlkL 0
T: sensorModel 16

**Environment:**
* OpenMQTTGateway version: development
* ESP32 BLE (esp32dev-ble env)

1technophile, 2020-08-30:
That's an interesting finding. Would it be possible to try the [Scan ](https://github.com/h2zero/NimBLE-Arduino/blob/master/examples/Refactored_original_examples/BLE_scan/BLE_scan.ino) example from NimBLE Arduino library? And see if you get the same behavior.

We are using the last version of the library.

1technophile, 2020-08-30:
> iNode Energy Meter sensor to your gateway

That's cool :-)

nemoo2580, 2020-08-30:
The problem is with active scanning ([line](https://github.com/1technophile/OpenMQTTGateway/blob/development/main/ZgatewayBT.ino#L486)) and low RSSI (that "sometimes" mean at the edge of the range).

Probably iNode Energy Meter is losing "Scan request" packet from ESP32 or ESP32 doesn't get a response and AdvertisedDeviceCallbacks is not invoked [here](https://github.com/h2zero/NimBLE-Arduino/blob/d8a535d28ab7af8a4074f719fb87406c4b244a69/src/NimBLEScan.cpp#L115).

Passive scaning works fine. Is active scanning needed for other currently supported sensors (getting additional data for advertisement)?

In my use case i need work with low RSSI and only one iNode sensor, so maybe I have to use my own compilation with `pBLEScan->setActiveScan(false)`.

By the way, thanks for you good job with this project :)

1technophile, 2020-08-31:
> Passive scaning works fine. Is active scanning needed for other currently supported sensors (getting additional data for advertisement)?

We have been using active scanning from the beginning, I may do some test to see if I'm able to retrieve the same amount of data with passive scanning.

>In my use case I need work with low RSSI and only one iNode sensor, so maybe I have to use my own compilation with pBLEScan->setActiveScan(false).

If you want to integrate the device on a maintainable way, we may set this boolean as a parameter, at build, at run time by MQTT or alternating between passive and active.

But I will do some tests first.

closed 2020-11-26 by stale[bot]