1technophile/OpenMQTTGateway #753: Check GPIOInput state on demand

orxaz, 2020-09-01:
I can't find if is there a way to check, once booted, the state of a defined GPIOInput.
OMG reports the defined pin state at boot, but, at least in my case, never more until switched on or off.

Is there a way to check this?

Thanks.

1technophile, 2020-09-01:
OMG should publish the change of state of the `GPIOInput` defined, could you share the version that you are using and an extract of the serial monitor?

orxaz, 2020-09-02:
In fact, OMG publish when state changes, but that I would like is to be able to check it on demand, asking on a topic for example, or the state to be reported periodically.

Now I am out, will try to get serial output later.

Version is current, at least 5-6 days ago:
Can't find in code, only found "version_tag"

Any way, I have downloaded master version on august 26th.

1technophile, 2020-09-02:
> In fact, OMG publish when state changes, but that I would like is to be able to check it on demand, asking on a topic for example, or the state to be reported periodically.

This is not available yet, would you be interested in proposing a pull request?

closed 2023-02-04 by 1technophile