1technophile/OpenMQTTGateway #758: Pilight doesn't reconnect after MQTT broker restart

melyux, 2020-09-04:
Before submitting a problem please check the troubleshooting section
https://docs.openmqttgateway.com/upload/troubleshoot.html

**Describe the bug**
I run the BME280 and the Pilight modules together on the same device with OMG. I turned off my MQTT server for a couple of hours and then turned it back on, and the BME280 sensor carried on sending readings to the MQTT broker after the broker came back up. However, the Pilight module stopped sending received signals to the MQTT broker. I had to send a `{"cmd":"restart"}` to the `home/OpenMQTTGateway/commands/MQTTtoSYS/config` topic to get the Pilight readings coming through again.

Maybe related to #550, but that was fixed... maybe they were using a different RF module other than Pilight and the fix only worked for that?

**To Reproduce**
Steps to reproduce the behavior:
1. Enable the Pilight module and use it with some 433MHz sensors.
2. Turn off the MQTT broker for a few seconds.
3. Turn the MQTT broker back on.
4. Observe the `home/OpenMQTTGateway/PilighttoMQTT` topic to see that Pilight messages no longer come through.

**Expected behavior**
Pilight messages should resume being sent to the MQTT broker after a broker outage.

**Screenshots**
N/A

**Environment (please complete the following information):**
* OpenMQTTGateway version used (V0.9.3, 0.8, development): 0.9.4
* Library version related to the problem you have: Pilight

**Additional context**
Running on Wemos D1 Mini with superheterodyne receiver and BME280 pressure sensor.

* You should not have a compilation error if you use the versions of the libraries linked into the libraries folder, this badges show you the state of the compilation 
[image]
* If you are not sure this is a bug or an enhancement post your question to the forum below
[image]

1technophile, 2020-09-25:
I will try to reproduce your issue and report back

melyux, 2020-11-27:
Were you able to reproduce it?

1technophile, 2020-11-29:
Not yet, let me try

1technophile, 2020-12-01:
If I switch OFF the broker for 20s I don't reproduce the problem (pilight continue to receive messages after the broker restart) with v0.9.5 (I don't have a BME280
I will test with several minutes

1technophile, 2020-12-01:
Could be the same thing as #784 ?

1technophile, 2020-12-01:
Working also if the broker is stopped several minutes

closed 2022-01-09 by stale[bot]