1technophile/OpenMQTTGateway #760: ZgatewayBT - XMTZC05HM process_scale_v2 - how to add isStabilized ??

zibous, 2020-09-06:
**Is your feature request related to a problem? Please describe.**
How can I set the value for isStabilized (**isStabilized = ctrlByte1 & (1<<5)**) and use it in processing?
Mi Body Composition Scale 2 , XMTZC05HM

**Describe the solution you'd like**
At the moment, data is always being evaluated and published, even if the measurement is not yet finished.
It would be better if only the data of the measurement are published when it has ended (isStabilized).

**Describe alternatives you've considered**
lolouk44 has a very good approach to processing it with Python that works perfectly.
see: https://github.com/lolouk44/xiaomi_mi_scale/blob/master/src/Xiaomi_Scale.py

**Additional context**
```
for (sdid, desc, data) in dev.getScanData():
  
  if data.startswith('1b18') and sdid == 22:
     
     data2 = bytes.fromhex(data[4:])
     ctrlByte1 = data2[1]

     isStabilized = ctrlByte1 & (1<<5)              <--- see here
     hasImpedance = ctrlByte1 & (1<<1)

     measunit = data[4:6]
     measured = int((data[28:30] + data[26:28]), 16) * 0.01
     unit = ''
	 
     if measunit == "03": unit = 'lbs'
     if measunit == "02": unit = 'kg' ; measured = measured / 2
     miimpedance = str(int((data[24:26] + data[22:24]), 16))
     
	 if unit and isStabilized:
	    ...
```

1technophile, 2020-09-12:
Hi,

This is a good idea, if I get it well, we need to connect to the scale so as to retrieve this flag? There is no possibility to have it as an advertised data?

zibous, 2020-09-13:
Hi,
I have tested this with python and only when the measurement has stabilized can the date of the measurement be recorded and the data published. Not all data is available beforehand and the result is incorrect.

```
#!/usr/bin/env python3
# -*- coding: utf-8 -*-

import sys
sys.path.append("..")

try:
    from datetime import datetime
    import time

    from conf import *
    from lib import logger
    
except Exception as e:
    print('Import error {}, check requirements.txt'.format(e))

log = logger.Log(__name__, LOG_LEVEL)

class Miscale2Decoder():

    resultData = None
    version = '1.0.0'

    def __init__(self, dev):

        log.debug("Init Miscale2Decoder")

        self.bledevice = dev
        self.scanData = dev.getScanData()
        self.data = None
        self.devicedata = list()

        self.loadRemoved = False
        self.isStabilized = False
        self.hasImpedance = False

        self.timestamp = None
        self.measured = 0.00
        self.calcweight = 0.00
        self.impedance = 0
        self.unit = None

        self.lastscan = str(datetime.today().strftime('%Y-%m-%d %H:%M:%S'))
        self.resultData = None

        if self.bledevice and self.scanData:
            self.__decodedata__()
        else:
            log.debug("✘ NO DEVICE OR DATA present !")

    def getData(self):
        return self.resultData

    def __decodeWeight__(self):

        if self.data:
            # get weight
            self.measured = 0.00
            self.unit = None

            self.measured = int((self.data[28:30] + self.data[26:28]), 16) * 0.01
            log.debug("✔︎ (Measured 1:{}, p1:{}, p2:{}".format(self.measured, self.data[28:30], self.data[26:28]))

            # data2=bytes.fromhex(self.data[4:])
            # mi_easured = (((data2[12] & 0xFF) << 8) | (data2[11] & 0xFF))  # /200 kg
            # log.debug("✔︎ (Measured 2:{}".format(mi_easured))
[... 122 lines cut ...]
                        # final check and build result data
                        if self.timestamp and (str(self.timestamp) > self.lastscan):
                            self.lastscan = str(self.timestamp)
                            self.__setResults__()

                    else:
                        log.debug("✘ Device {} {} is sleeping !".format(self.bledevice.addr, self.bledevice.addrType))

                elif sdid == MI2_DATAFLAG or sdid == MI2_NOSERVICE or sdid == 9:
                    # (1, 'Flags', '06'),
                    # (2, 'Incomplete 16b Services', '0000181b-0000-1000-8000-00805f9b34fb'),
                    log.debug(("✘ SSID:{}, Data:{}").format(sdid, data))
                    continue
                elif sdid == MI2_MANUFACTORID:
                    # (255, 'Manufacturer', '57015ccad34cee74')
                    log.debug(("✘ SSID:{}, Data:{}").format(sdid, data))
                    continue
                else:
                    log.debug("✘ New unknown packet: type={} data={} !".format(sdid, data))

        except Exception as e:
            log.error(f"Error {__name__}: {str(e)},  line {sys.exc_info()[-1].tb_lineno}")
            pass

```

zibous, 2020-09-26:
@1technophile 
Hi Florian,

I've tried it, but it doesn't quite work.

The data is currently published again and again, even if it has not changed. I would only like to receive data when a new measurement is made.

Only part of the information is published (unit, timestamps are missing) and the ESP32 always restarts.

**MQTT Payload**
```json
Received json : {
       "id":"5C:CA:D3:4C:EE:74",
        "name":"MIBFS",
        "manufacturerdata":"57015ccad34cee74",
        "rssi":-69,
        "distance":3.119501,
         "model":"XMTZC05HM",
         "device":"Mi Bodyscale v2",
         "user":"7",
         "measured":141.3,
         "weight":70.65,
         "impedance":496
        }
```

**Test:**
```script
// JsonObject& process_scale_v2(JsonObject& BLEdata) {
//   const char* servicedata = BLEdata["servicedata"].as<const char*>();

//   double weight = (double)value_from_service_data(servicedata, 22, 4, true) / 200;
//   double impedance = (double)value_from_service_data(servicedata, 18, 4, true);

//   //Set Json values
//   BLEdata.set("weight", (double)weight);
//   BLEdata.set("impedance", (double)impedance);

//   return BLEdata;
// }

// --------------------------- TEST MI BODY SCALE V2 ----------------------------------

// init the timestamp
String lastTimestamp = "1900-01-01 00:00:00";

int16_t stoi(String input, uint16_t index1) {
  return (int16_t)(strtol(input.substring(index1, index1 + 2).c_str(), NULL, 16));
}

int16_t stoi2(String input, uint16_t index1) {
  return (int16_t)(strtol((input.substring(index1 + 2, index1 + 4) + input.substring(index1, index1 + 2)).c_str(), NULL, 16));
}

JsonObject& process_scale_v2(JsonObject& BLEdata) {
  const char* pHex = BLEdata["servicedata"].as<const char*>();

  // std::string md = BLEdata["servicedata"];
  // uint8_t *mdp = (uint8_t *)d.getServiceData().data();
  // char *pHex = BLEUtils::buildHexData(nullptr, mdp, md.length());

  // # MiScale2 Raw Data Schema
  // # +------+------------------------+
  // # | byte |        function        |
  // # +------+------------------------+
  // # | 0    | Bit 0: unknown         |
  // # |      | Bit 1: kg              |
  // # |      | Bit 2: lbs             |
  // # |      | Bit 3: unknown         |
  // # |      | Bit 4: jin unit        |
  // # |      | Bit 5: stabilized      |
  // # |      | Bit 6: unknown         |
  // # |      | Bit 7: load removed    |
  // # +------+------------------------+
  // # | 1-2  | weight (little endian) |
  // # +------+------------------------+
  // # | 3-7  | unknown                |
  // # +------+------------------------+
  // # | 8-9  | sequence (big endian)  |
  // # +------+------------------------+

  String hex;
  hex = pHex;

  // free(pHex);

  // check controlbyte for stabilized, load removed..
  // bool loadRemoved = controlByte & (1 << 7)
  // bool isStabilized = controlByte & (1 << 5) != 0
  int controlByte = stoi2(hex, 4);
  Log.trace(F("Controlbyte is: %d" CR), controlByte);

  if (controlByte & (1 << 7) or controlByte & (1 << 5) != 0) {
    Log.trace(F("Not stabilized, skip and wait for the next one...."), CR);
    // return;
  }

  float impedance = stoi2(hex, 18);
  float measured = stoi2(hex, 22) * 0.01f;
  float weight = measured;

  int user = stoi(hex, 6);
  Log.trace(F("User is: %d" CR), user);

  int units = stoi(hex, 0);
  Log.trace(F("Units is: %d" CR), units);

  // decode the units and recalc the weight based on the unit
  String strUnits;

  if (units == 1) {
    // Chinese Catty
    strUnits = "jin";
    weight = measured * 0.50 * 0.50;
  } else if (units == 2) {
    //  MKS kg
    strUnits = "kg";
    weight = measured * 0.50;
  } else if (units == 3) {
    // Imperial pound
    weight = measured * 0.4536;
    strUnits = "lbs";
  }

  Log.trace(F("Weight is: %d" CR), weight);

  // decode the timestamp from the scale
  uint8_t month = stoi(hex, 8);
  uint8_t day = stoi(hex, 10);
  uint8_t hours = stoi(hex, 12);
  uint8_t minutes = stoi(hex, 14);
  uint8_t seconds = stoi(hex, 16);

  // build the timestamp
  String yearStr = String(String(stoi2(hex, 4)));
  String monthStr = month < 10 ? "0" + String(month) : String(month);
  String dayStr = day < 10 ? "0" + String(day) : String(day);
  String hoursStr = hours < 10 ? "0" + String(hours) : String(hours);
  String minuteStr = minutes < 10 ? "0" + String(minutes) : String(minutes);
  String secondStr = seconds < 10 ? "0" + String(seconds) : String(seconds);
  String time = yearStr + "-" + monthStr + "-" + dayStr + " " + hoursStr + ":" + minuteStr + ":" + secondStr;

  Log.trace(F("Time is: %s" CR), time);

  if (weight > 0 and impedance > 0 and lastTimestamp != time) {
    // publish only if we have new weight, impedance and a new scan
    BLEdata.set("device", "Mi Bodyscale v2");
    BLEdata.set("user", String(user));
    BLEdata.set("measured", (double)measured);
    BLEdata.set("weight", (double)weight);
    BLEdata.set("impedance", (double)impedance);
    BLEdata.set("unit", strUnits);
    BLEdata.set("scantime", time);
    BLEdata.set("lastscan", lastTimestamp);
    lastTimestamp = time;
    Log.trace(F("Publish new data: %s" CR), time);
    return BLEdata;
  }
}
```

dckiller51, 2020-10-22:
Hello,

I am using a BT-MQTT-GATEWAY on PIZERO. This function is ok on this model.

https://github.com/zewelor/bt-mqtt-gateway/blob/master/workers/miscale.py

1technophile, 2021-02-11:
@zibous 

Could it be possible to have some servicedata extract with different condition of weight(LBS/KG) and stabilisation. You can activate all the data publishing by setting all the parameters below this [link ](https://github.com/1technophile/OpenMQTTGateway/blob/1758746050bab3816d3cd5180cba9f415f69d631/main/config_BT.h#L66)to `true`

Here is some examples for XMTZC04HM:

name:"MI SCALE2" | manufacturerdata:"5701c8478cd2fc19" | rssi:-58 | distance:0.842869 | 223a2fb207010c171d1a" | servicedatauuid:"0x181d" | model:"XMTZC04HM" | weight:60.45} | stabilized | kg
-- | -- | -- | -- | -- | -- | -- | -- | -- | --
name:"MI SCALE2" | manufacturerdata:"5701c8478cd2fc19" | rssi:-62 | distance:1.429824 | 825000b207010c171a33" | servicedatauuid:"0x181d" | model:"XMTZC04HM" | weight:0.4} | stopped | kg
name:"MI SCALE2" | manufacturerdata:"5701c8478cd2fc19" | rssi:-59 | distance:1.01076 | 026c02b207010c171e2d" | servicedatauuid:"0x181d" | model:"XMTZC04HM" | weight:3.1} | not stabilized | kg

It would help me improving this for XMTZC05HM

1technophile, 2021-02-21:
I need data so as to implement this request. So I remove it from v0.9.6

closed 2022-01-09 by stale[bot]

DigiH, 2022-02-13:
I've done some further testing with my XMTZC05HM and found the following so far

"servicedata", "index", 2, "0" - unstabilised
"servicedata", "index", 2, "8" - unstabilised, stepped off scale
"servicedata", "index", 2, "2" - stabilised, on scale
"servicedata", "index", 2, "a" - stabilised, stepped off scale

Additionally, if "Weighing small Items" has been activated for the scale

"condition":["servicedata", 1, "6"] - unit = kg for objects
"condition":["servicedata", 1, "7"] - unit = lbs for objects

and in general if weighing objects or standing on the scale with shoes or sox, so that now impedance measuring is possible

"servicedata", "index", 3, "6" - impedance measuring ok
"servicedata", "index", 3, "4" - no impedance measuring

which could  be used as a condition impedance.

If anyone else with a XMTZC05HM can confirm these findings it would be great to implement them in the decoder.