1technophile/OpenMQTTGateway #761: Mijia BLE Temperature and Humidity Firmware 1.0.1_0068

tita2000, 2020-09-08:
Hi,

I don't know exactly, but I think since updating the sensors to the firmware above the battery status is no longer but in clear text to my mqtt broker. Temperature and humidity are still fine. The only thing i recognized is that the data is still send, but it seems to be hidden in the service data e.g. 5020aa0155aabbccddeeff0a100151. Found the following in another forum  https://www.domoticz.com/forum/viewtopic.php?t=20925&start=20

`Pos.	Value (Hex)	Description
0..1	95 FE	        UUID 16 bit (0xFE95)
2..5	50 20 AA 01	TBD (always 50 20 AA 01 for my three sensors)
6	17		Message counter (increments by 1 every message and then restart from 0 when reaches FF)
7..12	XX .. XX	Device BT address (6 bytes)
13	0D 	        Data type:
				0D -> Temperature & Humidity (4 bytes TL,TH,HL,HH)
				0A -> battery % (1 byte)
				06 -> only Humidity (2 bytes HL,HH)
				04 -> only Temperature (2 bytes TL, TH)
14	10		TBD (always 0x10 for my three sensors)
15	04		Nr of following bytes (depends on Data Type)
16..N	B9 00 5C 02	Data values:
				0x00B9 = 185 -> 18.5 C (Temperature)
				0x025C = 604 -> 60.4 % (Humidity)
`

This is the json payload I' able to receive:

`{
  "id": "FF:EE:DD:CC:BB:AA",
  -"rssi": -96,
  -"distance": 38.4873,
  -"servicedata": "5020aa0155aabbccddeeff0a100151"
  +"rssi": -87,
  +"distance": 18.07764,
  +"model": "LYWSDCGQ",
  +"tem": 21.3,
  +"hum": 47
}`

So there might be a change in how the battery status is send?!

I opened this issue as others might experience the same problem, hoping that there will be a solution.

Thanks in advance
Regards Timo

1technophile, 2020-09-08:
Hi,

Did you check your MQTT broker history, it is possible that the battery message is published apart from the temperature and humidity?

tita2000, 2020-09-09:
Yes, I did. Please see my second quote the temperature and humidity alternate with the servicedata. The last two bytes of the servicedata is in fact the battery status, but it isn't send in readyble json anymore.

tita2000, 2020-09-09:
Ok, I found a hint on the problem https://community.openmqttgateway.com/t/lywsdcgq-not-showing-battery-status/1086. So maybe in the next release it will be solved. Thank you

closed 2020-09-09 by tita2000