1technophile/OpenMQTTGateway #763: Any chance for OTA update possibility for the AVATTO S06 IR gateways?

DigiH, 2020-09-10:
Having seen the 0.9.5-beta release I thought I'd test it with my ESP8266 RF gateway, ESP32 BLE gateway and my IR gateway, for which I very much enjoy the AVATTO S06, after reading your blog entry.

For both the RF and BLE gateways the OTA updates worked flawlessly, but with the AVATTO S06 it failed, due to the lack of enough space.

Current from SYStoMQTT:
"freemem":35728

OTA upload error:
13:15:57 [INFO]: Upload size: 492608
13:15:57 [INFO]: Sending invitation to: 192.168.0.50
Authenticating...FAIL
13:16:00 [ERROR]: ERR: ERROR[4]: Not Enough Space

Since the Avatto would benefit the most from OTA updates - due to soldering required in contrast to the ESP8266/ESP32 - is there any leeway in reducing the Avatto specific build size so that it, and any future updates might fit into the 1mb flash for OTA updating? 

Probably just wishful thinking on my side, but I tought it might be worth asking.

Thanks

1technophile, 2020-09-10:
That's a good point, maybe playing with the partition sizes will enable to make the OTA works.

DigiH, 2020-09-10:
Where as previously with the default partition sizes I got

RAM:   [====      ]  38.3% (used 31392 bytes from 81920 bytes)
Flash: [=====     ]  49.5% (used 376980 bytes from 761840 bytes)

still failing with
17:01:47 [INFO]: Upload size: 381136
17:01:47 [INFO]: Sending invitation to: 192.168.0.50
Authenticating...FAIL
17:01:47 [ERROR]: ERR: ERROR[4]: Not Enough Space

now with adding the build_flag

-Wl,-Teagle.flash.1m64.ld
or 
board_build.ldscript = eagle.flash.1m64.ld

it's

RAM:   [====      ]  38.3% (used 31392 bytes from 81920 bytes)
Flash: [====      ]  39.3% (used 376980 bytes from 958448 bytes)

with a much better available total Flash size, but still failing

17:31:04 [INFO]: Upload size: 381136
17:31:04 [INFO]: Sending invitation to: 192.168.0.50
Authenticating...FAIL
17:31:04 [ERROR]: ERR: ERROR[4]: Not Enough Space

I suppose I need to install a build with such changes via hard wired conenctions first, as to have the partition sizes changes take effect. As the reported "freemem" is always 35728. Or is an OTA Flash split possible?

In that respect, is there a minimum Flash split that the IR build needs for spiffs/data storage, i.e. is 1m64.ld ok or 1m128.ld etc.?

Thanks

1technophile, 2020-09-10:
> I suppose I need to install a build with such changes via hard wired conenctions first, as to have the partition sizes changes take effect. As the reported "freemem" is always 35728. Or is an OTA Flash split possible?

You will need to hard wire, erase the flash and reupload it. The change of partition size should be done like that.

DigiH, 2020-09-10:
@1technophile thanks for the confirmation. Will wait until I receive my additional Avattos, then do a soldering/flashing session. Might be an idea to include an appropriate ldscript to the Avatto setting as a default, as most users might want to do OTA updates after the initial upload.

Any recommendations which one might be the most appropraite for when I heat up the soldering - …1m64.ld ok or …1m128.ld or even …1m.ld etc.?

Thanks

DigiH, 2020-09-24:
I'm using 

`board_build.ldscript = eagle.flash.1m64.ld`

now, which is fine, as **'-DESPWifiManualSetup=true'** does save quite a bit of space for the build.

Confirmed it's working fine, with serveral OTA updates done for testing now.

Closing

closed 2020-09-24 by DigiH

1technophile, 2020-09-25:
Maybe we should add:
`board_build.ldscript = eagle.flash.1m64.ld`

Into platformio.ini definition of the board?

DigiH, 2020-09-25:
`eagle.flash.1m64.ld` obviously has only 64 Kb less sketch space than my initial `eagle.flash.1m.ld`, but as the latter does have 0Kb spiff partition, and I seemed to have problems with OTA upload with it, I think `eagle.flash.1m64.ld` as a default in the Avatto platformio.ini definition would be a good idea, as it then assures possible future OTA updates.

The only problem seems to be when WiFi manager is not deactivated, the build requires

RAM:   [====      ]  41.6% (used 34108 bytes from 81920 bytes)
Flash: [=====     ]  50.7% (used 486012 bytes from 958448 bytes)

which looks too tight, and is the reason why I initially tried `eagle.flash.1m.ld`.

Also just asked in the 'loosing WiFi connection' community thread, as I saw that @Noobud was using `eagle.flash.1m.ld` there.

Have you tried both `eagle.flash.1m64.ld` and `eagle.flash.1m.ld`. Could the 0Kb spiff partition cause issues here?

DigiH, 2020-09-25:
If the 0Kb spiff partition really is the culprit - pending on the experiences of others with the `eagle.flash.1m.ld` - an alternative would be a custom

`eagle.flash.1m32.ld`

ldscript, which would also allow the sketch size to hold two 486012 bytes build which include WiFiManager for OTAs with the Avattos ;)

DigiH, 2020-09-26:
Too much fiddling with the above ;) and also still very tight, when WiFiManager is included, and might not be enough with future updates, which could end up slightly larger that the current 0.9.5 build.

> 
> Maybe we should add:
> board_build.ldscript = eagle.flash.1m64.ld
> 
> Into platformio.ini definition of the board?

So yes, I think including eagle.flash.1m64.ld as a default in the platformio.ini ist a good idea. In the docs it could then be stated, that for Avatto OTA updates WiFiManager has to be disabled, and the WiFi and MQTT info manually defined.